metadriverse/metadrive #461: How to change the environment used in metadrive/examples/train_generalization_experiment.py?

ZJP-c4cld, 2023-06-27:
I wan to train an autonomous driving policy in environments other than the default MetaDriveEnv, such as WaymoEnv, by running train_generalization_experiment.py. Once the env variable is changed from MetaDriveEnv to WaymoEnv, the following error occurs.

Failure # 1 (occurred at 2023-06-27_17-36-53)
Traceback (most recent call last):
  File "/root/anaconda3/envs/metadriver/lib/python3.10/site-packages/ray/tune/execution/ray_trial_executor.py", line 1070, in get_next_executor_event
    future_result = ray.get(ready_future)
  File "/root/anaconda3/envs/metadriver/lib/python3.10/site-packages/ray/_private/client_mode_hook.py", line 105, in wrapper
    return func(*args, **kwargs)
  File "/root/anaconda3/envs/metadriver/lib/python3.10/site-packages/ray/_private/worker.py", line 2311, in get
    raise value
ray.exceptions.RayActorError: The actor died because of an error raised in its creation task, [36mray::PPO.__init__()[39m (pid=24049, ip=172.30.64.140, repr=PPO)
  File "/root/anaconda3/envs/metadriver/lib/python3.10/site-packages/ray/rllib/evaluation/worker_set.py", line 239, in _setup
    self.add_workers(
  File "/root/anaconda3/envs/metadriver/lib/python3.10/site-packages/ray/rllib/evaluation/worker_set.py", line 612, in add_workers
    raise result.get()
  File "/root/anaconda3/envs/metadriver/lib/python3.10/site-packages/ray/rllib/utils/actor_manager.py", line 473, in __fetch_result
    result = ray.get(r)
ray.exceptions.RayActorError: The actor died because of an error raised in its creation task, [36mray::RolloutWorker.__init__()[39m (pid=24107, ip=172.30.64.140, repr=<ray.rllib.evaluation.rollout_worker.RolloutWorker object at 0x7f4af3f83d90>)
  File "/root/anaconda3/envs/metadriver/lib/python3.10/site-packages/ray/rllib/evaluation/rollout_worker.py", line 585, in __init__
    self.env = env_creator(copy.deepcopy(self.env_context))
  File "/root/anaconda3/envs/metadriver/lib/python3.10/site-packages/ray/rllib/algorithms/algorithm.py", line 2181, in <lambda>
    return env_id, lambda cfg: env_specifier(cfg)
  File "/root/aliyun/metadrive/metadrive/envs/real_data_envs/waymo_env.py", line 81, in __init__
    super(WaymoEnv, self).__init__(config)
  File "/root/aliyun/metadrive/metadrive/envs/base_env.py", line 199, in __init__
    merged_config = self._merge_extra_config(config)
  File "/root/aliyun/metadrive/metadrive/envs/real_data_envs/waymo_env.py", line 85, in _merge_extra_config
    config = self.default_config().update(config, allow_add_new_key=False)
  File "/root/aliyun/metadrive/metadrive/utils/config.py", line 122, in update
    raise KeyError(
KeyError: "'{'start_seed', 'random_traffic', 'traffic_density'}' does not exist in existing config. Please use config.update(...) to update the config. Existing keys: dict_keys(['random_agent_model', 'agent_policy', 'num_agents', 'is_multi_agent', 'allow_respawn', 'delay_done', 'return_single_space', 'manual_control', 'controller', 'decision_repeat', 'discrete_action', 'discrete_steering_dim', 'discrete_throttle_dim', 'use_multi_discrete', 'action_check', 'use_render', 'debug', 'disable_model_compression', 'cull_scene', 'use_chase_camera_follow_lane', 'camera_height', 'camera_dist', 'camera_pitch', 'camera_smooth', 'prefer_track_agent', 'draw_map_resolution', 'top_down_camera_initial_x', 'top_down_camera_initial_y', 'top_down_camera_initial_z', 'show_logo', 'vehicle_config', 'target_vehicle_configs', 'window_size', 'physics_world_step_size', 'show_fps', 'global_light', 'debug_physics_world', 'debug_static_world', 'pstats', 'image_observation', 'image_on_cuda', '_disable_detector_mask', 'rgb_clip', 'force_render_fps', 'force_destroy', 'num_buffering_objects', 'max_distance', '_debug_crash_object', 'horizon', 'max_step_per_agent', 'show_interface_navi_mark', 'show_mouse', 'show_skybox', 'show_terrain', 'show_interface', 'interface_panel', 'show_coordinates', 'multi_thread_render', 'multi_thread_render_mode', 'record_episode', 'replay_episode', 'only_reset_when_replay', 'force_reuse_object_name', 'waymo_data_directory', 'start_scenario_index', 'num_scenarios', 'store_map', 'store_map_buffer_size', 'sequential_seed', 'no_traffic', 'traj_start_index', 'traj_end_index', 'replay', 'no_static_traffic_vehicle', 'use_waymo_observation', 'success_reward', 'out_of_road_penalty', 'crash_vehicle_penalty', 'crash_object_penalty', 'driving_reward', 'speed_reward', 'use_lateral_reward', 'crash_vehicle_cost', 'crash_object_cost', 'out_of_road_cost', 'out_of_route_done', 'crash_vehicle_done', 'relax_out_of_road_done', 'coordinate_transform'])."

During handling of the above exception, another exception occurred:

[36mray::PPO.__init__()[39m (pid=24049, ip=172.30.64.140, repr=PPO)
  File "/root/anaconda3/envs/metadriver/lib/python3.10/site-packages/ray/rllib/algorithms/algorithm.py", line 441, in __init__
    super().__init__(
  File "/root/anaconda3/envs/metadriver/lib/python3.10/site-packages/ray/tune/trainable/trainable.py", line 169, in __init__
    self.setup(copy.deepcopy(self.config))
  File "/root/anaconda3/envs/metadriver/lib/python3.10/site-packages/ray/rllib/algorithms/algorithm.py", line 566, in setup
    self.workers = WorkerSet(
  File "/root/anaconda3/envs/metadriver/lib/python3.10/site-packages/ray/rllib/evaluation/worker_set.py", line 191, in __init__
    raise e.args[0].args[2]
KeyError: "'{'start_seed', 'random_traffic', 'traffic_density'}' does not exist in existing config. Please use config.update(...) to update the config. Existing keys: dict_keys(['random_agent_model', 'agent_policy', 'num_agents', 'is_multi_agent', 'allow_respawn', 'delay_done', 'return_single_space', 'manual_control', 'controller', 'decision_repeat', 'discrete_action', 'discrete_steering_dim', 'discrete_throttle_dim', 'use_multi_discrete', 'action_check', 'use_render', 'debug', 'disable_model_compression', 'cull_scene', 'use_chase_camera_follow_lane', 'camera_height', 'camera_dist', 'camera_pitch', 'camera_smooth', 'prefer_track_agent', 'draw_map_resolution', 'top_down_camera_initial_x', 'top_down_camera_initial_y', 'top_down_camera_initial_z', 'show_logo', 'vehicle_config', 'target_vehicle_configs', 'window_size', 'physics_world_step_size', 'show_fps', 'global_light', 'debug_physics_world', 'debug_static_world', 'pstats', 'image_observation', 'image_on_cuda', '_disable_detector_mask', 'rgb_clip', 'force_render_fps', 'force_destroy', 'num_buffering_objects', 'max_distance', '_debug_crash_object', 'horizon', 'max_step_per_agent', 'show_interface_navi_mark', 'show_mouse', 'show_skybox', 'show_terrain', 'show_interface', 'interface_panel', 'show_coordinates', 'multi_thread_render', 'multi_thread_render_mode', 'record_episode', 'replay_episode', 'only_reset_when_replay', 'force_reuse_object_name', 'waymo_data_directory', 'start_scenario_index', 'num_scenarios', 'store_map', 'store_map_buffer_size', 'sequential_seed', 'no_traffic', 'traj_start_index', 'traj_end_index', 'replay', 'no_static_traffic_vehicle', 'use_waymo_observation', 'success_reward', 'out_of_road_penalty', 'crash_vehicle_penalty', 'crash_object_penalty', 'driving_reward', 'speed_reward', 'use_lateral_reward', 'crash_vehicle_cost', 'crash_object_cost', 'out_of_road_cost', 'out_of_route_done', 'crash_vehicle_done', 'relax_out_of_road_done', 'coordinate_transform'])."

Failure # 2 (occurred at 2023-06-27_17-37-03)
Traceback (most recent call last):
  File "/root/anaconda3/envs/metadriver/lib/python3.10/site-packages/ray/tune/execution/ray_trial_executor.py", line 1070, in get_next_executor_event
    future_result = ray.get(ready_future)
  File "/root/anaconda3/envs/metadriver/lib/python3.10/site-packages/ray/_private/client_mode_hook.py", line 105, in wrapper
    return func(*args, **kwargs)
  File "/root/anaconda3/envs/metadriver/lib/python3.10/site-packages/ray/_private/worker.py", line 2311, in get
    raise value
ray.exceptions.RayActorError: The actor died because of an error raised in its creation task, [36mray::PPO.__init__()[39m (pid=24367, ip=172.30.64.140, repr=PPO)
  File "/root/anaconda3/envs/metadriver/lib/python3.10/site-packages/ray/rllib/evaluation/worker_set.py", line 239, in _setup
    self.add_workers(
  File "/root/anaconda3/envs/metadriver/lib/python3.10/site-packages/ray/rllib/evaluation/worker_set.py", line 612, in add_workers
    raise result.get()
  File "/root/anaconda3/envs/metadriver/lib/python3.10/site-packages/ray/rllib/utils/actor_manager.py", line 473, in __fetch_result
    result = ray.get(r)
[... cut ...] e', 'sequential_seed', 'no_traffic', 'traj_start_index', 'traj_end_index', 'replay', 'no_static_traffic_vehicle', 'use_waymo_observation', 'success_reward', 'out_of_road_penalty', 'crash_vehicle_penalty', 'crash_object_penalty', 'driving_reward', 'speed_reward', 'use_lateral_reward', 'crash_vehicle_cost', 'crash_object_cost', 'out_of_road_cost', 'out_of_route_done', 'crash_vehicle_done', 'relax_out_of_road_done', 'coordinate_transform'])."

 How to solve this problem? Thanks for help in advance.

QuanyiLi, 2023-06-27:
Please try the training script here: https://github.com/metadriverse/scenarionet/blob/main/scenarionet_training/scripts/train_waymo.py

This is a new one allowing you to train agents with more than large-scale scenarios like (50,000) with a ```MultiWorkerPPO```. You can also use the standard PPO by switching it to ```PPO```

c4cld, 2023-06-27:
@QuanyiLi Thank you for your selfless help  ^_^
I have run following code to install scenarionet:
      git clone git@github.com:metadriverse/scenarionet.git
      cd scenarionet
      pip install -e .
However, when I test scenarionet_training/scripts/train_waymo.py, the following error occurs:
ModuleNotFoundError: No module named 'metadrive.envs.gym_wrapper'

I check the https://github.com/metadriverse/metadrive/tree/main/metadrive/envs and find that "gym_wrapper" is not here.
How to solve this issue?

QuanyiLi, 2023-06-27:
Ah, I think you should ```git pull``` the recent metadrive main branch. We have many new features that don't release on PyPI yet. In addition, please use ```pip install -e .[train]``` to install the ScenarioNet repo. It will install the training framework together.

c4cld, 2023-06-28:
@QuanyiLi Thank you for you guidance!
I have installed the recent version of metadrive and ScenarioNet repo. Once I download the waymo dataset, I can run the code.
But I dont know which version of waymo dataset need to be downloaded.
[image]
Should I download all the data, or is it sufficient to download only the initial version of the data?

QuanyiLi, 2023-06-28:
The latest one is ok like 1.1 or 1.2. You can just download a portion of them like one or two files. Each file contains approximately 60 scenarios.

发自我的iPhone

QuanyiLi, 2023-06-29:
You can find the supported arguments of the script by adding ```-h``` flag. And there is one called ```--num_files``` which can be used to specify how many files or tf records are in the raw_data_dir. In your case, you can set it to 2 for fixing this issue.

c4cld, 2023-06-30:
@QuanyiLi Thank you very much. I set --num_files to 2 and the issue is addressed.

However, scenarionet.converter.waymo.utils seems unable to read the Waymo dataset, with the following error message.

INFO:scenarionet.converter.waymo.utils:
 Reading raw data
INFO:scenarionet.converter.waymo.utils:Use one worker, as num_scenario < num_workers:
  0%|                                                     | 0/2 [00:00<?, ?it/s]
Traceback (most recent call last):
  File "/root/anaconda3/envs/scenarionet/lib/python3.8/runpy.py", line 192, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/root/anaconda3/envs/scenarionet/lib/python3.8/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/root/aliyun/scenarionet/scenarionet/convert_waymo.py", line 64, in <module>
    scenarios = get_waymo_scenarios(
  File "/root/aliyun/scenarionet/scenarionet/converter/waymo/utils.py", line 456, in get_waymo_scenarios
    all_result = read_from_files([waymo_data_directory, file_list])
  File "/root/aliyun/scenarionet/scenarionet/converter/waymo/utils.py", line 481, in read_from_files
    scenario.ParseFromString(data)
google.protobuf.message.DecodeError: Error parsing message with type 'waymo.open_dataset.Scenario'

There is a Q&A involving a similar question to mine, the website is https://github.com/waymo-research/waymo-open-dataset/issues/343.  I found that your approach to reading the Waymo dataset is basically the same as the solution in this Q &A.  Therefore, I suspect that some of my third-party libraries are of the wrong version. I checked my third-party versions, as shown below.

absl-py                      1.4.0
aiohttp                      3.6.0
aiohttp-cors                 0.7.0
aioredis                     2.0.1
aiosignal                    1.3.1
ale-py                       0.7.5
array-record                 0.4.0
astunparse                   1.6.3
async-timeout                3.0.1
atari-py                     0.2.9
attrs                        23.1.0
auditwheel                   5.3.0
beautifulsoup4               4.12.2
blessed                      1.20.0
cachetools                   5.3.1
certifi                      2023.5.7
chardet                      3.0.4
charset-normalizer           3.1.0
click                        8.1.3
click-plugins                1.1.1
cligj                        0.7.2
cloudpickle                  2.2.1
cmake                        3.26.4
colorama                     0.4.6
colorful                     0.5.5
configparser                 5.3.0
contourpy                    1.1.0
cycler                       0.11.0
dask                         2023.3.1
decorator                    5.1.1
dm-tree                      0.1.8
docker-pycreds               0.4.0
einsum                       0.3.0
etils                        1.3.0
exceptiongroup               1.1.1
Farama-Notifications         0.0.4
filelock                     3.12.2
Fiona                        1.9.4.post1
flatbuffers                  23.5.26
fonttools                    4.40.0
frozenlist                   1.3.3
fsspec                       2023.6.0
gast                         0.4.0
geopandas                    0.13.2
gitdb                        4.0.10
GitPython                    3.1.31
google                       3.0.0
[... 122 lines cut ...]
termcolor                    2.3.0
threadpoolctl                3.1.0
tifffile                     2023.4.12
toml                         0.10.2
tomli                        2.0.1
toolz                        0.12.0
torch                        2.0.1
tqdm                         4.65.0
trimesh                      3.22.2
triton                       2.0.0
typeguard                    2.13.3
typing_extensions            4.6.3
tzdata                       2023.3
urllib3                      1.26.16
wandb                        0.12.1
waymo-open-dataset-tf-2.11.0 1.5.0
wcwidth                      0.2.6
Werkzeug                     2.3.6
wheel                        0.38.4
wrapt                        1.14.1
yapf                         0.30.0
yarl                         1.9.2
zipp                         3.15.0

Is it because of the wrong version of the third-party library that causes the inability to read the data from the Waymo dataset? If so, how should it be corrected? If not, what could be the cause of the problem?

QuanyiLi, 2023-06-30:
Yes, you should install waymo toolkit: ```pip install waymo-open-dataset-tf-2-11-0==1.5.0```. That's weird. I think it will throw an error message, if you don't have this toolkit. Anyway, I will throw this error message to where your bug occurs. Thanks.

c4cld, 2023-07-02:
@QuanyiLi  Hello, thank you for your help. Now I am able to determine that my problem is caused by the incompatibility between the versions of third-party libraries.

Firstly, I installed a virtual environment by following command.
conda create -n scenarionet python==3.8
conda activate scenarionet

Secondly,  I switched to the path to ./scenarionet and ran the following command.
pip install -e .

Thirdly, I switched to the path to ./metadrive and ran the following command.
pip install -e .

Fourthly, I switched to the path to ./scenarionet and tried to convert tfrecord by following command.
python -m scenarionet.convert_waymo -d waymo --raw_data_path /path/to/tfrecords --num_workers=16
I received an error message indicating that I should install waymo-open-dataset-tf-2-11-0==1.5.0.
So I installed the library by pip install waymo-open-dataset-tf-2-11-0==1.5.0.
During the installation process, some versions of third-party libraries were modified.

Then, I tried to convert tfrecord again and received following error message.
INFO:scenarionet.converter.waymo.utils:
Reading raw data
INFO:scenarionet.converter.waymo.utils:Use one worker, as num_scenario < num_workers:
0%| | 0/2 [00:00<?, ?it/s]
Traceback (most recent call last):
File "/root/anaconda3/envs/scenarionet/lib/python3.8/runpy.py", line 192, in _run_module_as_main
return _run_code(code, main_globals, None,
File "/root/anaconda3/envs/scenarionet/lib/python3.8/runpy.py", line 85, in _run_code
exec(code, run_globals)
File "/root/aliyun/scenarionet/scenarionet/convert_waymo.py", line 64, in
scenarios = get_waymo_scenarios(
File "/root/aliyun/scenarionet/scenarionet/converter/waymo/utils.py", line 456, in get_waymo_scenarios
all_result = read_from_files([waymo_data_directory, file_list])
File "/root/aliyun/scenarionet/scenarionet/converter/waymo/utils.py", line 481, in read_from_files
scenario.ParseFromString(data)
google.protobuf.message.DecodeError: Error parsing message with type 'waymo.open_dataset.Scenario'

Now, when I switch to the path to ./scenarionet and run 'pip install -e .' again, I receive the following errors.
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
tensorflow 2.11.0 requires protobuf<3.20,>=3.9.2, but you have protobuf 3.20.3 which is incompatible.
waymo-open-dataset-tf-2-11-0 1.5.0 requires matplotlib==3.6.1, but you have matplotlib 3.7.1 which is incompatible.
waymo-open-dataset-tf-2-11-0 1.5.0 requires numpy==1.21.5, but you have numpy 1.24.2 which is incompatible.

As a result, I am able to determine that my problem is caused by the incompatibility between the versions of third-party libraries. 
Could you share a setting that ensures codes run properly, including the Python version number and the version numbers of relevant third-party libraries?

QuanyiLi, 2023-07-02:
Could you try python 3.9? The rest steps remain the same. This is still a bug from Waymo package and protobuf. Changing to py3.9 allows you to install tf 2.11 and protobuf 3.20

发自我的iPhone

 

 
 
 
 
 
 
 tensorflow 2.11.0 requires protobuf<3.20,&gt;=3.9.2, but you have protobuf 3.20.3 which is incompatible.

c4cld, 2023-07-03:
@QuanyiLi Thank you for your help.
I have tried python 3.9 and the rest steps remain the same. However, incompatibility issues still remain. When I converted tfrecord, same error messages occur. 
So I checked versions of third-party libraries, and found that the version of protobuf is 3.19.6 rather than 3.20.
When I tried to install protobuf 3.20.0, following warnings occurs:
Looking in indexes: http://mirrors.cloud.aliyuncs.com/pypi/simple/
Collecting protobuf==3.20
  Downloading http://mirrors.cloud.aliyuncs.com/pypi/packages/7f/d9/6b9e97c6498a29c5e99badce383a8711c4f0ff586a464851b3f8b06cc66d/protobuf-3.20.0-cp39-cp39-manylinux_2_5_x86_64.manylinux1_x86_64.whl (1.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.0/1.0 MB 24.0 MB/s eta 0:00:00
Installing collected packages: protobuf
  Attempting uninstall: protobuf
    Found existing installation: protobuf 3.19.6
    Uninstalling protobuf-3.19.6:
      Successfully uninstalled protobuf-3.19.6
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
googleapis-common-protos 1.59.1 requires **protobuf!=3.20.0,!**=3.20.1,!=4.21.1,!=4.21.2,!=4.21.3,!=4.21.4,!=4.21.5,<5.0.0.dev0,>=3.19.5, but you have protobuf 3.20.0 which is incompatible.
tensorflow 2.11.0 requires **protobuf<3.20,>=3.9.2**, but you have protobuf 3.20.0 which is incompatible.
metadrive-simulator 0.3.0.1 requires numpy<=1.24.2,>=1.21.6, but you have numpy 1.21.5 which is incompatible.
metadrive-simulator 0.3.0.1 requires protobuf==3.20.3, but you have protobuf 3.20.0 which is incompatible.
Successfully installed protobuf-3.20.0

What's more, how to choose the version of numpy is another issue. In the above warnings, metadrive-simulator 0.3.0.1 requires numpy<=1.24.2,>=1.21.6. When I tried to install numpy 1.21.6, the following warnings occurs:
Looking in indexes: http://mirrors.cloud.aliyuncs.com/pypi/simple/
Collecting numpy==1.21.6
  Downloading http://mirrors.cloud.aliyuncs.com/pypi/packages/e7/f2/0bdcf2c40ef144cbbc9e0947eea831a145a98b0e5f8438fc09cf7fda0b35/numpy-1.21.6-cp39-cp39-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (15.7 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 15.7/15.7 MB 83.5 MB/s eta 0:00:00
Installing collected packages: numpy
  Attempting uninstall: numpy
    Found existing installation: numpy 1.21.5
    Uninstalling numpy-1.21.5:
      Successfully uninstalled numpy-1.21.5
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
seaborn 0.12.2 requires matplotlib!=3.6.1,>=3.1, but you have matplotlib 3.6.1 which is incompatible.
tensorflow 2.11.0 requires protobuf<3.20,>=3.9.2, but you have protobuf 3.20.0 which is incompatible.
waymo-open-dataset-tf-2-11-0 1.5.0 requires numpy==1.21.5, but you have numpy 1.21.6 which is incompatible.
metadrive-simulator 0.3.0.1 requires protobuf==3.20.3, but you have protobuf 3.20.0 which is incompatible.
Successfully installed numpy-1.21.6
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv

waymo-open-dataset-tf-2-11-0 1.5.0 requires numpy==1.21.5

How to address these issues? Thank you for your help in advance ^_^

QuanyiLi, 2023-07-03:
That's strange. By using ```conda list```, my env shows this:
```
_libgcc_mutex             0.1                        main  
_openmp_mutex             5.1                       1_gnu  
absl-py                   1.4.0                    pypi_0    pypi
affine                    2.4.0                    pypi_0    pypi
aioboto3                  11.0.1                   pypi_0    pypi
aiobotocore               2.4.2                    pypi_0    pypi
aiofiles                  22.1.0                   pypi_0    pypi
aiohttp                   3.8.4                    pypi_0    pypi
aioitertools              0.11.0                   pypi_0    pypi
aiosignal                 1.3.1                    pypi_0    pypi
aiosqlite                 0.19.0                   pypi_0    pypi
antlr4-python3-runtime    4.8                      pypi_0    pypi
anyio                     3.6.2                    pypi_0    pypi
appdirs                   1.4.4                    pypi_0    pypi
argon2-cffi               21.3.0                   pypi_0    pypi
argon2-cffi-bindings      21.2.0                   pypi_0    pypi
array-record              0.2.0                    pypi_0    pypi
arrow                     1.2.3                    pypi_0    pypi
asttokens                 2.2.1                    pypi_0    pypi
astunparse                1.6.3                    pypi_0    pypi
async-generator           1.10                     pypi_0    pypi
async-timeout             4.0.2                    pypi_0    pypi
attrs                     23.1.0                   pypi_0    pypi
auditwheel                5.3.0                    pypi_0    pypi
babel                     2.12.1                   pypi_0    pypi
backcall                  0.2.0                    pypi_0    pypi
beautifulsoup4            4.12.2                   pypi_0    pypi
bleach                    6.0.0                    pypi_0    pypi
bokeh                     2.4.3                    pypi_0    pypi
boto3                     1.24.59                  pypi_0    pypi
botocore                  1.27.59                  pypi_0    pypi
ca-certificates           2023.01.10           h06a4308_0  
cachetools                5.3.0                    pypi_0    pypi
casadi                    3.6.2                    pypi_0    pypi
certifi                   2022.12.7                pypi_0    pypi
cffi                      1.15.1                   pypi_0    pypi
cfgv                      3.3.1                    pypi_0    pypi
charset-normalizer        3.1.0                    pypi_0    pypi
click                     8.1.3                    pypi_0    pypi
click-plugins             1.1.1                    pypi_0    pypi
cligj                     0.7.2                    pypi_0    pypi
cloudpickle               2.2.1                    pypi_0    pypi
cmake                     3.26.3                   pypi_0    pypi
comm                      0.1.3                    pypi_0    pypi
contourpy                 1.0.7                    pypi_0    pypi
control                   0.9.1                    pypi_0    pypi
coverage                  7.2.5                    pypi_0    pypi
cryptography              40.0.2                   pypi_0    pypi
cycler                    0.11.0                   pypi_0    pypi
dask                      2023.3.1                 pypi_0    pypi
debugpy                   1.6.7                    pypi_0    pypi
decorator                 5.1.1                    pypi_0    pypi
defusedxml                0.7.1                    pypi_0    pypi
descartes                 1.1.0                    pypi_0    pypi
distlib                   0.3.6                    pypi_0    pypi
dm-tree                   0.1.8                    pypi_0    pypi
docker                    6.1.0                    pypi_0    pypi
docker-pycreds            0.4.0                    pypi_0    pypi
einsum                    0.3.0                    pypi_0    pypi
etils                     1.2.0                    pypi_0    pypi
exceptiongroup            1.1.1                    pypi_0    pypi
executing                 1.2.0                    pypi_0    pypi
fastjsonschema            2.16.3                   pypi_0    pypi
filelock                  3.12.0                   pypi_0    pypi
fiona                     1.9.3                    pypi_0    pypi
fire                      0.5.0                    pypi_0    pypi
flatbuffers               23.3.3                   pypi_0    pypi
fonttools                 4.39.3                   pypi_0    pypi
[... 246 lines cut ...]
webcolors                 1.13                     pypi_0    pypi
webencodings              0.5.1                    pypi_0    pypi
websocket-client          1.5.1                    pypi_0    pypi
werkzeug                  2.3.3                    pypi_0    pypi
wheel                     0.38.4           py39h06a4308_0  
widgetsnbextension        4.0.7                    pypi_0    pypi
wrapt                     1.15.0                   pypi_0    pypi
wsproto                   1.2.0                    pypi_0    pypi
xmltodict                 0.13.0                   pypi_0    pypi
xz                        5.4.2                h5eee18b_0  
y-py                      0.5.9                    pypi_0    pypi
yapf                      0.30.0                   pypi_0    pypi
yarl                      1.9.2                    pypi_0    pypi
ypy-websocket             0.8.2                    pypi_0    pypi
[... cut ...]
zlib                      1.2.13               h5eee18b_0  
```
My env creation follows these steps:
1. ```conda create -n scenarionet python=3.9```
2. git clone latest metadrive, and ```cd metadrive``` then run ```pip install -e .```
3. git clone scenarionet, and ```cd scenarionet``` then run ```pip install -e .```

I also tested the convert. I use files downloaded here: waymo_open_dataset_motion_v_1_2_0/uncompressed/scenario/training_20s

Everything works well. I guess maybe you don't use 20s version Waymo data, which causes this problem

QuanyiLi, 2023-07-03:
by the way, the Error of pip can be viewed as a warning actually. The pkgs you required are still installed successfully but may cause some conflicts. I usually ignore this error as conflicts rarely happen in my experience.

发自我的iPhone

 

 googleapis-common-protos 1.59.1 requires protobuf!=3.20.0,!=3.20.1,!=4.21.1,!=4.21.2,!=4.21.3,!=4.21.4,!=4.21.5,<5.0.0.dev0,&gt;=3.19.5, but you have protobuf 3.20.0 which is incompatible.
 tensorflow 2.11.0 requires protobuf<3.20,&gt;=3.9.2, but you have protobuf 3.20.0 which is incompatible.
 metadrive-simulator 0.3.0.1 requires numpy<=1.24.2,&gt;=1.21.6, but you have numpy 1.21.5 which is incompatible.
 
What's more, how to choose the version of numpy is another issue. In the above warnings, metadrive-simulator 0.3.0.1 requires numpy<=1.24.2,&gt;=1.21.6. When I tried to install numpy 1.21.6, the following warnings occurs:
 seaborn 0.12.2 requires matplotlib!=3.6.1,&gt;=3.1, but you have matplotlib 3.6.1 which is incompatible.
 tensorflow 2.11.0 requires protobuf<3.20,&gt;=3.9.2, but you have protobuf 3.20.0 which is incompatible.

c4cld, 2023-07-04:
@QuanyiLi Thank you very much.  "google.protobuf.message.DecodeError: Error parsing message with type 'waymo.open_dataset.Scenario'" issue is really caused by not using the data of version 20s. 

When I use 20s version Waymo data, another issue about data occurs:
(scenarionet) root@iZ8vb2jw1bhl7u7wkujvc4Z:~/aliyun/scenarionet# python -m scenarionet.convert_waymo -d waymo --raw_data_path /root/aliyun/scenarionet/tfrecord  --num_workers=16
/root/anaconda3/envs/scenarionet/lib/python3.9/site-packages/gym/envs/registration.py:440: UserWarning: WARN: The `registry.env_specs` property along with `EnvSpecTree` is deprecated. Please use `registry` directly as a dictionary instead.
  logger.warn(
Successfully registered the following environments: ['MetaDrive-validation-v0', 'MetaDrive-10env-v0', 'MetaDrive-100envs-v0', 'MetaDrive-1000envs-v0', 'SafeMetaDrive-validation-v0', 'SafeMetaDrive-10env-v0', 'SafeMetaDrive-100envs-v0', 'SafeMetaDrive-1000envs-v0', 'MARLTollgate-v0', 'MARLBottleneck-v0', 'MARLRoundabout-v0', 'MARLIntersection-v0', 'MARLParkingLot-v0', 'MARLMetaDrive-v0'].
INFO:scenarionet.converter.waymo.utils:
 Reading raw data
INFO:scenarionet.converter.waymo.utils:Use one worker, as num_scenario < num_workers:
100%|█████████████████████████████████████████████| 1/1 [00:01<00:00,  1.33s/it]
INFO:scenarionet.converter.waymo.utils:
 Find 61 waymo scenarios from 1 files
multiprocessing.pool.RemoteTraceback:
"""
Traceback (most recent call last):
  File "/root/anaconda3/envs/scenarionet/lib/python3.9/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/root/aliyun/scenarionet/scenarionet/converter/utils.py", line 131, in writing_to_directory_wrapper
    return write_to_directory_single_worker(
  File "/root/aliyun/scenarionet/scenarionet/converter/utils.py", line 179, in write_to_directory_single_worker
    summary_file = SD.DATASET.SUMMARY_FILE
AttributeError: type object 'ScenarioDescription' has no attribute 'DATASET'
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/root/anaconda3/envs/scenarionet/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/root/anaconda3/envs/scenarionet/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/root/aliyun/scenarionet/scenarionet/convert_waymo.py", line 68, in <module>
    write_to_directory(
  File "/root/aliyun/scenarionet/scenarionet/converter/utils.py", line 125, in write_to_directory
    ret = list(p.imap(func, argument_list))
  File "/root/anaconda3/envs/scenarionet/lib/python3.9/multiprocessing/pool.py", line 870, in next
    raise value
AttributeError: type object 'ScenarioDescription' has no attribute 'DATASET'

Additionally, I created another virtual environment which include third-party libraries with same version numbers in your last reply. And when I convert, same errors occur.

How to solve this issue?  If the issue is difficult to solve, would you be able to share the converted Waymo scenario files with me? I would like to use these scenario files to run the train_waymo program.

c4cld, 2023-07-05:
@QuanyiLi After updating the metadrive in my env, I have successfully converted waymo tfrecord ^_^
Now I start training ppo agents in waymo scenarios, but wandb library can not be imported, the error message is as follows:
```Traceback (most recent call last):
  File "/root/anaconda3/envs/scenarionet/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/root/anaconda3/envs/scenarionet/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/root/aliyun/scenarionet-20230705/scenarionet_training/scripts/train_waymo.py", line 81, in <module>
    train(
  File "/root/aliyun/scenarionet-20230705/scenarionet_training/train_utils/utils.py", line 140, in train
    from scenarionet_training.wandb_utils.our_wandb_callbacks_ray100 import OurWandbLogger
  File "/root/aliyun/scenarionet-20230705/scenarionet_training/wandb_utils/our_wandb_callbacks_ray100.py", line 3, in <module>
    from ray.tune.integration.wandb import WandbLogger, _clean_log, _set_api_key
ImportError: cannot import name 'WandbLogger' from 'ray.tune.integration.wandb' (/root/anaconda3/envs/scenarionet/lib/python3.9/site-packages/ray/tune/integration/wandb.py)

I have tried to change versions of related libraries (such as wandb and ray[rllib]), but the issue is not solved. So I have asked this question in the github websites of wandb and ray.

QuanyiLi, 2023-07-05:
Good. For training, you can install ScenarioNet via ```pip install -e .[train]```. Then all dependencies can be installed automatically. **However, you need to downgrade the Python version to 3.7 or 3.8 to install ```ray==1.0.0```.** We are working on upgrading the ray dependency to the latest version, but it isn't ready yet.

c4cld, 2023-07-06:
@QuanyiLi Thank you for your valuable suggestions!

Could you share your setting that ensures training codes run properly, including the Python version number and the version numbers of relevant third-party libraries?

I have encountered some issues caused by version incompatibility. For example, an error message is as follows.

wandb: Currently logged in as: openrobot (use `wandb login --relogin` to force relogin)
Successfully registered the following environments: ['MetaDrive-validation-v0', 'MetaDrive-10env-v0', 'MetaDrive-100envs-v0', 'MetaDrive-1000envs-v0', 'SafeMetaDrive-validation-v0', 'SafeMetaDrive-10env-v0', 'SafeMetaDrive-100envs-v0', 'SafeMetaDrive-1000envs-v0', 'MARLTollgate-v0', 'MARLBottleneck-v0', 'MARLRoundabout-v0', 'MARLIntersection-v0', 'MARLParkingLot-v0', 'MARLMetaDrive-v0'].
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/root/anaconda3/envs/scenarionet-train/lib/python3.8/multiprocessing/spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "/root/anaconda3/envs/scenarionet-train/lib/python3.8/multiprocessing/spawn.py", line 125, in _main
    prepare(preparation_data)
  File "/root/anaconda3/envs/scenarionet-train/lib/python3.8/multiprocessing/spawn.py", line 234, in prepare
    _fixup_main_from_name(data['init_main_from_name'])
  File "/root/anaconda3/envs/scenarionet-train/lib/python3.8/multiprocessing/spawn.py", line 258, in _fixup_main_from_name
    main_content = runpy.run_module(mod_name,
  File "/root/anaconda3/envs/scenarionet-train/lib/python3.8/runpy.py", line 207, in run_module
    return _run_module_code(code, init_globals, run_name, mod_spec)
  File "/root/anaconda3/envs/scenarionet-train/lib/python3.8/runpy.py", line 97, in _run_module_code
    _run_code(code, mod_globals, init_globals,
  File "/root/anaconda3/envs/scenarionet-train/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/root/aliyun/scenarionet/scenarionet_training/scripts/train_waymo.py", line 5, in <module>
    from scenarionet_training.train_utils.multi_worker_PPO import MultiWorkerPPO
  File "/root/aliyun/scenarionet/scenarionet_training/train_utils/multi_worker_PPO.py", line 4, in <module>
    from ray.rllib.agents.ppo.ppo import PPOTrainer
  File "/root/anaconda3/envs/scenarionet-train/lib/python3.8/site-packages/ray/__init__.py", line 14, in <module>
    version_info = pkg_resources.require("pickle5")
  File "/root/anaconda3/envs/scenarionet-train/lib/python3.8/site-packages/pkg_resources/__init__.py", line 966, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/root/anaconda3/envs/scenarionet-train/lib/python3.8/site-packages/pkg_resources/__init__.py", line 827, in resolve
    dist = self._resolve_dist(
  File "/root/anaconda3/envs/scenarionet-train/lib/python3.8/site-packages/pkg_resources/__init__.py", line 868, in _resolve_dist
    raise DistributionNotFound(req, requirers)
**pkg_resources.DistributionNotFound: The 'pickle5' distribution was not found and is required by the application**

I have search some Q&As related to this issue (such as https://github.com/ray-project/ray/issues/10637). These Q&As confirm that this issue is caused by pickle5's version incompatibility.

QuanyiLi, 2023-07-06:
Generally, running ```pip install -e .``` and  ```pip install -e .[train]``` in metadrive and scenarionet repo respectively can satisfy all requirements automatically. This is verified by other users, see: https://github.com/metadriverse/metadrive/pull/445#issuecomment-1589848384

If it is still too sticky to solve, I would recommend you use another training framework and only use ScenarioNet as a data conversion tool.

c4cld, 2023-07-09:
@QuanyiLi Thank you for your help!
I have solved all issues related to version incompatibility by carefully adjusting the versions of third-party libraries. Currently I am training PPO models by running the train_waymo.py.

QuanyiLi, 2023-07-09:
Great! You can include more scenarios in the training dataset. Besides, you could set up the Wandb for monitoring the training process.

closed 2023-08-21 by QuanyiLi