metadriverse/metadrive #465: acceleration and steering before normalization

liuxy10, 2023-07-11:
Hi, 

In the [doc](https://metadrive-simulator.readthedocs.io/en/latest/action_and_dynamics.html), it is mentioned that MetaDrive receives normalized action as input to control each target vehicle. Could you provide a target acceleration and steering angle range before the normalization (i.e. the maximum acceleration and maximum steering angle input)?

pengzhenghao, 2023-07-11:
The steering range is determined here: https://github.com/metadriverse/metadrive/blob/2db10d3eea092fdea0d134ccff8d2203f47b6a0a/metadrive/component/pg_space.py#L237

and it is default to +-40 degree as the RL agent is controlling the BaseVehicle. That is, if the input is +1, then we will assume current steering angle is +40 degree and vice versa.

Similarly, the throttle control how many engine force we apply, and the config name is `max_engine_force`. We will randomly sample from `BoxSpace(750, 850)`. That is when applying acceleration = +1 (the second dim of the action) we will apply engine force = 800 (this value might vary).

Here is the definition of the config:
https://github.com/metadriverse/metadrive/blob/2db10d3eea092fdea0d134ccff8d2203f47b6a0a/metadrive/component/pg_space.py#L235

You can easily change the config by setting:

```
env_config = {"vehicle_config": {"max_steering": 60, "max_engine_force": 1000}}
env(env_config)
```

Last, for your information, here is the function we translate the input "normalized action" to a real control signal of vehicle:

https://github.com/metadriverse/metadrive/blob/2db10d3eea092fdea0d134ccff8d2203f47b6a0a/metadrive/component/vehicle/base_vehicle.py#L524C12-L524C12

jinning-li, 2023-07-11:
Thank you very much for the detailed explanation! I still have several further questions:
1. I was wondering what is the unit of `steering` [here](https://github.com/metadriverse/metadrive/blob/2db10d3eea092fdea0d134ccff8d2203f47b6a0a/metadrive/component/vehicle/base_vehicle.py#L524C12-L524C12)? As I understand from your answer above, it should be degree. But in the `IDMPolicy` [here](https://github.com/metadriverse/metadrive/blob/2db10d3eea092fdea0d134ccff8d2203f47b6a0a/metadrive/policy/idm_policy.py#L299C1-L300C1), it seems the unit of steering is radius?

2. Also for the second dim of action (which I assume is acceleration), is the unit $m/s^2$? 

Actually, we would like to train a behavior cloning policy from waymo dataset, where we'd like to define [heading angle, acceleration] as the output of the BC policy. I believe we have to preprocess these actions before we input them to the simulator. We plan to use a PID controller to obtain the steering angle from the heading angle, just as the IDMPolicy. But for the acceleration, we can only obtain the actual acceleration from the dataset. So is there any suggested way to normalize it to [-1, 1]? Or should we normalize it at all?

pengzhenghao, 2023-07-11:
> I was wondering what is the unit of steering [...]

The input to the PID controller is in degree but the output of the PID controller has no unit. 

> Also for the second dim of action (which I assume is acceleration), is the unit 

The unit is engine force, whose unit is Newton `N`.

> Actually, we would like to train a behavior cloning policy from waymo dataset, where we'd like to define [heading angle, acceleration] as the output of the BC policy. I believe we have to preprocess these actions before we input them to the simulator. We plan to use a PID controller to obtain the steering angle from the heading angle, just as the IDMPolicy. But for the acceleration, we can only obtain the actual acceleration from the dataset. So is there any suggested way to normalize it to [-1, 1]? Or should we normalize it at all because I noticed that the [IDMPolicy](https://github.com/metadriverse/metadrive/blob/2db10d3eea092fdea0d134ccff8d2203f47b6a0a/metadrive/policy/idm_policy.py#L264-L265) didn't do any normalization?

In waymo dataset the native data is velocity in `m/s`. You don't have to translate it to thrust as thrust is the 2nd-order derivative of velocity. I think you can follow IDM policy and create a velocity PID to control it. I believe we already provided that in somewhere.

Again the PID controller's output is a control signal which might not have clear unit there. Though you can control it somewhat by setting the `P` parameters.

closed 2023-08-22 by QuanyiLi

yangsl-njfu, 2025-06-18:
> The steering range is determined here:
> 
> [metadrive/metadrive/component/pg_space.py](https://github.com/metadriverse/metadrive/blob/2db10d3eea092fdea0d134ccff8d2203f47b6a0a/metadrive/component/pg_space.py#L237)
> 
> Line 237 in [2db10d3](/metadriverse/metadrive/commit/2db10d3eea092fdea0d134ccff8d2203f47b6a0a)
> 
>  max_steering=ConstantSpace(40), 
> and it is default to +-40 degree as the RL agent is controlling the BaseVehicle. That is, if the input is +1, then we will assume current steering angle is +40 degree and vice versa.
> 
> Similarly, the throttle control how many engine force we apply, and the config name is `max_engine_force`. We will randomly sample from `BoxSpace(750, 850)`. That is when applying acceleration = +1 (the second dim of the action) we will apply engine force = 800 (this value might vary).
> 
> Here is the definition of the config:
> 
> [metadrive/metadrive/component/pg_space.py](https://github.com/metadriverse/metadrive/blob/2db10d3eea092fdea0d134ccff8d2203f47b6a0a/metadrive/component/pg_space.py#L235)
> 
> Line 235 in [2db10d3](/metadriverse/metadrive/commit/2db10d3eea092fdea0d134ccff8d2203f47b6a0a)
> 
>  max_engine_force=BoxSpace(750, 850), 
> You can easily change the config by setting:
> 
> ```
> env_config = {"vehicle_config": {"max_steering": 60, "max_engine_force": 1000}}
> env(env_config)
> ```
> 
> Last, for your information, here is the function we translate the input "normalized action" to a real control signal of vehicle:
> 
> https://github.com/metadriverse/metadrive/blob/2db10d3eea092fdea0d134ccff8d2203f47b6a0a/metadrive/component/vehicle/base_vehicle.py#L524C12-L524C12

How to get the mapping of default_vehicle to the action[steering,acceleration] range in the real environment? Besides, I see that the braking engine force is smaller than the acceleration engine force. Why is this?
 Also, what are the specific acceleration values mapped to this engine force?

BASE_VEHICLE = dict( wheel_friction=ConstantSpace(0.9), max_engine_force=BoxSpace(750, 850), max_brake_force=BoxSpace(80, 180), max_steering=ConstantSpace(40), max_speed_km_h=ConstantSpace(80), ) DEFAULT_VEHICLE = BASE_VEHICLE