metadriverse/metadrive #466: Get the lidar Observation and Image Observation at the same time

Sue0316, 2023-07-13:
Hi, I would like to ask is there any argument that I can adjust to get both of the image observation and lidar observation at the same time? Or is there any way to do so? Thank you so much! BTW. I'm using version 0.2.3

QuanyiLi, 2023-07-13:
Actually, if the ```image_observation``` in your env config is set to ```True```, you are allowed to access the camera and lidar together by```env.vehicle.lidar.perceive(env.vehicle)``` and ```env.vehicle.image_sensor["rgb_camera"].get_pixels_array(env.vehicle, self.rgb_clip)```. If you wanna get the results of both sensors in the returned value of ```env.step()```, you can just write your own observation class by merging ```LidarStateObservation``` and ```ImageStateObservation```.

Sue0316, 2023-07-13:
I see! Got it, Thank you so much for the answer. Really appreciate it.

closed 2023-08-21 by QuanyiLi