metadriverse/metadrive #469: [Discussion] Env reset seems to be much slower than previous versions

jimmydengpeng, 2023-07-16:
I timed `env.reset()` method on my M1 chip MacBook Pro, and it took almost 10 seconds, that is much slower than before. I tracked down deep into the code calling stack and found it was caused by the line in `BaseEngine.warmup():line672`:
```python
Pedestrian.init_pedestrian_model()
```
I stopped digging deeper and just comment out that line, and everything works fine as usual since I don't need any pedestrians 😂

I don't know the exact reason for this or it just happens on macOS and won't be slower in other platforms. 

Hope it will be fixed in future 🙏 🌹

QuanyiLi, 2023-07-16:
Thanks for letting me know. 🌹

This preload process is for avoiding the jitter when a human model is created for the first time. It does take some time to load the model, so I preload it to memory when the simulation is initialized. This preload is enabled by default. You can turn off this by setting ```preload_models=False``` in ```env_config```. Actually, it doesn't take too much time on Windows and Linux. But given your feedback, it would be better to disable the preload function or determine when to do the preload more wisely.

jimmydengpeng, 2023-07-17:
Thanks for the suggestion, it works🙏!

closed 2023-07-17 by jimmydengpeng