metadriverse/metadrive #473: rgb_camera observation shape

WeizhenWang-1210, 2023-07-19:
It seems like the observation is of shape ` {'image': (84, 84, 3, 3), 'state': (21,)}`. Why is there an extra dimension in image?

QuanyiLi, 2023-07-19:
We stack the latest 3 images to inform temporal information. This parameter can be modified in env_config, I think. You can check the ```class ImageStateObservation``` to know more about it

pengzhenghao, 2023-07-19:
It shouldn’t be a (3,3) at the end. If we stack 3 frames, we either grayscale the image and use channel=3, or we stack all channels to have 9 channels.Best,Zhenghao

QuanyiLi, 2023-07-19:
The default image observation is set to RGB now. I believe one can build the observation on their own. It doesn't matter.

pengzhenghao, 2023-07-19:
Yes I know it’s customizable but a default one have shape (3,3) is a bug I guess (not sure how Weizheng has this)Best,Zhenghao

QuanyiLi, 2023-07-19:
Actually, I made this modification before...

closed 2023-07-24 by WeizhenWang-1210