metadriverse/metadrive #477: Errors in running python -m metadrive.examples.train_generalization_experiment

Liuxueyi, 2023-07-26:
I fall into trouble with the following error.txt, what's the problem?Please.
Failure # 1 (occurred at 2023-07-26_16-42-25)
[36mray::PPO.train()[39m (pid=1477765, ip=10.1.20.90, repr=PPO)
  File "/home/oem/anaconda3/envs/meta/lib/python3.9/site-packages/ray/tune/trainable/trainable.py", line 384, in train
    raise skipped from exception_cause(skipped)
  File "/home/oem/anaconda3/envs/meta/lib/python3.9/site-packages/ray/tune/trainable/trainable.py", line 381, in train
    result = self.step()
  File "/home/oem/anaconda3/envs/meta/lib/python3.9/site-packages/ray/rllib/algorithms/algorithm.py", line 792, in step
    results, train_iter_ctx = self._run_one_training_iteration()
  File "/home/oem/anaconda3/envs/meta/lib/python3.9/site-packages/ray/rllib/algorithms/algorithm.py", line 2811, in _run_one_training_iteration
    results = self.training_step()
  File "/home/oem/anaconda3/envs/meta/lib/python3.9/site-packages/ray/rllib/algorithms/ppo/ppo.py", line 403, in training_step
    train_batch = synchronous_parallel_sample(
  File "/home/oem/anaconda3/envs/meta/lib/python3.9/site-packages/ray/rllib/execution/rollout_ops.py", line 85, in synchronous_parallel_sample
    sample_batches = worker_set.foreach_worker(
  File "/home/oem/anaconda3/envs/meta/lib/python3.9/site-packages/ray/rllib/evaluation/worker_set.py", line 722, in foreach_worker
    handle_remote_call_result_errors(remote_results, self._ignore_worker_failures)
  File "/home/oem/anaconda3/envs/meta/lib/python3.9/site-packages/ray/rllib/evaluation/worker_set.py", line 75, in handle_remote_call_result_errors
    raise r.get()
ray.exceptions.RayTaskError(ValueError): [36mray::RolloutWorker.apply()[39m (pid=1478256, ip=10.1.20.90, repr=<ray.rllib.evaluation.rollout_worker.RolloutWorker object at 0x7fd34b626b80>)
ValueError: The two structures don't have the same nested structure.

First structure: type=tuple str=(array([0.29166666, 0.29166666, 0.5       , 0.01234568, 0.5       ,
       0.5       , 0.5       , 0.        , 0.5       , 0.55      ,
       0.5       , 0.        , 0.5       , 0.5       , 0.95      ,
       0.5       , 0.        , 0.5       , 0.5       , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
[... 217 lines cut ...]
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        ], dtype=float32), {'overtake_vehicle_num': 0, 'velocity': 0.0, 'steering': 0.0, 'acceleration': 0.0, 'step_energy': 0.0, 'episode_energy': 0.0, 'policy': 'EnvInputPolicy', 'crash_vehicle': False, 'crash_object': False, 'crash_building': False, 'out_of_road': False, 'arrive_dest': False, 'max_step': False, 'env_seed': 5000, 'crash': False, 'step_reward': 0.0, 'cost': 0})

Second structure: type=ndarray str=[0.11674915 0.30998474 0.6659199  0.6621079  0.34599248 0.71242267
 0.30595112 0.9044908  0.05226379 0.59416175 0.8165471  0.17465371

QuanyiLi, 2023-07-27:
Hi,

I guess it is due to the upgrade from ```gym``` to ```gymnasium```.

Could you try ```from metadrive.env.gym_wrapper import GymEnvWrapper``` and do the following modification to the training config?

```
 env=GymEnvWrapper,
        env_config={
            "inner_class": MetaDriveEnv,
            "inner_config": dict(
            num_scenarios=tune.grid_search([1, 3, 5, 1000]),
            start_seed=tune.grid_search([5000]),
            random_traffic=False,
            traffic_density=tune.grid_search([0.1, 0.3])
        )
        }
```

Liuxueyi, 2023-07-28:
Thanks for your reply! I have run the PPO experiments successfully using the Stable-Baselines3 with the newest MetaDrive version. I think there may be some problems with the version of rllib. By the way, when I use the env "MetaDrive-10env-v0" and set the config "use_render" to true, the map index is always 0,  is there any settings that cause this situation?

QuanyiLi, 2023-07-31:
The number of maps in each environment is controlled by ```num_environments```, which is a field in env_config. Could you share your training config with me?

Liuxueyi, 2023-07-31:
In addition, my training script is :
[image]

QuanyiLi, 2023-08-01:
I replicate your error successfully. This is because you pass a new config dict ```{"use_render": True}```, which overwrites the original config where the ```num_scenarios``` is set to 1000. 

The overwrite of ```gym.make``` doesn't happen recursively, so you can ```{"use_render": True, "num_scenarios": 1000}``` instead.

Liuxueyi, 2023-08-02:
It works! Thanks for your guidance！

closed 2023-08-04 by QuanyiLi