metadriverse/metadrive #479: [Question]: How to disable terminal message

Gaiejj, 2023-08-04:
Thanks to your contribution to this meaningful safe reinforcement learning tasks. I am a safe reinforcement learning researcher and I want to run some safe reinforcement learning algorithms on MetaDrive.
I try to make the environment following the instruction in your [document](https://metadrive-simulator.readthedocs.io/en/latest/rl_environments.html)
```python
from metadrive import SafeMetaDriveEnv

env=SafeMetaDriveEnv()
```
then I run ``PPOLag`` but find that lots of information comes to my screen, like
```bash
INFO:/home/metadrive/metadrive/metadrive/envs/base_env.py:Episode ended! Index: 76 Reason: out_of_road.
INFO:/home/metadrive/metadrive/metadrive/envs/base_env.py:Episode ended! Index: 76 Reason: out_of_road.
INFO:/home/metadrive/metadrive/metadrive/envs/base_env.py:Episode ended! Index: 76 Reason: out_of_road.
INFO:/home/metadrive/metadrive/metadrive/envs/base_env.py:Episode ended! Index: 76 Reason: out_of_road.
INFO:/home/metadrive/metadrive/metadrive/envs/base_env.py:Episode ended! Index: 76 Reason: out_of_road.
```
Those information is a little troublesome and I check the ``metadrive/envs/metadrive_env`` file. I notice all of them come form ``done_function`` in line 157
```python
    def done_function(self, vehicle_id: str):
        vehicle = self.vehicles[vehicle_id]
        done = False
        done_info = {
            TerminationState.CRASH_VEHICLE: False,
            TerminationState.CRASH_OBJECT: False,
            TerminationState.CRASH_BUILDING: False,
            TerminationState.OUT_OF_ROAD: False,
            TerminationState.SUCCESS: False,
            TerminationState.MAX_STEP: False,
            # TerminationState.CURRENT_BLOCK: self.vehicle.navigation.current_road.block_ID(),
            TerminationState.ENV_SEED: self.current_seed,
            # crash_vehicle=False, crash_object=False, crash_building=False, out_of_road=False, arrive_dest=False,
        }
        if self._is_arrive_destination(vehicle):
            done = True
            self.logger.info("Episode ended! Index: {} Reason: arrive_dest.".format(self.current_seed))
            done_info[TerminationState.SUCCESS] = True
        if self._is_out_of_road(vehicle):
            done = True
            self.logger.info("Episode ended! Index: {} Reason: out_of_road.".format(self.current_seed))
            done_info[TerminationState.OUT_OF_ROAD] = True
        if vehicle.crash_vehicle and self.config["crash_vehicle_done"]:
            done = True
            self.logger.info("Episode ended! Index: {} Reason: crash vehicle ".format(self.current_seed))
            done_info[TerminationState.CRASH_VEHICLE] = True
        if vehicle.crash_object and self.config["crash_object_done"]:
            done = True
            done_info[TerminationState.CRASH_OBJECT] = True
            self.logger.info("Episode ended! Index: {} Reason: crash object ".format(self.current_seed))
        if vehicle.crash_building:
            done = True
            done_info[TerminationState.CRASH_BUILDING] = True
            self.logger.info("Episode ended! Index: {} Reason: crash building ".format(self.current_seed))
        if self.config["max_step_per_agent"] is not None and \
                self.episode_lengths[vehicle_id] >= self.config["max_step_per_agent"]:
            done = True
            done_info[TerminationState.MAX_STEP] = True
            self.logger.info("Episode ended! Index: {} Reason: max step ".format(self.current_seed))

        if self.config["horizon"] is not None and \
                self.episode_lengths[vehicle_id] >= self.config["horizon"] and not self.is_multi_agent:
            # single agent horizon has the same meaning as max_step_per_agent
            done = True
            done_info[TerminationState.MAX_STEP] = True
            self.logger.info("Episode ended! Index: {} Reason: max step ".format(self.current_seed))

        # for compatibility
        # crash almost equals to crashing with vehicles
        done_info[TerminationState.CRASH] = (
            done_info[TerminationState.CRASH_VEHICLE] or done_info[TerminationState.CRASH_OBJECT]
            or done_info[TerminationState.CRASH_BUILDING]
        )
        return done, done_info
```

Is there any solution to prevent them logging in to terminal?

QuanyiLi, 2023-08-04:
Yes. set ```env.logger.setLevel(logging.FATAL)```

Gaiejj, 2023-08-04:
It worked for me, thanks a lot !

closed 2023-08-04 by Gaiejj