metadriverse/metadrive #496: How to test an RL policy after training?

Shuijing725, 2023-09-12:
Hi,

Thank you for providing and maintaining this repo! 
I have a question: After running `metadrive/metadrive/examples/train_generalization_experiment.py`, how to test the trained policy (for example, how to run some new unseen episodes and calculate success rate, collision rate, etc) ? Is there an example script?

QuanyiLi, 2023-09-13:
Basically, you can write an evaluation script in the following way (You need to do some modifications based on it).
```python

def eval_ckpt(config,
              ckpt_path,
              scenario_data_path,
              num_scenarios,
              start_scenario_index,
              horizon=600,
              render=False,
              # PPO is a stochastic policy, turning off exploration can reduce jitter but may harm performance
              explore=True,
              log_interval=None,
              ):
    initialize_ray(test_mode=False, num_gpus=1)
    # 27 29 30 37 39
    env_config = get_eval_config(config)["env_config"]
    env_config.update(dict(
        start_scenario_index=start_scenario_index,
        num_scenarios=num_scenarios,
        sequential_seed=True,
        curriculum_level=1,  # disable curriculum
        target_success_rate=1,
        horizon=horizon,
        episodes_to_evaluate_curriculum=num_scenarios,
        data_directory=scenario_data_path,
        use_render=render))
    env = createGymWrapper(ScenarioEnv)(env_config)

    super_data = defaultdict(list)
    EPISODE_NUM = env.config["num_scenarios"]
    compute_actions = get_function(ckpt_path, explore=explore, config=config)

    o = env.reset(seed=0)
    assert env.current_seed == start_scenario_index, "Wrong start seed!"

    total_cost = 0
    total_reward = 0
    success_rate = 0
    ep_cost = 0
    ep_reward = 0
    success_flag = False
    step = 0

    def log_msg():
        print("CKPT:{} | success_rate:{}, mean_episode_reward:{}, mean_episode_cost:{}".format(epi_num,
                                                                                               success_rate / epi_num,
                                                                                               total_reward / epi_num,
                                                                                               total_cost / epi_num))

    for epi_num in tqdm.tqdm(range(0, EPISODE_NUM)):
        step += 1
        action_to_send = compute_actions(o)["default_policy"]
        o, r, d, info = env.step(action_to_send)
        if env.config["use_render"]:
            env.render(text={"reward": r})
        total_reward += r
        ep_reward += r
        total_cost += info["cost"]
        ep_cost += info["cost"]
        if d or step > horizon:
            if info["arrive_dest"]:
                success_rate += 1
                success_flag = True
            o = env.reset(seed=epi_num)

            super_data[0].append(
                {"reward": ep_reward,
                 "success": success_flag,
                 "out_of_road": info[TerminationState.OUT_OF_ROAD],
                 "cost": ep_cost,
                 "seed": env.current_seed,
                 "route_completion": info["route_completion"]})

            ep_cost = 0.0
            ep_reward = 0.0
            success_flag = False
            step = 0

            if log_interval is not None and epi_num % log_interval == 0:
                log_msg()
    if log_interval is not None:
        log_msg()
    del compute_actions
    env.close()
    with open("eval_ret_{}_{}_{}.json".format(start_scenario_index,
                                              start_scenario_index + num_scenarios,
                                              get_time_str()), "w") as f:
        json.dump(super_data, f)
    return super_data
```

The workflow here is to 
1. Create a new environment with a new set of scenarios.
2. Build a loop traversing all scenarios by using `env.reset(seed=scenario_index)`.
3. Build a new loop for simulation with env.step(policy(observation))
4. Collect statistics after the episode
5. Save the result.

In addition, if you are using a ray, you can do the evaluation in a more convenient and efficient way. An example is at here https://github.com/metadriverse/scenarionet/blob/main/scenarionet_training/scripts/multi_worker_eval.py
What this script does is restore the checkpoint point and update the `eval_config` including the environment config. Then it can evaluate the trained policy with multi-worker, which is super fast. In our recent paper, ScenarioNet, we use this script to do evaluation as there are thousands of scenarios to be tested and efficiency matters.

Shuijing725, 2023-09-13:
Hi @QuanyiLi ,

Thank you for your quick reply! In your evaluation script, where are `get_eval_config` and `get_function` defined?

QuanyiLi, 2023-09-14:
The `get_function` should return a policy function that takes the observation as input and outputs an action. An example is like:
```
def get_function(ckpt, explore, config):
    trainer = MultiWorkerPPO(get_eval_config(config))
    trainer.restore(ckpt)

    def _f(obs):
        ret = trainer.compute_actions({"default_policy": obs}, explore=explore)
        return ret

    return _f
```
And `get_eval_config` returns an environment config which is used to build the evaluation environment. If you are going to setup an environment different from the training one for evaluation, you should use a different config to generate a new environment. An example is like:
```
# suppose you have train_cfg={"num_scenarios":1000, "start_seed":0}, you want a non-overlapping testset
def get_eval_config():
    return {"num_scenarios":100, "start_seed": 1000}
```

pengzhenghao, 2023-09-14:
Interesting. I am thinking should we formalize the concept of “Policy”? We already have an “external policy” use internally inside MetaDrive. But this is not relevant to how user should define a policy. Should we define a “Policy” class so user know what to define a policy? And then we can prepare a eval functionality that takes:

1. env config (including metrics config)
2. Policy instance

as input and return/print/report/log the metrics.

Best,
Zhenghao

Shuijing725, 2023-09-14:
> Interesting. I am thinking should we formalize the concept of “Policy”? We already have an “external policy” use internally inside MetaDrive. But this is not relevant to how user should define a [...]

I agree this would be a very useful feature.

QuanyiLi, 2023-09-15:
> Interesting. I am thinking should we formalize the concept of “Policy”? We already have an “external policy” use internally inside MetaDrive. But this is not relevant to how user should define a [...]

Indeed, we should release some official evaluation scripts or functionalities/APIs. Actually, I use ray to do parallel evaluation, which is pretty fast. achieving 1000 episodes/5min. So we can consider supporting this feature with ray.

Shuijing725, 2023-09-18:
Hi @QuanyiLi ,

Thank you for your previous reply! I still have an issue about your testing script:

In both your script and `multi_worker_eval.py`, we need to install ScenarioNet. But after installation, I found ScenarioNet is based on some previous version of ray (for example, `from ray.rllib.agents.ppo.ppo` v.s. `from ray.rllib.algorithms.ppo.ppo`).
 
So can you tell me which version of ray works with these two testing methods? The [ScenarioNet github repo says 1.0.0](https://github.com/metadriverse/scenarionet/blob/403af36ba149d84afb2ee52fe96d617d481302f9/setup.py#L50) but the oldest version I can install with pip is 1.4.1, which is also inconsistent with the grammar in ScenarioNet.

QuanyiLi, 2023-09-19:
You need to lower the version of Python to 3.7 or 3.8. And then run `pip install -e .[train]` in ScenarioNet repo.

@pengzhenghao Actually ray 2.6.0 works well for `ScenarioNet`, but `wandb` seems not compatible with it in my previous trial. Let's see if we can fix it and update the ray dependency.

Shuijing725, 2023-09-21:
Hi @QuanyiLi ,

After creating a new conda environment with Python 3.8 and ray 1.0.0, I still get an error when running `multi_worker_eval.py`. The environment I trained the policy with is `MetaDriveEnv`. Do you know why this happens?

```
Traceback (most recent call last):
  File "/fs/scratch/projects/proj-ai-planning/archive/adv_rl_planning/metadrive_all/scenarionet/scenarionet_training/scripts/multi_worker_eval.py", line 68, in <module>
    trainer = MultiWorkerPPO(config)
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/agents/trainer_template.py", line 101, in __init__
    Trainer.__init__(self, config, env, logger_creator)
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/agents/trainer.py", line 476, in __init__
    super().__init__(config, logger_creator)
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/tune/trainable.py", line 249, in __init__
    self.setup(copy.deepcopy(self.config))
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/agents/trainer.py", line 629, in setup
    self._init(self.config, self.env_creator)
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/agents/trainer_template.py", line 123, in _init
    self.workers = self._make_workers(env_creator, self._policy_class,
  File "/fs/scratch/projects/proj-ai-planning/archive/adv_rl_planning/metadrive_all/scenarionet/scenarionet_training/train_utils/multi_worker_PPO.py", line 41, in _make_workers
    return AnisotropicWorkerSet(
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/evaluation/worker_set.py", line 72, in __init__
    self._local_worker = self._make_worker(RolloutWorker, env_creator,
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/evaluation/worker_set.py", line 268, in _make_worker
    worker = cls(
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/evaluation/rollout_worker.py", line 418, in __init__
    self.policy_map, self.preprocessors = self._build_policy_map(
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/evaluation/rollout_worker.py", line 1010, in _build_policy_map
    policy_map[name] = cls(obs_space, act_space, merged_conf)
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/policy/torch_policy_template.py", line 200, in __init__
    dist_class, logit_dim = ModelCatalog.get_action_dist(
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/models/catalog.py", line 201, in get_action_dist
    raise NotImplementedError("Unsupported args: {} {}".format(
NotImplementedError: Unsupported args: Box(-1.0, 1.0, (2,), float32) None
```

QuanyiLi, 2023-09-21:
I guess it is because `MetaDrive` is using `gymnasium` now, while `ray 1.0.0` still uses `gym`.

Please try this:

```
from metadrive.envs.gym_wrapper import createGymWrapper
config = dict(
    env=createGymWrapper(MetaDriveEnv),
    ...
    )
```

Shuijing725, 2023-09-21:
I did what you said but got this error:
```
Traceback (most recent call last):
  File "/fs/scratch/projects/proj-ai-planning/archive/adv_rl_planning/metadrive_all/scenarionet/scenarionet_training/scripts/multi_worker_eval.py", line 71, in <module>
    trainer = MultiWorkerPPO(config)
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/agents/trainer_template.py", line 101, in __init__
    Trainer.__init__(self, config, env, logger_creator)
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/agents/trainer.py", line 476, in __init__
    super().__init__(config, logger_creator)
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/tune/trainable.py", line 249, in __init__
    self.setup(copy.deepcopy(self.config))
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/agents/trainer.py", line 629, in setup
    self._init(self.config, self.env_creator)
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/agents/trainer_template.py", line 123, in _init
    self.workers = self._make_workers(env_creator, self._policy_class,
  File "/fs/scratch/projects/proj-ai-planning/archive/adv_rl_planning/metadrive_all/scenarionet/scenarionet_training/train_utils/multi_worker_PPO.py", line 41, in _make_workers
    return AnisotropicWorkerSet(
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/evaluation/worker_set.py", line 72, in __init__
    self._local_worker = self._make_worker(RolloutWorker, env_creator,
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/evaluation/worker_set.py", line 268, in _make_worker
    worker = cls(
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/evaluation/rollout_worker.py", line 332, in __init__
    self.env = _validate_env(env_creator(env_context))
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/agents/trainer.py", line 1175, in <lambda>
    register_env(name, lambda config: env_object(config))
  File "/fs/scratch/projects/proj-ai-planning/archive/adv_rl_planning/metadrive_all/metadrive/metadrive/envs/gym_wrapper.py", line 82, in __init__
    super().__setattr__("_inner", actual_inner_class(config=config))
  File "/fs/scratch/projects/proj-ai-planning/archive/adv_rl_planning/metadrive_all/metadrive/metadrive/envs/metadrive_env.py", line 106, in __init__
    super(MetaDriveEnv, self).__init__(config)
  File "/fs/scratch/projects/proj-ai-planning/archive/adv_rl_planning/metadrive_all/metadrive/metadrive/envs/base_env.py", line 235, in __init__
    merged_config = self.default_config().update(config, False, ["target_vehicle_configs", "sensors"])
  File "/fs/scratch/projects/proj-ai-planning/archive/adv_rl_planning/metadrive_all/metadrive/metadrive/utils/config.py", line 122, in update
    raise KeyError(
KeyError: "'{'curriculum_level', 'start_scenario_index', 'no_light', 'steering_range_penalty', 'crash_human_penalty', 'no_static_vehicles', 'lateral_penalty', 'max_lateral_dist', 'target_success_rate', 'no_negative_reward', 'on_lane_line_penalty', 'reactive_traffic', 'heading_penalty', 'sequential_seed', 'data_directory'}' does not exist in existing config. Please use config.update(...) to update the config. Existing keys: dict_keys(['random_agent_model', 'agent_policy', 'num_agents', 'is_multi_agent', 'allow_respawn', 'delay_done', 'return_single_space', 'manual_control', 'controller', 'decision_repeat', 'discrete_action', 'discrete_steering_dim', 'discrete_throttle_dim', 'use_multi_discrete', 'action_check', 'rgb_clip', 'rgb_to_grayscale', 'stack_size', 'image_observation', 'use_render', 'debug', 'disable_model_compression', 'cull_scene', 'use_chase_camera_follow_lane', 'camera_height', 'camera_dist', 'camera_pitch', 'camera_smooth', 'camera_fov', 'prefer_track_agent', 'draw_map_resolution', 'top_down_camera_initial_x', 'top_down_camera_initial_y', 'top_down_camera_initial_z', 'show_logo', 'vehicle_config', 'sensors', 'auto_resize_window', 'target_vehicle_configs', 'window_size', 'physics_world_step_size', 'show_fps', 'global_light', 'debug_physics_world', 'debug_static_world', 'pstats', 'image_on_cuda', '_render_mode', '_disable_detector_mask', 'force_render_fps', 'force_destroy', 'num_buffering_objects', 'render_pipeline', 'daytime', 'debug_panda3d', 'drivable_region_extension', 'height_scale', '_debug_crash_object', 'horizon', 'max_step_per_agent', 'show_interface_navi_mark', 'show_mouse', 'show_skybox', 'show_terrain', 'show_interface', 'show_policy_mark', 'show_coordinates', 'interface_panel', 'multi_thread_render', 'multi_thread_render_mode', 'preload_models', 'log_level', 'record_episode', 'replay_episode', 'only_reset_when_replay', 'force_reuse_object_name', 'start_seed', 'num_scenarios', 'environment_num', 'map', 'block_dist_config', 'random_lane_width', 'random_lane_num', 'map_config', 'store_map', 'traffic_density', 'need_inverse_traffic', 'traffic_mode', 'random_traffic', 'traffic_vehicle_config', 'accident_prob', 'static_traffic_object', 'use_AI_protector', 'save_level', 'random_spawn_lane_index', 'success_reward', 'out_of_road_penalty', 'crash_vehicle_penalty', 'crash_object_penalty', 'driving_reward', 'speed_reward', 'use_lateral_reward', 'crash_vehicle_cost', 'crash_object_cost', 'out_of_road_cost', 'out_of_route_done', 'on_continuous_line_done', 'crash_vehicle_done', 'crash_object_done'])."
```

QuanyiLi, 2023-09-21:
This is because the `config["env"]=createGymWrapper(MetaDriveEnv)`, but the `config["evaluation_config"]["env_config"]` and `config["env_config"]` are still for the 'ScenarioEnv' instead of 'MetaDriveEnv'. Just modify both config dict by removing the keys that don't appear in `metadrive/envs/metadrive_env.py: METADRIVE_DEFAULT_CONFIG` or use your own training environment config.

Shuijing725, 2023-09-21:
I changed my testing config so that it is consistent with the training environment config in [`train_generalization_experiment.py`](https://github.com/metadriverse/metadrive/blob/main/metadrive/examples/train_generalization_experiment.py). My testing config is shown below:
```
config = dict(

        # ===== Training Environment =====
        # Train the policies in scenario sets with different number of scenarios.
        env=createGymWrapper(MetaDriveEnv),
        env_config=dict(
            num_scenarios=100,
            start_seed=0,
            random_traffic=False,
            traffic_density=0.1
        ),

        # ===== Framework =====
        framework="torch",

        # ===== Evaluation =====
        # Evaluate the trained policies in unseen 200 scenarios.
        evaluation_interval=2,
        evaluation_num_episodes=40,
        metrics_smoothing_episodes=200,
        evaluation_config=dict(env_config=dict(num_scenarios=200, start_seed=0)),
        evaluation_num_workers=5,
    )
```
However, I still run into the same error about gym v.s. gymnasium: 
```
Traceback (most recent call last):
  File "/fs/scratch/projects/proj-ai-planning/archive/adv_rl_planning/metadrive_all/scenarionet/scenarionet_training/scripts/multi_worker_eval.py", line 111, in <module>
    trainer = MultiWorkerPPO(config)
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/agents/trainer_template.py", line 101, in __init__
    Trainer.__init__(self, config, env, logger_creator)
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/agents/trainer.py", line 476, in __init__
    super().__init__(config, logger_creator)
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/tune/trainable.py", line 249, in __init__
    self.setup(copy.deepcopy(self.config))
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/agents/trainer.py", line 629, in setup
    self._init(self.config, self.env_creator)
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/agents/trainer_template.py", line 123, in _init
    self.workers = self._make_workers(env_creator, self._policy_class,
  File "/fs/scratch/projects/proj-ai-planning/archive/adv_rl_planning/metadrive_all/scenarionet/scenarionet_training/train_utils/multi_worker_PPO.py", line 41, in _make_workers
    return AnisotropicWorkerSet(
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/evaluation/worker_set.py", line 72, in __init__
    self._local_worker = self._make_worker(RolloutWorker, env_creator,
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/evaluation/worker_set.py", line 268, in _make_worker
    worker = cls(
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/evaluation/rollout_worker.py", line 418, in __init__
    self.policy_map, self.preprocessors = self._build_policy_map(
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/evaluation/rollout_worker.py", line 1010, in _build_policy_map
    policy_map[name] = cls(obs_space, act_space, merged_conf)
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/policy/torch_policy_template.py", line 200, in __init__
    dist_class, logit_dim = ModelCatalog.get_action_dist(
  File "/home/lis2syv/.conda/envs/old_metadrive/lib/python3.8/site-packages/ray/rllib/models/catalog.py", line 201, in get_action_dist
    raise NotImplementedError("Unsupported args: {} {}".format(
NotImplementedError: Unsupported args: Box(-1.0, 1.0, (2,), float32) None
```

QuanyiLi, 2023-09-22:
That's weird. I can not help more given the limited information. By setting `initialize_ray(test_mode=True, num_gpus=1, local_mode=True)`, you should be able to put a breakpoint at the line causing the bug and check why this happens. The `rllib` agent should choose the `Box space` branch in the decision tree. 

Another way is to give me a runnable script to reproduce this error. Once I reproduced it, I can help you with it.

Shuijing725, 2023-09-22:
[files.zip](https://github.com/metadriverse/metadrive/files/12704929/files.zip)
In the zip folder, my conda environment is in env.yml, and my modified testing script is `multi_worker_eval.py`. Let me know if you need anything else from me. Thank you in advance!

QuanyiLi, 2023-09-23:
I can not reproduce the issue. My action is 1) create conda env from `env.yml` 2) `cd metadrive & pip install -e .` 3) `cd scenarionet & pip install -e .`.

There are some bugs with your 'multi_worker_evel' script. I fixed them and everything works fine for me. The fixed script now is like:

```
import argparse
import pickle
import json
import os

import numpy as np

from scenarionet_training.scripts.train_nuplan import config
from scenarionet_training.train_utils.callbacks import DrivingCallbacks
from ray.rllib.agents.ppo.ppo import PPOTrainer
from scenarionet_training.train_utils.utils import initialize_ray

from metadrive.envs.gym_wrapper import createGymWrapper
from metadrive import MetaDriveEnv

class NumpyEncoder(json.JSONEncoder):
    def default(self, obj):
        if isinstance(obj, np.ndarray):
            return obj.tolist()
        elif isinstance(obj, np.int32):
            return int(obj)
        elif isinstance(obj, np.int64):
            return int(obj)
        return json.JSONEncoder.default(self, obj)

if __name__ == '__main__':
    parser = argparse.ArgumentParser()
    parser.add_argument("--start_index", type=int, default=0)
    # parser.add_argument("--ckpt_path", type=str, required=True)
    parser.add_argument("--database_path", type=str, required=False)
    parser.add_argument("--id", type=str, default="")
    parser.add_argument("--num_scenarios", type=int, default=5000)
    parser.add_argument("--num_workers", type=int, default=10)
    parser.add_argument("--horizon", type=int, default=600)
    parser.add_argument("--allowed_more_steps", type=int, default=50)
    parser.add_argument("--max_lateral_dist", type=int, default=2.5)
    parser.add_argument("--overwrite", action="store_true")

    args = parser.parse_args()
    # file = "eval_{}_{}_{}".format(args.id, os.path.basename(args.ckpt_path), os.path.basename(args.database_path))
    file = "eval_{}_{}".format(args.id, "test")
    initialize_ray(test_mode=True, num_gpus=0, local_mode=False)

    config = dict(

        # ===== Training Environment =====
        # Train the policies in scenario sets with different number of scenarios.
        env=createGymWrapper(MetaDriveEnv),
        env_config=dict(
            num_scenarios=100,
            start_seed=0,
            random_traffic=False,
            traffic_density=0.1
        ),

        # ===== Framework =====
        framework="torch",

        # ===== Evaluation =====
        # Evaluate the trained policies in unseen 200 scenarios.
        evaluation_interval=2,
        evaluation_num_episodes=40,
        metrics_smoothing_episodes=200,
        evaluation_config=dict(env_config=dict(num_scenarios=200, start_seed=0)),
        evaluation_num_workers=10,

        # ===== Training =====
        # Hyper-parameters for PPO
        # horizon=1000,
        # rollout_fragment_length=200,
        # sgd_minibatch_size=256,
        # train_batch_size=20000,
        # num_sgd_iter=10,
        # lr=3e-4,
        num_workers=0,
        # **{"lambda": 0.95},

        # ===== Resources Specification =====
        # num_gpus=0.25 if args.num_gpus != 0 else 0,
        # num_cpus_per_worker=0.2,
        # num_cpus_for_driver=0.5,
    )

    trainer = PPOTrainer(config)
    # trainer.restore(args.ckpt_path)

    ret = trainer._evaluate()["evaluation"]
    with open(file + ".json", "w") as f:
        json.dump(ret, f, cls=NumpyEncoder)

    with open(file + ".pkl", "wb+") as f:
        pickle.dump(ret, f)

```
The output on my machine is 
[image]

If it takes too much time to use this script. Let's just fall back to the naive evaluatoin method. Indeed, it is slow due to single-process run, but you can do some parallel with python's multi-process lib.

Shuijing725, 2023-09-24:
May I ask which version of MetaDrive did you install? After following all of your actions and running the corrected script, a new error appears and might be caused by some files in metadrive:
```
            |
Traceback (most recent call last):
  File "/fs/scratch/projects/proj-ai-planning/archive/adv_rl_planning/metadrive_all/scenarionet/scenarionet_training/scripts/multi_worker_eval.py", line 89, in <module>
    ret = trainer._evaluate()["evaluation"]
  File "/home/lis2syv/.conda/envs/metadrive/lib/python3.8/site-packages/ray/rllib/agents/trainer.py", line 747, in _evaluate
    ray.get([
  File "/home/lis2syv/.conda/envs/metadrive/lib/python3.8/site-packages/ray/worker.py", line 1428, in get
    raise value.as_instanceof_cause()
ray.exceptions.RayTaskError(FileExistsError): ray::RolloutWorker.sample() (pid=2940764, ip=10.124.70.60)
  File "python/ray/_raylet.pyx", line 484, in ray._raylet.execute_task
  File "python/ray/_raylet.pyx", line 438, in ray._raylet.execute_task.function_executor
  File "/home/lis2syv/.conda/envs/metadrive/lib/python3.8/site-packages/ray/rllib/evaluation/rollout_worker.py", line 579,in sample
    batches = [self.input_reader.next()]
  File "/home/lis2syv/.conda/envs/metadrive/lib/python3.8/site-packages/ray/rllib/evaluation/sampler.py", line 93, in next
    batches = [self.get_data()]
  File "/home/lis2syv/.conda/envs/metadrive/lib/python3.8/site-packages/ray/rllib/evaluation/sampler.py", line 209, in get_data
    item = next(self.rollout_provider)
  File "/home/lis2syv/.conda/envs/metadrive/lib/python3.8/site-packages/ray/rllib/evaluation/sampler.py", line 548, in _env_runner
    base_env.poll()
  File "/home/lis2syv/.conda/envs/metadrive/lib/python3.8/site-packages/ray/rllib/env/base_env.py", line 325, in poll
    self.new_obs = self.vector_env.vector_reset()
  File "/home/lis2syv/.conda/envs/metadrive/lib/python3.8/site-packages/ray/rllib/env/vector_env.py", line 133, in vector_reset
    return [e.reset() for e in self.envs]
  File "/home/lis2syv/.conda/envs/metadrive/lib/python3.8/site-packages/ray/rllib/env/vector_env.py", line 133, in <listcomp>
    return [e.reset() for e in self.envs]
  File "/fs/scratch/projects/proj-ai-planning/archive/adv_rl_planning/metadrive_all/metadrive/metadrive/envs/gym_wrapper.py", line 94, in reset
    obs, _ = self._inner.reset(**not_none_params)
  File "/fs/scratch/projects/proj-ai-planning/archive/adv_rl_planning/metadrive_all/metadrive/metadrive/envs/base_env.py",line 480, in reset
    self.lazy_init()  # it only works the first time when reset() is called to avoid the error when render
  File "/fs/scratch/projects/proj-ai-planning/archive/adv_rl_planning/metadrive_all/metadrive/metadrive/envs/base_env.py",line 366, in lazy_init
    initialize_engine(self.config)
  File "/fs/scratch/projects/proj-ai-planning/archive/adv_rl_planning/metadrive_all/metadrive/metadrive/engine/engine_utils.py", line 12, in initialize_engine
    cls.singleton = cls(env_global_config)
  File "/fs/scratch/projects/proj-ai-planning/archive/adv_rl_planning/metadrive_all/metadrive/metadrive/engine/base_engine.py", line 35, in __init__
    self.try_pull_asset()
  File "/fs/scratch/projects/proj-ai-planning/archive/adv_rl_planning/metadrive_all/metadrive/metadrive/engine/base_engine.py", line 699, in try_pull_asset
    pull_asset(False)
  File "/fs/scratch/projects/proj-ai-planning/archive/adv_rl_planning/metadrive_all/metadrive/metadrive/pull_asset.py", line 65, in pull_asset
    zip_ref.extractall(TARGET_DIR)
  File "/home/lis2syv/.conda/envs/metadrive/lib/python3.8/zipfile.py", line 1647, in extractall
    self._extract_member(zipinfo, path, pwd)
  File "/home/lis2syv/.conda/envs/metadrive/lib/python3.8/zipfile.py", line 1697, in _extract_member
    os.mkdir(targetpath)
FileExistsError: [Errno 17] File exists: '/fs/scratch/projects/proj-ai-planning/archive/adv_rl_planning/metadrive_all/metadrive/metadrive/assets'
```

QuanyiLi, 2023-09-24:
Could you run `python -m metadrive.pull_assets --update`?

Shuijing725, 2023-09-24:
Running `python -m metadrive.pull_assets --update` returns:
`/home/lis2syv/.conda/envs/metadrive/bin/python: No module named metadrive.pull_assets`.
So instead, I cd into the directory with `pull_assets.py` and run `python pull_asset.py --update`, and it returns: 
`Fail to pull. Assets is already up-to-date, version: 0.4.1.2. MetaDrive version: 0.4.1.2`

QuanyiLi, 2023-09-24:
hmmm, could you rerun the previous failed script?

Shuijing725, 2023-09-24:
Running the previous script without `--ckpt_path` works and I got the same output as you. Thanks a lot for your help!

However, when I uncomment and set the `--ckpt_path` to my trained checkpoint path from `train_generalization_experiment.py`, I got this error:
```
Traceback (most recent call last):
  File "/fs/scratch/projects/proj-ai-planning/archive/adv_rl_planning/metadrive_all/scenarionet/scenarionet_training/scripts/multi_worker_eval.py", line 88, in <module>
    trainer.restore(args.ckpt_path)
  File "/home/lis2syv/.conda/envs/metadrive/lib/python3.8/site-packages/ray/tune/trainable.py", line 462, in restore
    with open(checkpoint_path + ".tune_metadata", "rb") as f:
FileNotFoundError: [Errno 2] No such file or directory: '/fs/scratch/projects/proj-ai-planning/archive/adv_rl_planning/metadrive_all/metadrive_modify/experiments/ppo_50000scenarios/PPO_GymEnvWrapper_e2619_00001_1_num_scenarios/checkpoint_000500.tune_metadata'
```

The structure of my trained checkpoints looks like `checkpoint_path/output.png` below. 
[checkpoint_path.zip](https://github.com/metadriverse/metadrive/files/12709711/checkpoint_path.zip)

My question is: Is the reason due to 
1. The training and testing script require different output structures? Thus, I need to re-train the policy with the scripts in `scenarionet/scenarionet_training/scripts`
2. Or something else?

QuanyiLi, 2023-09-24:
There is a `.tune_metadata`. 

So try setting the ckpt path as `/fs/scratch/projects/proj-ai-planning/archive/adv_rl_planning/metadrive_all/metadrive_modify/experiments/ppo_50000scenarios/PPO_GymEnvWrapper_e2619_00001_1_num_scenarios/checkpoint_000500/`
The last slash `/` matters, I guess.

Shuijing725, 2023-09-24:
After adding the slash, the error becomes:
```
Traceback (most recent call last):
  File "multi_worker_eval.py", line 88, in <module>
    trainer.restore(args.ckpt_path)
  File "/home/lis2syv/.conda/envs/metadrive/lib/python3.8/site-packages/ray/tune/trainable.py", line 463, in restore
    metadata = pickle.load(f)
ModuleNotFoundError: No module named 'ray.rllib.core'
```
Reinstalling ray by `pip install ray[rllib]==1.0.0` and `pip install ray==1.0.0` did not solve the above error.

QuanyiLi, 2023-09-24:
I guess that is because your agent is trained with ray in another version. When pickle loads data, it will load the python module as well. If it can not find the module when loading, it means that the module is in another version when you save the `.pkl`.

Shuijing725, 2023-09-25:
I see. Running `train_generalization_experiment.py` in the conda environment with ray=1.0.0 (the same environment that you reproduced) also gives errors and requires some debugging.

**Before bothering you to look at my error trace again, I want to consult your advice on a higher level:**

**My goal is to train and test ANY RL policy in ANY environment with reactive agents, meaning that the agents are controlled by some policies such as IDM, instead of logged data. Ideally, the agents should be able to change lanes as well.**

**So in your opinion, what is the easiest way of achieving my goal?**

For reference, when I run `train_generalization_experiment.py`, it returns 
```
2023-09-25 18:58:21,955 ERROR trial_runner.py:567 -- Trial PPO_GymEnvWrapper_90e19_00000: Error processing event.
Traceback (most recent call last):
  File "/home/lis2syv/.conda/envs/metadrive/lib/python3.8/site-packages/ray/tune/trial_runner.py", line 515, in _process_trial
    result = self.trial_executor.fetch_result(trial)
  File "/home/lis2syv/.conda/envs/metadrive/lib/python3.8/site-packages/ray/tune/ray_trial_executor.py", line 488, in fetch_result
    result = ray.get(trial_future[0], timeout=DEFAULT_GET_TIMEOUT)
  File "/home/lis2syv/.conda/envs/metadrive/lib/python3.8/site-packages/ray/worker.py", line 1428, in get
    raise value.as_instanceof_cause()
ray.exceptions.RayTaskError(TypeError): ray::PPO.train() (pid=451625, ip=10.124.70.58)
  File "python/ray/_raylet.pyx", line 484, in ray._raylet.execute_task
  File "python/ray/_raylet.pyx", line 438, in ray._raylet.execute_task.function_executor
  File "/home/lis2syv/.conda/envs/metadrive/lib/python3.8/site-packages/ray/rllib/agents/trainer.py", line 519, in train
    raise e
  File "/home/lis2syv/.conda/envs/metadrive/lib/python3.8/site-packages/ray/rllib/agents/trainer.py", line 505, in train
    result = Trainable.train(self)
  File "/home/lis2syv/.conda/envs/metadrive/lib/python3.8/site-packages/ray/tune/trainable.py", line 398, in train
    self.log_result(result)
  File "/home/lis2syv/.conda/envs/metadrive/lib/python3.8/site-packages/ray/rllib/agents/trainer.py", line 549, in log_result
    self.callbacks.on_train_result(trainer=self, result=result)
TypeError: on_train_result() missing 1 required keyword-only argument: 'algorithm'
```

QuanyiLi, 2023-09-25:
The problem is at https://github.com/metadriverse/metadrive/blob/2f8379a789c97956a9acd994c11df4d07086796e/metadrive/examples/train_generalization_experiment.py#L80

Make it  `def on_train_result(self, *, trainer, result: dict, **kwargs):` should fix the bug.

For another question (which can be referred to as reactive traffic simulation), I don't have a good answer about it. Modeling reactive behaviors for surrounding vehicles is a challenging task. A general solution now is allowing surrounding vehicles to follow a trajectory with an IDM to make the traffic flow reactive. But it is still flexible enough and can not perform lane-changing. It is still an open-problem and needs further exploration.

QuanyiLi, 2023-09-28:
OK. Now my understanding is that:

You want to train agents in real-world scenarios/synthetic scenarios in reactive traffic mode where surrounding vehicles may yield the ego car if there is a potential collision. And you want your ego vehicle to reach a destination without restricting its trajectory like forcing it to follow the logged path. After training, some behaviors like lane-changing should emerge.

To this end, yes. MetaDrive is able to achieve this from the environment side. For the training side, you can use any RL frameworks such as sb3, rllib, and so on. We believe they are all compatible with MetaDrive. Of course, you can adapt `train_generalization_experiment.py` and `multi_worker_eval.py` to suit your needs as they are built on rllib.

Shuijing725, 2023-09-28:
I see. I have one last question: In your previous papers such as "[Expert Guided Policy Optimization](https://github.com/decisionforce/EGPO#expert-guided-policy-optimization-egpo)" and "[Guarded Policy Optimization with Imperfect Online Demonstrations](https://github.com/metadriverse/TS2C#guarded-policy-optimization-with-imperfect-online-demonstrations)", how did you get the quantitative testing results (such as Table 1 and Fig. 4 bottom in the first paper)?

QuanyiLi, 2023-09-29:
Firstly, we have several metrics for each episode, like success/collision/episode return/episode cost, and out-of-road. Then we run the policy on all test environments, i.e. 50 environments, 1 episode per environment, and collect the metrics for each episode. After this, we average the metrics on all environments to get the statistics. For example, if the police manages to arrive the destination in 40 environments, then we get the test success rate as 40/50 = 80 %.

Shuijing725, 2023-09-29:
Is [this file](https://github.com/decisionforce/EGPO/blob/main/training_script/evaluate_egpo_with_human.py) one of the testing scripts you refer to?

QuanyiLi, 2023-10-01:
Yes, it is an evaluation script.