metadriverse/metadrive #497: Some questions about the state space and reward function of MetaDrive-based environments

c4cld, 2023-09-18:
In paper 'MetaDrive: Composing Diverse Driving Scenarios for Generalizable Reinforcement Learning' , the authors introduce the observation and reward function as follows.

The observation of RL agents is as follows:
(a) A 240-dimensional (72-dimensional in MARL) vector denoting the 2D-Lidar-like point clouds with 50m maximum detecting distance centering at the target vehicle. Each entry is in [0,1] representing the relative distance of the nearest obstacle in the specified
direction.
(b) A vector summarizing the target vehicle’s state such as the steering, heading, velocity, and relative distance to the left and right boundaries.
(c) The navigation information that guides the target vehicle toward the destination. We sparsely spread a set of checkpoints, 50m apart on average, in the route and use the relative positions toward future checkpoints as observation to the target vehicle.

The reward function is composed of three parts as follows:
R = c_{1} * R_{disp} + c_{2} * R_{speed} + R_{term}
R_{disp}
The displacement reward R_{disp} = d_{t} - d_{t-1}, wherein the d_{t} and d_{t-1} denote the longitudinal movement of the target vehicle in Frenet coordinates of current lane between two consecutive time steps, provides dense reward to encourage agent to move forward.
R_{speed}
The speed reward R_{speed} = v_{t}/v_{max} incentivizes agent to drive fast. v_{t} and v_{max} denote the current
velocity and the maximum velocity (80 km/h), respectively.
R_{term}
A sparse terminal reward R_{term}, which is non-zero only at the last time step. At that step.  R_{term} is set to +10 if the vehicle reaches the destination, -5 for crashing others or violating traffic rules such as locating at undrivable area.
c_{1} is set to 1 and c_{2} is set to 0.1.

I have some questions about the state space and reward function of MetaDrive-based environments as follows:
1 What is the value range of target vehicle’s state such as the steering, heading, velocity, and relative distance to the left and right boundaries?
2 What is the value range of navigation information? 
3 What is the value range of R_{disp}?

QuanyiLi, 2023-09-19:
It is generally between 0-2

c4cld, 2023-09-21:
Thank you very much

c4cld, 2024-01-12:
In R_{speed}, the agent is already motivated to drive as fast as possible. In this case, is it necessary to consider the completion time of the task in the reward function? 
In other words, in a fixed environment with a specific task, is motivating the agent to drive as fast as possible in the reward function equivalent to motivating the agent to finish the episode in the shortest possible time in the reward function?

closed 2024-01-19 by QuanyiLi