metadriverse/metadrive #513: About VaryingDynamicsEnv

c4cld, 2023-10-17:
About the details of VaryingDynamicsEnv, I have some questions as follows:
1 What's the number of obstacles? What's the properties (static or dynamic) of the obstacles? If there is no obstacle, how can I add them?
2 How many other vehicles are included in this environment? Do these vehicles use IDM policies? If there is no other vehicle, how can I add them?
3 What's the structure (Straight/Ramp/Fork/Roundabout/Curve/T-Intersection/) of roads ? Are they unban roads or highway roads? 
4 What's the max episode length of this environment? What's the max episode reward of this environment?

QuanyiLi, 2023-10-17:
1. I cannot give you a certain answer, as the map is not fixed for each episode. For a specific map, the algorithm will automatically scatter some obstacles and vehicles onto it. For adding obstacles, set `config["accident_prob"]` to a number greater than 0 
2. Same as the previous question. I can not give you a certain answer. Now the `config["traffic_density"]==0.15`, so there will be approximately 6 vehicles in each episode. You can increase the number to add more vehicles.
3. They are urban roads. 
4. We don't have a maximum episode length currently. You can set the max_episode_length by setting `config["horizon"]`. When reaching that episode length, the episode will be terminated automatically. For the reward, I guess it would be in the range of [200, 400]. It depends on the map.

pengzhenghao, 2023-10-17:
VaryingDynamicsEnv is basically the single-agent MetaDrive env (with PG maps). So all the things apply to it.

c4cld, 2023-10-18:
1 The paper of Metadrive mentions that the map generated by the Procedural Generation method is composed of several blocks, and each block is one of the following types: Straight, Ramp, Fork, Roundabout, Curve, T-Intersection, and Intersection. What is a reasonable number of blocks to choose?
2 If the maximum episode reward is in the range of [200, 400], what is the approximate range for the number of blocks?
3 Could you please tell me the conversion relationship between the traffic_density parameter and the number of other vehicles?

QuanyiLi, 2023-10-18:
1. three. the use `env_config["map"]=3` or `env_config["map"]="OXC"` to change the map. The former one creates map from the number of blocks, while the latter one creates from block type.
2. The reward depends mainly on the length of the routes. so the longer the path or the more blocks, the more reward.
3. There is no rule to calculate the exact number of vehicles from the `traffic_density`. But we usually use `traffic_density<0.2`
4. Same as the last problem. It is because that each block may have varying sizes, thus the number of vehicles or objects can be contained in it is different and case by case. Thus these two parameters are too rough to get an exact number of objects/cars.

c4cld, 2023-10-19:
Thank you very much for your answer.

closed 2023-10-19 by c4cld