metadriverse/metadrive #520: docstring coverage message bot raises error when merge PR

pengzhenghao, 2023-10-20:
@QuanyiLi 
@BoSmallEar 

Do we have any idea to fix this..

pengzhenghao, 2023-10-20:
It's working when the PR is not merged. It can find the PR webpage successfully so it can make comment on that PR. But when the PR is merged, it can't find the correct destination..

pengzhenghao, 2023-10-20:
In repo main page:

[image]

In PR:

[image]

QuanyiLi, 2023-10-20:
Can you specify when to trigger the bot? Only trigger it when we are committing things to PR.

QuanyiLi, 2023-10-20:
Finished in #521. The action will bypass the docstring check now: https://github.com/metadriverse/metadrive/actions/runs/6587476374

closed 2023-10-20 by QuanyiLi