metadriverse/metadrive #522: If the values of the parameters accident_prob and traffic_density are altered, will the dimensionality of the observation be altered as well?

c4cld, 2023-10-23:
In the 196th line of the file `state_obs.py` in the [Metadrive repository](https://github.com/metadriverse/metadrive/blob/main/metadrive/obs/state_obs.py), the dimension of the observation is determined to be 9 (Ego states) + 10 (Navi info) + 4 * closest vehicle info + 240 (Lidar points).

If the values of the parameters accident_prob and traffic_density are altered, will the dimensionality of the observation be altered as well?

QuanyiLi, 2023-10-24:
No, the observation will persist. The change in the number of vehicles and objects will be reflected by Lidar result

c4cld, 2023-10-25:
I have a question concerning observation.

You clarify about the observation as follows：The observation of RL agents is composed of "A vector summarizing the target vehicle’s state", "The navigation information" and "A 240-dimensional vector denoting the 2D-Lidar-like point clouds". According to this clarification, the dimensionality of observation vector is 9 + 10 + 240 = 259.

But you  write in https://github.com/metadriverse/metadrive/blob/main/metadrive/obs/state_obs.py line 196 as follows: observation in 9 + 10 + 16 + 240 dim. So observation vector is a 275-dimensional vector.

This raises the question of whether the 16-dimensional vector representing the closest vehicle information is included in the observation vector. Is the observation vector a 259-dimensional vector or a 275-dimensional vector?

c4cld, 2023-10-25:
Thank you very much for your answer.

closed 2023-10-25 by c4cld

mimjing, 2024-12-31:
> the 16 dim vector which represents the closest 4 vehicles from the observation, as we find it is not very helpful for the performance

Haven't you make experiments on the scenarionet-env?  I want to know whether need to add the closet vehicles' info when i train the agent in nuplan dataset? It always doesn't work well. The reward curve does not converge.
[image]