metadriverse/metadrive #544: [bug]assets path error

CarlDegio, 2023-11-05:
Ubuntu22.04, miniconda, python3.9
I just uninstall metadrive using pip and delete metadrive folder(delete assets folder). And install metadrive by `pip install git+https://github.com/metadriverse/metadrive.git`. However, it can't run test `python -m metadrive.examples.drive_in_single_agent_env` with this error:
```
[INFO] MetaDrive version: 0.4.1.2
[INFO] Sensors: [lidar: Lidar(50,), side_detector: SideDetector(), lane_line_detector: LaneLineDetector(), dashboard: DashBoard(), main_camera: MainCamera(1200, 900)]
[INFO] Render Mode: onscreen
[WARNING] Assets folder doesn't exist. Begin to download assets... (base_engine.py:704)
[INFO] Pull assets from https://github.com/metadriverse/metadrive/releases/download/MetaDrive-0.4.1.2/assets.zip
100% |#########################################################################|
[INFO] Successfully download assets, version: 0.4.1.2. MetaDrive version: 0.4.1.2
Known pipe types:
  glxGraphicsPipe
(1 aux display modules not yet loaded.)
:shader(error): Could not find shader file: /home/lzh/miniconda3/envs/metadrive/lib/python3.9/site-packages/metadrive/assets/../shaders/skybox.vert.glsl
Traceback (most recent call last):
  File "/home/lzh/miniconda3/envs/metadrive/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/lzh/miniconda3/envs/metadrive/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/lzh/miniconda3/envs/metadrive/lib/python3.9/site-packages/metadrive/examples/drive_in_single_agent_env.py", line 78, in <module>
    raise e
  File "/home/lzh/miniconda3/envs/metadrive/lib/python3.9/site-packages/metadrive/examples/drive_in_single_agent_env.py", line 52, in <module>
    o, _ = env.reset(seed=21)
  File "/home/lzh/miniconda3/envs/metadrive/lib/python3.9/site-packages/metadrive/envs/base_env.py", line 485, in reset
    self.lazy_init()  # it only works the first time when reset() is called to avoid the error when render
  File "/home/lzh/miniconda3/envs/metadrive/lib/python3.9/site-packages/metadrive/envs/base_env.py", line 371, in lazy_init
    initialize_engine(self.config)
  File "/home/lzh/miniconda3/envs/metadrive/lib/python3.9/site-packages/metadrive/engine/engine_utils.py", line 12, in initialize_engine
    cls.singleton = cls(env_global_config)
  File "/home/lzh/miniconda3/envs/metadrive/lib/python3.9/site-packages/metadrive/engine/base_engine.py", line 36, in __init__
    EngineCore.__init__(self, global_config)
  File "/home/lzh/miniconda3/envs/metadrive/lib/python3.9/site-packages/metadrive/engine/core/engine_core.py", line 288, in __init__
    self.sky_box = SkyBox(not self.global_config["show_skybox"])
  File "/home/lzh/miniconda3/envs/metadrive/lib/python3.9/site-packages/metadrive/engine/core/sky_box.py", line 53, in __init__
    skybox.set_shader(skybox_shader)
TypeError: NodePath.set_shader() argument 1 must be Shader, not NoneType
```
maybe there are some path error? the assets are download successfully.
Besides, I install it with `pip install metadrive-simulator`, no error.

QuanyiLi, 2023-11-05:
Thank you. I reproduced it. It should be fixed now. Can you try it again?

CarlDegio, 2023-11-06:
It works well now! Thank you!

closed 2023-11-06 by CarlDegio