metadriverse/metadrive #553: [Help Requst]Imcompatibale version with argoverse-api

l1xiao, 2023-11-16:
Hi,
I followed the "Read Data from Dataset" chapter as the doc described. And found that the argoverse-api requires numpy==1.19 but metadrive will install numpy=1.24.2. I set numpy==1.24.2 in argoverse-api setup.py, but still failed to run `python -m metadrive.examples.drive_in_argoverse`:
/usr/bin/python: No module named metadrive.examples.drive_in_argoverse

Is there any solution to this problem?

QuanyiLi, 2023-11-16:
Sorry, Argoverse-related API and environment have been deprecated for quite a long time. However, [ScenarioNet](https://github.com/metadriverse/scenarionet) is going to support Argoverse soon! We expect to release Argoverse convertor before the end of this month.

l1xiao, 2023-11-16:
Thanks for your reply!

closed 2023-11-16 by l1xiao