metadriverse/metadrive #556: [Help Request] Failed to ran 'python3.8 train_generalization_experiment.py '

l1xiao, 2023-11-16:
Hi, I ran `python3.8 train_generalization_experiment.py ` but failed. Logs are following:
Traceback (most recent call last):
  File "train_generalization_experiment.py", line 16, in <module>
    from metadrive.envs.gym_wrapper import createGymWrapper
ImportError: cannot import name 'createGymWrapper' from 'metadrive.envs.gym_wrapper' (/home/xx/metadrive/metadrive/envs/gym_wrapper.py)

I tried other examples and all worked well.

System: Ubuntu 20.04
Commit:
commit 0df2f979e97e39282b3634a7984a2e20c77e4263 (HEAD -> main, origin/main, origin/HEAD)
Author: Quanyi Li <quanyili0057@gmail.com>
Date:   Tue Nov 14 00:47:35 2023 +0000
    Cleanup (#551)
    * rename return raw style
    * clean lanes
    * optimize sidewalk for PGMap
 
Is there any solution to this problem?

l1xiao, 2023-11-16:
I modified as follow, 
[image]
[image]
and install ray==1.2.0. There is still has error:
/home/xx/metadrive/venv/bin/python /home/lixiao07/metadrive/metadrive/examples/train_generalization_experiment.py 
```
Traceback (most recent call last):
  File "/home/xx/metadrive/metadrive/examples/train_generalization_experiment.py", line 19, in <module>
    import ray
  File "/home/xx/metadrive/venv/lib/python3.8/site-packages/ray/__init__.py", line 63, in <module>
    import ray._raylet  # noqa: E402
  File "python/ray/_raylet.pyx", line 102, in init ray._raylet
  File "/home/xx/metadrive/venv/lib/python3.8/site-packages/ray/exceptions.py", line 7, in <module>
    from ray.core.generated.common_pb2 import RayException, Language, PYTHON
  File "/home/xx/metadrive/venv/lib/python3.8/site-packages/ray/core/generated/common_pb2.py", line 33, in <module>
    _descriptor.EnumValueDescriptor(
  File "/home/xx/metadrive/venv/lib/python3.8/site-packages/google/protobuf/descriptor.py", line 789, in __new__
    _message.Message._CheckCalledFromGeneratedFile()
TypeError: Descriptors cannot be created directly.
If this call came from a _pb2.py file, your generated code is out of date and must be regenerated with protoc >= 3.19.0.
If you cannot immediately regenerate your protos, some other possible workarounds are:
 1. Downgrade the protobuf package to 3.20.x or lower.
 2. Set PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python (but this will use pure-Python parsing and will be much slower).
```

QuanyiLi, 2023-11-17:
Please use `ray==1.0.0` or `ray=1.2.0` to run the experiments. As for the `gymWrapper` error, did you install `gym`, which should be the `openai.gym` instead of `gymnasium`?

closed 2024-01-19 by QuanyiLi