metadriverse/metadrive #564: Change Meta-Drive to Multi-Objective Reinforcement learning environment

man469, 2023-11-22:
I am trying to applying morl-baselines library from https://github.com/LucasAlegre/morl-baselines to meta-drive. However, it requires meta-drive to do some modifications, especially in reward function. Moreover, the meta-drive should follow the mo-gymnasium environment. Does anyone know how to make this? Thank you!

QuanyiLi, 2023-11-22:
It seems that the only thing needed is decomposing the reward function into a set of scalars. For example, in the reward function of `MetaDriveEnv`, the reward function is calculated as mainly the weighted sum of displacement reward and velocity reward: `r=w_1*r_d + w_2*r_v`. You can modify the reward function to return `[r_d,  r_v]` instead of `r` and it should be fine with the morl algorithms.

pengzhenghao, 2023-11-22:
A workaround is to adding additional reward function into the info dict. And we can write a wrapper outside to interpret those values in info dict to a list of reward:

```
class Wrapper:

  def step(..):
    o, r, tm, tc, i = super().step(...)
    new_reward = [i["metareward1"], i["metareward2], ...]
    return o, new_reward, tm, tc, i
    
```

closed 2024-01-19 by QuanyiLi