metadriverse/metadrive #565: The difference between cost and terminals reward?

CarlDegio, 2023-11-22:
I test MetaDriveEnv and SafeMetaDriveEnv. And found when the collision happen, the reward is calculated by terminals reward(-5.0). But in the [doc](https://metadrive-simulator.readthedocs.io/en/latest/reward_cost_and_termination_function.html), looks more like the collision results in a penalty rather than the terminals. 

I found in the source code that cost only adds info and has nothing to do with rewards. Maybe the cost can be described as the number of violations rather than the penalty value, otherwise it may cause confusion (reward+cost).

QuanyiLi, 2023-11-22:
As we all know, RL strongly relies on the design of these functions and our implementation may not be the best.  Users are encouraged to define the reward/cost/termination functions on their own.

pengzhenghao, 2023-11-22:
The reason we provide cost in hte infor is because in Safe RL the cost is used in another pipeline, such as estimating a value network solely for the cost. So we don't want to implement the reward shaping: reward = reward - cost inside the environment.