metadriverse/metadrive #573: local coordinates in env.engine.make_point_drawer is invalid.

CarlDegio, 2023-12-07:
I want to draw points in local coordinates, so I imitate the way in `env.engine.make_line_drawer(env.vehicle.origin,thickness=5)` as `env.engine.make_point_drawer(env.vehicle.origin, scale=1)`.

However, this seems to be invalid. Firstly there are no setH in make points, then drawer.draw_points(points, colors) seems show nothing. (Even if there is no rotation by setH, it should still be drawn on the side of the car)

Additionally, I suggest unifying the interfaces for drawing lines and drawing points. `lines = [line_1, line_2]` and `points = line_1 + line_2` may cause some confusion.

ref: https://metadrive-simulator.readthedocs.io/en/latest/points_and_lines.html

QuanyiLi, 2023-12-07:
I merged a PR which fixes this

CarlDegio, 2023-12-18:
Sorry for the late test. I found there is a running bug in the latest version. I uninstall metadrive and reinstall use:

`pip install git+https://github.com/metadriverse/metadrive.git`

Then run a simple test with error:
```
[INFO] Environment: MetaDriveEnv
[INFO] MetaDrive version: 0.4.1.2
[INFO] Sensors: [lidar: Lidar(), side_detector: SideDetector(), lane_line_detector: LaneLineDetector(), dashboard: DashBoard(), main_camera: MainCamera(1200, 900)]
[INFO] Render Mode: onscreen
[INFO] Horizon (Max steps per agent): 1000
/home/lzh/miniconda3/envs/metadrive/lib/python3.9/site-packages/gymnasium/spaces/box.py:130: UserWarning: WARN: Box bound precision lowered by casting to float32
  gym.logger.warn(f"Box bound precision lowered by casting to {self.dtype}")
[INFO] Assets version: 0.4.1.2
[INFO] Known Pipes: glxGraphicsPipe
Traceback (most recent call last):
  File "/home/lzh/PycharmProjects/meta_frame/traj_follow_policy.py", line 169, in <module>
    o, info = env.reset(seed=100)
  File "/home/lzh/miniconda3/envs/metadrive/lib/python3.9/site-packages/metadrive/envs/base_env.py", line 555, in reset
    self.lazy_init()  # it only works the first time when reset() is called to avoid the error when render
  File "/home/lzh/miniconda3/envs/metadrive/lib/python3.9/site-packages/metadrive/envs/base_env.py", line 431, in lazy_init
    initialize_engine(self.config)
  File "/home/lzh/miniconda3/envs/metadrive/lib/python3.9/site-packages/metadrive/engine/engine_utils.py", line 12, in initialize_engine
    cls.singleton = cls(env_global_config)
  File "/home/lzh/miniconda3/envs/metadrive/lib/python3.9/site-packages/metadrive/engine/base_engine.py", line 55, in __init__
    EngineCore.__init__(self, global_config)
  File "/home/lzh/miniconda3/envs/metadrive/lib/python3.9/site-packages/metadrive/engine/core/engine_core.py", line 270, in __init__
    self.pbrpipe = init(
  File "/home/lzh/miniconda3/envs/metadrive/lib/python3.9/site-packages/metadrive/third_party/simplepbr/__init__.py", line 350, in init
    return Pipeline(**kwargs)
  File "/home/lzh/miniconda3/envs/metadrive/lib/python3.9/site-packages/metadrive/third_party/simplepbr/__init__.py", line 139, in __init__
    self._recompile_pbr()
  File "/home/lzh/miniconda3/envs/metadrive/lib/python3.9/site-packages/metadrive/third_party/simplepbr/__init__.py", line 238, in _recompile_pbr
    pbr_vert_str = _load_shader_str('simplepbr.vert', pbr_defines)
  File "/home/lzh/miniconda3/envs/metadrive/lib/python3.9/site-packages/metadrive/third_party/simplepbr/__init__.py", line 57, in _load_shader_str
    with open(os.path.join(shader_dir, shaderpath + ".glsl")) as shaderfile:
FileNotFoundError: [Errno 2] 没有那个文件或目录: '/home/lzh/miniconda3/envs/metadrive/lib/python3.9/site-packages/metadrive/third_party/simplepbr/shaders/simplepbr.vert.glsl'
```
Then I found there are no shaders dir. like this:
[image]
Maybe some files were missed during installation
.

QuanyiLi, 2023-12-19:
It is a trivial bug that happens when using `pip+git` install. It is fixed now.

CarlDegio, 2023-12-21:
Test successfully. Points and lines can be drawn in the local coordinate of the car, and the colors are also rendered successfully.

closed 2023-12-21 by CarlDegio