metadriverse/metadrive #576: [bug] brake abnormal

CarlDegio, 2023-12-09:
I found this code may have a bug:

```
    def _apply_throttle_brake(self, throttle_brake):
        max_engine_force = self.config["max_engine_force"]
        max_brake_force = self.config["max_brake_force"]
        for wheel_index in range(4):
            if throttle_brake >= 0:
                self.system.setBrake(2.0, wheel_index)
                if self.speed_km_h > self.max_speed_km_h:
                    self.system.applyEngineForce(0.0, wheel_index)
                else:
                    self.system.applyEngineForce(max_engine_force * throttle_brake, wheel_index)
            else:
                if self.enable_reverse:
                    self.system.applyEngineForce(max_engine_force * throttle_brake, wheel_index)
                    self.system.setBrake(0, wheel_index)
                else:
                    self.system.applyEngineForce(0.0, wheel_index)
                    self.system.setBrake(abs(throttle_brake) * max_brake_force, wheel_index)  # cause reverse
```
in base_vehicle.py line 469.https://github.com/metadriverse/metadrive/blob/b8a66c92ac34a5f80944b41600fed08d8e3a90ee/metadrive/component/vehicle/base_vehicle.py#L469

if throttle_brake<0 and self.enable_reverse is False(default), the car will reverse due to `self.system.setBrake(abs(throttle_brake) * max_brake_force, wheel_index) `. It makes the car wheel rotate back then the car will move backward. 

I think when reversing is not allowed, the wheel rotation(setBrake) should also be set to 0. And when it is allowed, setBrake is set as rotate value.

CarlDegio, 2023-12-12:
It looks like there is still a problem with setBrake being 0 because the moving car will not slow down. 

There are two main points that I don’t quite understand.
1. `self.system.setBrake(2.0, wheel_index)` in moving forward. Looks like it has the potential to cause a slowdown.
2. setBrake and applyEngineForce work in different way. Causes forward and braking acceleration settings to be different. The braking acceleration is set to a small value, but the braking is obviously faster.

QuanyiLi, 2023-12-12:
The initial issue is not a problem. The reverse is not so clear and only happens when the speed is close to 0. I tried the following code.
```python
  self.system.applyEngineForce(0.0, wheel_index)
  force = abs(throttle_brake) * max_brake_force if self.speed > 0.025 else 0
  self.system.setBrake(force, wheel_index)
```
But it can not alleviate this issue. As it is not a serious problem, I suggest just let it be.

For the two following questions:
1. We add this to simulate a friction effect, as the default fiction is not strong enough
2. We didn't find this. All we know is just `applyEngineForce` and `setBrake` are used to move forward and backward. We didn't investigate deeply the powertrain model of the car and various parameters, as MetaDrive is not for powertrain simulation. Thus a car with an acceptable dynamic model is enough for current research. If you are interested in this part and want to tune the dynamics of the car to make it more realistic, you can find the implementation of the car physics at https://github.com/bulletphysics/bullet3/blob/6bb8d1123d8a55d407b19fd3357c724d0f5c9d3c/src/BulletDynamics/Vehicle/btRaycastVehicle.cpp#L256 Also, if you want to learn more about car dynamics simulation, you can take a look at [Project Chrono](https://projectchrono.org/), which aims at simulating the vehicle dynamics.

closed 2023-12-12 by CarlDegio