metadriverse/metadrive #579: How to have longer roads？

AntaiXie, 2023-12-12:
Hello, map="SSSSS" can make the straight path longer. How can I directly define the length of the straight path? Because I hope an episode has more timesteps

AntaiXie, 2023-12-12:
Another question I would like to ask is can the render window follow the movement of the ego vehicle?

QuanyiLi, 2023-12-12:
For the first question, if you want to precisely control the length of a road, you can build your own map with parameters filled on your own instead of the random number engine. An example is as follows
https://github.com/metadriverse/metadrive/blob/f247bdd3781790bfbd887427f396cd578cfc110f/metadrive/envs/marl_envs/marl_parking_lot.py#L144
But in general, using map="SSSSSS" is enough as the more blocks the longer the path.

For the second question, are you using topdown renderer? If so, you can cancel the `camera_position` parameter in the `env.render`. Then the camera will follow the car. Also, add `target_vehicle_heading_up=True` in `env.render` will make the ego car's heading upward. But it may lower the rendering efficiency.

AntaiXie, 2023-12-12:
Thank you for your reply. Maybe I should create my map. Because too many "S" can cause some problems

安泰
***@***.***

 

 https://github.com/metadriverse/metadrive/blob/f247bdd3781790bfbd887427f396cd578cfc110f/metadrive/envs/marl_envs/marl_parking_lot.py#L144

closed 2024-01-19 by QuanyiLi