metadriverse/metadrive #591: ImportError: cannot import name '_load_shader_str' from 'simplepbr'

axelbr, 2024-01-05:
When installing metadrive and running the examples, I got the following error:
`ImportError: cannot import name '_load_shader_str' from 'simplepbr'`

It seems that the dependency `panda3d-simplepbr` does not have this member anymore. Downgrading to a version prior to 0.11 helped to fix this error.
I am using python 3.10

QuanyiLi, 2024-01-07:
Hi,

Thanks for the feedback: ) 

Did you install it with `pip`? If it is installed from `git clone`, this should be addressed.

axelbr, 2024-01-08:
I installed it with pip. But I guess it should be addressed anyways, since it's one of the recommended ways of installation.

axelbr, 2024-01-08:
It works when I install it directly from Github.

QuanyiLi, 2024-01-08:
Indeed, we should release a new PyPI package. Thank you for reminding us!

closed 2024-01-19 by QuanyiLi