metadriverse/metadrive #621: Assets not pulled yet. Waiting for 10 seconds... forever

jakethesnake420, 2024-01-26:
Was working yesterday
 
`python -m metadrive.examples.profile_metadrive`

```
Start to profile the efficiency of MetaDrive with 1000 maps and ~4 vehicles!
[INFO] Environment: MetaDriveEnv
[INFO] MetaDrive version: 0.4.2.2
[INFO] Sensors: [lidar: Lidar(), side_detector: SideDetector(), lane_line_detector: LaneLineDetector()]
[INFO] Render Mode: none
[INFO] Horizon (Max steps per agent): None
[INFO] Another instance of this program is already running. Wait for the asset pulling finished from another program...
[INFO] Assets not pulled yet. Waiting for 10 seconds...
[INFO] Assets not pulled yet. Waiting for 10 seconds...
[INFO] Assets not pulled yet. Waiting for 10 seconds...
```
the repeats forever. I know its not my machine because github action tests that worked yesterday are failing today.

QuanyiLi, 2024-01-26:
@pengzhenghao Could it be raised due to the new lock system?

jakethesnake420, 2024-01-26:
import unittest
import filelock
import os
import time
import urllib.request
import zipfile
from unittest.mock import patch
import requests
from requests.exceptions import RequestException
import traceback

VERSION = "0.4.2.2"
class TestFileLockWithRealDownload(unittest.TestCase):
    def setUp(self):
        self.zip_lock = 'test.lock'
        self.zip_path = 'test.zip'
        self.ASSET_URL = "https://github.com/metadriverse/metadrive/releases/download/MetaDrive-{}/assets.zip".format(VERSION)  # Replace with your version
        self.TARGET_DIR = 'test_extraction_dir'

        # Create target directory if it doesn't exist
        if not os.path.exists(self.TARGET_DIR):
            os.makedirs(self.TARGET_DIR)

    def tearDown(self):
        # Cleanup downloaded and extracted files
        if os.path.exists(self.zip_path):
            os.remove(self.zip_path)
        if os.path.exists(self.zip_lock):
            os.remove(self.zip_lock)

    def test_file_lock_with_real_download_fails(self):
        lock = filelock.FileLock(self.zip_lock, timeout=1)
        try:
            with lock.acquire():
                print("Pull assets from {} to {}".format(self.ASSET_URL, self.zip_path))
                try:
                    urllib.request.urlretrieve(self.ASSET_URL, self.zip_path)
                except Exception as e:
                    print("Error during file download: {}".format(e))
                    traceback.print_exc()  # This will print the stack trace
                with zipfile.ZipFile(self.zip_path, 'r') as zip_ref:
                    zip_ref.extractall(self.TARGET_DIR)
                print("Successfully downloaded and extracted assets")

        except filelock.Timeout:
            print("Another instance of this program is already running...")
            while os.path.exists(self.zip_lock):
                print("Assets not pulled yet. Waiting for 10 seconds...")
                time.sleep(10)

            print("Assets are now available.")
            
    def test_file_lock_with_real_download_works(self):
        lock = filelock.FileLock(self.zip_lock, timeout=1)
        try:
            with lock.acquire():
                # Simulate logger
                print("Pull assets from {} to {}".format(self.ASSET_URL, self.zip_path))

                # Implementing retry mechanism
                max_retries = 3
                for _ in range(max_retries):
                    try:
                        response = requests.get(self.ASSET_URL, stream=True, timeout=10)
                        with open(self.zip_path, 'wb') as f:
                            for chunk in response.iter_content(chunk_size=8192):
                                f.write(chunk)
                        break
                    except RequestException as e:
                        print(f"Failed to download file: {e}")
                        time.sleep(5)  # wait before retrying
                else:
                    raise Exception("Failed to download file after multiple attempts")

                with zipfile.ZipFile(self.zip_path, 'r') as zip_ref:
                    zip_ref.extractall(self.TARGET_DIR)
                print("Successfully downloaded and extracted assets")

        except filelock.Timeout:
            print("Another instance of this program is already running. Wait for the asset pulling finished from another program...")
            while os.path.exists(self.zip_lock):
                print("Assets not pulled yet. Waiting for 10 seconds...")
                time.sleep(10)

            print("Assets are now available.")

if __name__ == '__main__':
    unittest.main()

pengzhenghao, 2024-01-26:
Hi did you install metadrive via git clone or pip install metadrive-simulator? It should be fixed by #607 

Thanks for reminding me maybe I should create a new release so the user install via pip install can avoid this issue.

pengzhenghao, 2024-01-26:
Now I've made a new release #622 

Could you try update MD and see if that works?

pengzhenghao, 2024-01-26:
[image]

@jakethesnake420 I ran your code it works well with me

jakethesnake420, 2024-01-27:
> @jakethesnake420 I ran your code it works well with me

Interesting, I get the `Exception: Lost Connection` when i run test_file_lock_with_real_download_fails(). It must be some dependency issue in my openpilot enviroment on github actions and my local PC. I will test your latest fixes.

jakethesnake420, 2024-01-27:
Tried the latest pip version 0.4.2.3 . 

Just doing `pip install metadrive-simulator` did not fix the issue. I had to do `pip-autoremove metadrive-simulator` and then `pip install metadrive-simulator --no-cache-dir` to make it work.

jakethesnake420, 2024-01-27:
So actually when I install it in openpilot, i still get the same `Exception: Lost Connection ` even after updating poetry and pip-autoremove metadrive-simulator and then pip install metadrive-simulator --no-cache-dir

jakethesnake420, 2024-01-27:
I made a script to recursively check all dependency  versions:

```
import subprocess
import json

def get_package_info(package):
    """ Get the version and dependencies of a package. """
    try:
        print(f"Checking {package}...")
        package_info = subprocess.check_output(["pip", "show", package], text=True)
    except subprocess.CalledProcessError:
        # Package not found
        print(f"Package {package} not found.")
        return None, []

    version = None
    dependencies = []

    for line in package_info.splitlines():
        if line.startswith("Version:"):
            version = line.split("Version:")[1].strip()
        elif line.startswith("Requires"):
            deps = line.split("Requires:")[1].strip()
            if deps:
                dependencies = deps.split(", ")

    return version, dependencies

def get_dependency_tree(package, checked_packages=None):
    """ Recursively get the dependency tree of a package."""
    if checked_packages is None:
        checked_packages = {}

    if package in checked_packages:
        # Package already processed
        print(f"{package} has been already processed.")
        return {package: {'version': checked_packages[package], 'dependencies': {}}}

    version, dependencies = get_package_info(package)
    if version is None:
        return {package: "not found"}

    checked_packages[package] = version
    tree = {'version': version, 'dependencies': {}}

    for dep in dependencies:
        sub_deps = get_dependency_tree(dep, checked_packages)
        if sub_deps:
            tree['dependencies'][dep] = sub_deps

    return tree

dependency_tree = get_dependency_tree("metadrive-simulator")
print(json.dumps(dependency_tree, indent=4))

```

Here is the full dependency tree for my installation so you can compare:

```
{
    "version": "0.4.2.3",
    "dependencies": {
        "filelock": {
            "version": "3.13.1",
            "dependencies": {}
        },
        "geopandas": {
            "version": "0.14.2",
[... 359 lines cut ...]
        "yapf": {
            "version": "0.40.2",
            "dependencies": {
                "importlib-metadata": {
                    "version": "7.0.1",
                    "dependencies": {
                        "zipp": {
                            "version": "3.17.0",
                            "dependencies": {}
                        }
                    }
                },
                "platformdirs": {
                    "version": "4.1.0",
                    "dependencies": {}
                },
                "tomli": {
                    "version": "2.0.1",
                    "dependencies": {}
                }
            }
        }
    }
}
```

closed 2024-12-15 by pengzhenghao