metadriverse/metadrive #629: Invalid image observations when using experimental render pipeline

fredyshox, 2024-01-30:
I was trying to use Metadrive 4.2.3 with experimental render pipeline, with the intent of rendering more advanced graphics and gaining control the time of the day using `daytime` attribute.

Im using the code from `drive_in_single_agent_env` example (just with `render_pipeline=True`, rest is exactly the same) and while the frames rendered by the MainCamera look correct, the frames from the RGBCamera have distorted geometry and lack skybox or lightning (see the attached image).

When setting debug=True, I can see constantly repeated line:
```
:display:gsg:glgsg(error): GL_INVALID_VALUE error generated. Invalid image format.
```

Is there a way to make it work? Or is it not ready to be used yet? 

Main Camera
[image]

RGB Camera
[image]

QuanyiLi, 2024-01-31:
It is as expected. The RGBCamera in this script is a newly created one and the render_pipeline's effect doesn't apply to it. Actually, the compatibility between the render_pipeline and all sensors is not fully tested yet. Thus, the only RGBSensor you can use with the render pipeline is the main camera. You can do this by replacing the code snippet with the following one in `drive_in_single_agent.py`.

```python
    if args.observation == "rgb_camera":
        config.update(
            dict(
                render_pipeline=True,
                image_observation=True,
                sensors=dict(main_camera=()),
                camera_dist=-1,
                camera_pitch=15,
                camera_height=1,
                window_size=(400, 300),
                interface_panel=["dashboard"],
                vehicle_config=dict(show_lidar=True,
                                    show_navi_mark=False,
                                    show_line_to_navi_mark=False,
                                    image_source="main_camera"),
                show_interface=False,
                show_fps=False,
            )
        )
```   

It gives you a front-view camera as if it is mounted on the car. In each frame, this camera can render scenes from different views as if there are multiple RGBCamera in the scene. This can be achieved by doing this: https://metadrive-simulator.readthedocs.io/en/latest/obs.html#customization-multiview

pengzhenghao, 2024-03-02:
link to this: 

https://github.com/metadriverse/metadrive/issues/652#issuecomment-1974409193