metadriverse/metadrive #632: Headless rendering: tonemap_quad is None

alexander-toch, 2024-02-05:
Hi, I'm trying to perform headless rendering on a Tesla T4 with Metadrive version: 0.4.2.3.

Following commands work:
- `python3 -m metadrive.examples.verify_headless_installation --camera rgb`
- `python3 -m metadrive.examples.verify_headless_installation --camera rgb --cuda`
-  `python3 -m metadrive.examples.verify_headless_installation --camera depth --cuda`

But following command and my custom example with `use_render=False`, `image_observation=True` and `image_on_cuda=True`) fail: `python verify_image_observation.py --cuda`

I enabled following settings for debugging:
```
loadPrcFileData("", "gl-debug true")
loadPrcFileData("", "notify-level-egldisplay spam")
loadPrcFileData("", "notify-level-display debug")
```

Trace:
```
[INFO] Environment: MetaDriveEnv
[INFO] MetaDrive version: 0.4.2.3
[INFO] Sensors: [lidar: Lidar(), side_detector: SideDetector(), lane_line_detector: LaneLineDetector(), main_camera: MainCamera(1200, 900)]
[INFO] Render Mode: offscreen
[INFO] Horizon (Max steps per agent): None
Use main_camera with resolution (900, 1200)
[INFO] Assets version: 0.4.2.3
:display:egldisplay(debug): Supported EGL client extensions:
  EGL_EXT_platform_base
  EGL_EXT_device_base
  EGL_KHR_client_get_all_proc_addresses
  EGL_EXT_client_extensions
  EGL_KHR_debug
  EGL_KHR_platform_x11
  EGL_EXT_platform_x11
  EGL_EXT_platform_device
:display:egldisplay(debug): Successfully initialized EGL display, got version 1.5
--- TRUNCATED---
[INFO] Known Pipes: glxGraphicsPipe
Traceback (most recent call last):
  File "~/miniconda3/envs/ld3.9/lib/python3.9/site-packages/metadrive/examples/verify_image_observation.py", line 95, in <module>
    _test_rgb_camera_as_obs(args.render, image_on_cuda=args.cuda, debug=args.debug, camera=args.camera)
  File "~/miniconda3/envs/ld3.9/lib/python3.9/site-packages/metadrive/examples/verify_image_observation.py", line 53, in _test_rgb_camera_as_obs
    o, i = env.reset()
  File "~/miniconda3/envs/ld3.9/lib/python3.9/site-packages/metadrive/envs/base_env.py", line 519, in reset
    self.lazy_init()  # it only works the first time when reset() is called to avoid the error when render
  File "~/miniconda3/envs/ld3.9/lib/python3.9/site-packages/metadrive/envs/base_env.py", line 406, in lazy_init
    initialize_engine(self.config)
  File "~/miniconda3/envs/ld3.9/lib/python3.9/site-packages/metadrive/engine/engine_utils.py", line 12, in initialize_engine
    cls.singleton = cls(env_global_config)
  File "~/miniconda3/envs/ld3.9/lib/python3.9/site-packages/metadrive/engine/base_engine.py", line 55, in __init__
    EngineCore.__init__(self, global_config)
  File "~/miniconda3/envs/ld3.9/lib/python3.9/site-packages/metadrive/engine/core/engine_core.py", line 284, in __init__
    self.pbrpipe = init(
  File "~/miniconda3/envs/ld3.9/lib/python3.9/site-packages/metadrive/third_party/simplepbr/__init__.py", line 350, in init
    return Pipeline(**kwargs)
  File "~/miniconda3/envs/ld3.9/lib/python3.9/site-packages/metadrive/third_party/simplepbr/__init__.py", line 142, in __init__
    self._setup_tonemapping()
  File "~/miniconda3/envs/ld3.9/lib/python3.9/site-packages/metadrive/third_party/simplepbr/__init__.py", line 287, in _setup_tonemapping
    self.tonemap_quad.set_shader(tonemap_shader)
AttributeError: 'NoneType' object has no attribute 'set_shader'
```

Could the error be related with `Known Pipes: glxGraphicsPipe`? Is there any way to change to [eglGraphicsPipe](https://www.panda3d.org/reference/cxx/classeglGraphicsPipe.html) (which should be supported?).

Any help would be very appreciated!

QuanyiLi, 2024-02-06:
Could you try adding `vglrun` before `python xyz`?

We have experienced a similar thing on another server. It works by doing so.

QuanyiLi, 2024-02-06:
Another interesting thing to notice is that `verify_headless_installation` firstly runs an environment with {"use_render": False, "image_observation": False}), and then runs a new environment with {image_observation=True} where the rendered images are generated.

One of our users said that without running the first physics-only environment, the second environment will fail to capture figures due to the same error you are experiencing. Actually, I haven't figured it out yet. So a potential workaround for this is initializing a physics-only environment and then closing it before initializing the environment you are actually working on.

alexander-toch, 2024-02-06:
I just reused the physics-only environment from the `verify_headless_installation` script, launched it before the second environment, and it just worked!

Also tried with `vglrun -d /dev/dri/card0 python ...` but that failed with `:ShowBase(warning): Unable to open 'offscreen' window.` and `Exception: Could not open window.`.

However, the headless rendering finally works, thank you!

QuanyiLi, 2024-02-06:
Great!

fredyshox, 2024-02-07:
I've also experienced this bug in headless environment, and came out with the same workaround. Then I did some investigation on my own (didn't end up solving it), from which I'll share my findings.

To reproduce the bug on regular PC with desktop environment (i'm using ubuntu 20.04), just `unset DISPLAY` in your shell before running metadrive. 

First I noticed that, you don't need to initialize physics environment; calling `initialize_engine(config)` and then `close_engine()` is sufficient for the next environment to work. So, before calling `_test_rgb_camera_as_obs`:

```py
    from metadrive.envs.base_env import BASE_DEFAULT_CONFIG
    from metadrive.engine.engine_utils import initialize_engine, close_engine
    config = BASE_DEFAULT_CONFIG.copy()
    config["debug"] = True
    initialize_engine(config)
    close_engine()
```

Lets refer to the vanilla `verify_image_observation.py` as A, and modified version with cold-run of the engine init as B.
Looking at debug logs from panda3d display module (attached below), which can be enabled using:
```py
    from panda3d.core import loadPrcFileData
    loadPrcFileData("", "notify-level-display debug")
    loadPrcFileData("", "notify-level-x11display debug")
```
I can see that in both cases panda3d is using `eglGraphicsPipe`, but in case A the gsg gets OpenGL ES context (with some ancient version):
```
:display:gsg:glgsg(debug): GL_VENDOR = NVIDIA Corporation
:display:gsg:glgsg(debug): GL_RENDERER = NVIDIA GeForce RTX 3080/PCIe/SSE2
:display:gsg:glgsg(debug): GL_VERSION = OpenGL ES 1.1 NVIDIA 545.23.08
:display:gsg:glgsg(debug): GL_VERSION decoded to: 1.1
:display:gsg:glgsg(debug): EGL_VERSION = 1.5
:display:gsg:glgsg(debug): HAS EXT GL_ARB_shading_language_100 0
:display:gsg:glgsg(debug): Detected GLSL version: 0.0
```
Despite the fact that the panda3d from pip is compiled without GLES support, and that it binds `EGL_OPENGL_API`. In case B I see following output, which means it gets the modern GL context as expected:
```
:display:gsg:glgsg(debug): GL_VENDOR = NVIDIA Corporation
:display:gsg:glgsg(debug): GL_RENDERER = NVIDIA GeForce RTX 3080/PCIe/SSE2
:display:gsg:glgsg(debug): GL_VERSION = 4.6.0 NVIDIA 545.23.08
:display:gsg:glgsg(debug): GL_VERSION decoded to: 4.6
:display:gsg:glgsg(debug): EGL_VERSION = 1.5
:display:gsg:glgsg(debug): GL_SHADING_LANGUAGE_VERSION = 4.60 NVIDIA
:display:gsg:glgsg(debug): Detected GLSL version: 4.60
```

This seem to propagate failures later on, since the "invalid" context does not support certain features (like `textures_power_2`), which leads to `tonemap_quad` being None.

I'm not exactly sure, but I feel like this could be behind it. I tried to replicate that on my own, by writing simple EGL program, imitating the initialization steps of `eglGraphicsPipe`/`eglGraphicsStateGuardian`, but couldn't reproduce the issue. 

Hope it helps somehow!

[case-a.log](https://github.com/metadriverse/metadrive/files/14188773/case-a.log)
[case-b.log](https://github.com/metadriverse/metadrive/files/14188777/case-b.log)
[context-info.txt](https://github.com/metadriverse/metadrive/files/14188800/context-info.txt)

pengzhenghao, 2024-02-13:
Thanks for the investigation! @fredyshox 

I am reopening this PR as we should proactively address this issue in our side. I would try to make a PR to fix this.

My idea is to manually add a script like this:

```python
    from metadrive.envs.base_env import BASE_DEFAULT_CONFIG
    from metadrive.engine.engine_utils import initialize_engine, close_engine
    config = BASE_DEFAULT_CONFIG.copy()
    config["debug"] = True
    initialize_engine(config)
    close_engine()
```

closed 2024-02-20 by pengzhenghao

Hanna-Li, 2025-02-06:
In my server(ubuntu1804) when I run [python local_test_headless_rendering.py], also get the ERROR [AttributeError: 'NoneType' object has no attribute 'set_shader']. I use the main branch code at 2025/1/20.
In my local ubuntu1804 cloud desktop is OK.

https://github.com/metadriverse/metadrive/issues/632#issuecomment-1942744648