metadriverse/metadrive #642: Unable to run in ubuntu 22.04

abastola0, 2024-02-13:
I am experiencing issues running the basic implementations from provided example files. I'm stuck at 1.0fps for some reason. I see the gpu usage spike sometimes but then it comes down to 0% and the simulator freezes for some reason. I tried reinstalling everything and following everything in the installation steps but still am not able to get it to work.

[image]

QuanyiLi, 2024-02-13:
Some users experience the similar thing before: https://github.com/metadriverse/metadrive/issues/154 . There must be something wrong with the graphics card driver or OpenGL. Could you set `debug_panda3d=True` and `debug=True` in the config file when launching the simulation? The debug message may contain something helpful.

abastola0, 2024-02-15:
I reinstalled everything from X11 setup and nvidia drivers and now it works fine when i run it natively with native monitor. I tried using virtualgl with X11, with XPra and now running over TurboVNC but still it does not work. I'm able to run other graphics applications over vnc though. But for metadrive i'm stuck at 0.2fps or something like that. I could not find a better way to remote access it. Default RDP does not work as well.

abastola0, 2024-02-16:
I had a really hard time figuring this issue but this was related with .xconf file. When trying to run headless and then also run glx applications can be very complicated. Thus both x11 and turbovnc both with vgl worked well in my case. x11 was fine as i had a gigabit ethernet connected to the server. Here are the changes that I made to the .xconf file. 
In Section "Monitor":     
          ......
          #Option         "DPMS"
          Option         "dpms" "false"

In Section "Device"
          ......
          Option "AllowEmptyInitialConfiguration" "true"
 
and finally added a new section "ServerFlags"

Section "ServerFlags"
 	Option "BlankTime" "0"
	Option "StandbyTime" "0"
	Option "SuspendTime" "0"
	Option "OffTime" "0"
	Option "dpms" "false"
EndSection

Here is my full xorg.conf file:
# nvidia-xconfig: X configuration file generated by nvidia-xconfig
# nvidia-xconfig:  version 535.154.05

Section "ServerLayout"
    Identifier     "Layout0"
    Screen      0  "Screen0"
    InputDevice    "Keyboard0" "CoreKeyboard"
    InputDevice    "Mouse0" "CorePointer"
EndSection

Section "Files"
EndSection

Section "InputDevice"
    # generated from default
    Identifier     "Mouse0"
    Driver         "mouse"
    Option         "Protocol" "auto"
    Option         "Device" "/dev/psaux"
    Option         "Emulate3Buttons" "no"
    Option         "ZAxisMapping" "4 5"
EndSection

Section "InputDevice"
    # generated from default
    Identifier     "Keyboard0"
    Driver         "kbd"
EndSection

Section "Monitor"
    Identifier     "Monitor0"
    VendorName     "Unknown"
    ModelName      "Unknown"
    #Option         "DPMS"
    Option         "dpms" "false"
EndSection

Section "Device"
    Identifier     "Device0"
    Driver         "nvidia"
    VendorName     "NVIDIA Corporation"
    Option "AllowEmptyInitialConfiguration" "true"
EndSection

Section "Screen"
    Identifier     "Screen0"
    Device         "Device0"
    Monitor        "Monitor0"
    DefaultDepth    24
    SubSection     "Display"
        Depth       24
    EndSubSection
EndSection

Section "ServerFlags"
	Option "BlankTime" "0"
	Option "StandbyTime" "0"
	Option "SuspendTime" "0"
	Option "OffTime" "0"
	Option "dpms" "false"
EndSection

Please find the reference here:
https://groups.google.com/g/virtualgl-users/c/d8k1ujwKJXo/m/79QYFBf8AAAJ

closed 2024-02-16 by abastola0