metadriverse/metadrive #643: Document broken: https://metadrive-simulator.readthedocs.io/en/latest/sensors.html

pengzhenghao, 2024-02-13:
https://metadrive-simulator.readthedocs.io/en/latest/sensors.html

[image]

QuanyiLi, 2024-02-13:
I left this here on purpose. I wanna know if the readthedoc sever can render 3D scenes and this bug occurs. It is likely we can fix it. I will do this when I am free, so we can generate 3D content in the doc compile time.

pengzhenghao, 2024-02-13:
It can be solved by #644

pengzhenghao, 2024-02-20:
How I make this notebook compiled again? The #644 should already fix this?

QuanyiLi, 2024-02-21:
Firstly, you can build the doc locally, to see if this cell runs and outputs the expected results. If the cell does not run, you have to enable the running in compiling time for that `.ipynb` according to the readme in the documentation. If the cell runs and produces correct results, that means it is still not fixed by #644. As when you merge #644, the doc should be compiled again...

PR #669 (2024-03-02): Add more examples for semantic observation/sensor in docs
## What changes do you make in this PR?

* Please describe why you create this PR

close #650 by providing better example in documentation.

close #643

## Checklist

* [x] I have merged the latest main branch into current branch.
* [x] I have run `bash scripts/format.sh` before merging.
* Please use "squash and merge" mode.
closed 2024-03-02 by pengzhenghao