metadriverse/metadrive #654: Cannot find "/usr/lib/dri/swrast_dri.so"

rgu01, 2024-02-20:
OS: Linux Ubuntu 22.04.3 LTS
Processor: Intel® Core™ i5-8250U CPU @ 1.60GHz × 4
Graphics: llvmpipe (LLVM 15.0.7, 256 bits)
Virtual machine: Oracle VirtualBox Version 7.0.10 r158379
Issue: After strictly following the installation steps, running "python -m metadrive.examples.drive_in_single_agent_env" generates the following the errors:

[INFO] Environment: MetaDriveEnv
[INFO] MetaDrive version: 0.4.2.3
[INFO] Sensors: [lidar: Lidar(), side_detector: SideDetector(), lane_line_detector: LaneLineDetector(), main_camera: MainCamera(1200, 900), dashboard: DashBoard()]
[INFO] Render Mode: onscreen
[INFO] Horizon (Max steps per agent): None
[INFO] Assets version: 0.4.2.3
Xlib:  extension "XFree86-DGA" missing on display ":0".
**libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so**: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)

Running the following command can obtain the path of swrast_dri.so: find / -name swrast_dri.so 2>/dev/null
Result: **/usr/lib/x86_64-linux-gnu/dri/swrast_dri.so**

Apparently, the path used by MetaDrive is wrong.

QuanyiLi, 2024-02-21:
Do you try creating a soft link for swrast_dri.so?

rgu01, 2024-02-21:
> Do you try creating a soft link for swrast_dri.so?

No, I did not try any fix. I thought it was a problem with the tool as I just followed the instructions in a new Ubuntu VM and did nothing complex. Do you mean add "/usr/lib/x86_64-linux-gnu/dri/swrast_dri.so" in the PATH variable?

QuanyiLi, 2024-02-21:
No, I mean you can do `ln -s  /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so /usr/lib/dri/swrast_dri.so`, which will create a soft copy for the missing `.so`

QuanyiLi, 2024-02-21:
We didn't run MetaDrive in a Virtual machine before... I think the virtual machine has poor support for GPU, so the rendering could be slow. If you are playing Ubuntu on windows platform, maybe you can try `WSL2`, which provides native support for linux on windows.

QuanyiLi, 2024-02-21:
Do you install from `pip`? The PR I merged just now should solve this if you are installing from github... Also, Mac with M chips can not do 3D rendering as the OpenGL is deprecated on MacOS

rgu01, 2024-02-22:
Yes, I pulled the latest code, but I did not run "pip install" again. Now the problem is an exception in "Showcase.py", cannot open the window. The detail is as below:

After running "python -m metadrive.examples.drive_in_single_agent_env", I got
[INFO] Environment: MetaDriveEnv
[INFO] MetaDrive version: 0.4.2.3
[INFO] Sensors: [lidar: Lidar(), side_detector: SideDetector(), lane_line_detector: LaneLineDetector(), main_camera: MainCamera(1200, 900), dashboard: DashBoard()]
[INFO] Render Mode: onscreen
[INFO] Horizon (Max steps per agent): None
[INFO] Assets version: 0.4.2.3
:ShowBase(warning): Unable to open 'onscreen' window.
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/Users/rgu01/Library/CloudStorage/OneDrive-Mälardalensuniversitet/Documents/Programs/scenarionet/metadrive/metadrive/examples/drive_in_single_agent_env.py", line 51, in <module>
    o, _ = env.reset(seed=21)
           ^^^^^^^^^^^^^^^^^^
  File "/Users/rgu01/Library/CloudStorage/OneDrive-Mälardalensuniversitet/Documents/Programs/scenarionet/metadrive/metadrive/envs/base_env.py", line 517, in reset
    self.lazy_init()  # it only works the first time when reset() is called to avoid the error when render
    ^^^^^^^^^^^^^^^^
  File "/Users/rgu01/Library/CloudStorage/OneDrive-Mälardalensuniversitet/Documents/Programs/scenarionet/metadrive/metadrive/envs/base_env.py", line 409, in lazy_init
    initialize_engine(self.config)
  File "/Users/rgu01/Library/CloudStorage/OneDrive-Mälardalensuniversitet/Documents/Programs/scenarionet/metadrive/metadrive/engine/engine_utils.py", line 38, in initialize_engine
    cls.singleton = cls(env_global_config)
                    ^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/rgu01/Library/CloudStorage/OneDrive-Mälardalensuniversitet/Documents/Programs/scenarionet/metadrive/metadrive/engine/base_engine.py", line 55, in __init__
    EngineCore.__init__(self, global_config)
  File "/Users/rgu01/Library/CloudStorage/OneDrive-Mälardalensuniversitet/Documents/Programs/scenarionet/metadrive/metadrive/engine/core/engine_core.py", line 194, in __init__
    super(EngineCore, self).__init__(windowType=self.mode)
  File "/Users/rgu01/miniconda3/envs/metadrive/lib/python3.11/site-packages/direct/showbase/ShowBase.py", line 341, in __init__
    self.openDefaultWindow(startDirect = False, props=props)
  File "/Users/rgu01/miniconda3/envs/metadrive/lib/python3.11/site-packages/direct/showbase/ShowBase.py", line 1026, in openDefaultWindow
    self.openMainWindow(*args, **kw)
  File "/Users/rgu01/miniconda3/envs/metadrive/lib/python3.11/site-packages/direct/showbase/ShowBase.py", line 1061, in openMainWindow
    self.openWindow(*args, **kw)
  File "/Users/rgu01/miniconda3/envs/metadrive/lib/python3.11/site-packages/direct/showbase/ShowBase.py", line 806, in openWindow
    raise Exception('Could not open window.')
Exception: Could not open window.

Could it be a MacBook Pro problem?

QuanyiLi, 2024-02-22:
Yes. As I said before, Mac with M chips can not do 3D rendering for MetaDrive as OpenGL is deprecated on MacOS

rgu01, 2024-02-22:
Thanks. Is there a way to disable the 3D rendering and keep using the rest of MetaDrive? Now, I simply want to run some examples but have not managed once.

QuanyiLi, 2024-02-22:
Of course. You can still use the2D renderer. All examples in our documentation are generated with 2D renderer: https://metadrive-simulator.readthedocs.io/ All examples can run on both colab and local machines. They also show you how to use this 2D renderer as well.

rgu01, 2024-02-23:
> Of course. You can still use the2D renderer. All examples in our documentation are generated with 2D renderer: https://metadrive-simulator.readthedocs.io/ All examples can run on both colab and local machines. They also show you how to use this 2D renderer as well.

Perfect. I meant turning on/off the 3D render in your examples, such as python -m metadrive.examples.drive_in_single_agent_env. Anyway, I am running the 2D codes locally on my own laptop. Thanks for the help.

closed 2024-02-23 by rgu01