metadriverse/metadrive #664: git install issue

CarlDegio, 2024-02-26:
clone metadrive git repo or fetch will lead to an error:

```
remote: Enumerating objects: 791, done.
remote: Counting objects: 100% (750/750), done.
remote: Compressing objects: 100% (416/416), done.
error: RPC 失败。curl 92 HTTP/2 stream 0 was not closed cleanly: CANCEL (err 8)
error: 预期仍然需要 7614 个字节的正文
fetch-pack: unexpected disconnect while reading sideband packet
fatal: 过早的文件结束符（EOF）
fatal: fetch-pack：无效的 index-pack 输出
```

This affects installing pip via git (pip install git+), `git config --global http.postBuffer 20000000` will not make pip valid.

I'm not sure if the repo is too big causing this problem?

QuanyiLi, 2024-02-28:
It works from my side. And we have a unit test for checking this and it works fine as well. So I guess there might be something wrong on your side.

AIDefender, 2024-03-12:
It might be related to the poor network connection to github.

neugzy, 2024-07-11:
Most likely it's a network issue. You can try several times at different times, and I've also encountered this problem.
大概率是网络问题，可以在不同时间段多尝试几次，我也遇到了这个问题。

closed 2024-12-07 by CarlDegio