metadriverse/metadrive #666: RGBCamera is not deterministic

fredyshox, 2024-02-28:
Images captured by RGBCamera sensor in the same setting are not always the same. Slight differences in pixel values are spotted from time to time, when using MetaDriveEnv on a simple PG map without any kind of randomization and pre-seeding. This hurts scenario reproducibility, if your driving policy is using images.

Example script:
```python
import random
import numpy as np
from metadrive import MetaDriveEnv
from metadrive.component.map.base_map import BaseMap
from metadrive.component.map.pg_map import MapGenerateMethod
from metadrive.component.sensors.rgb_camera import RGBCamera
import cv2

random.seed(0)
np.random.seed(0)

map_config={
        BaseMap.GENERATE_TYPE: MapGenerateMethod.BIG_BLOCK_SEQUENCE,
        BaseMap.GENERATE_CONFIG: "SOT",
        BaseMap.LANE_WIDTH: 3.5,
        BaseMap.LANE_NUM: 3,
        "exit_length": 50,
    }

config = dict(
        use_render=False,
        manual_control=False,
        traffic_density=0.0, #10, #0.15,
        traffic_mode="respawn",
        num_scenarios=1,
        random_agent_model=False,
        random_lane_width=False,
        random_lane_num=False,
        random_traffic=False,
        decision_repeat=1,
        start_seed=0,
        force_destroy=True,
        map_config=map_config, # set it here
        random_spawn_lane_index=False,
        success_reward=100.,
        image_observation=True,
        norm_pixel=False,
        stack_size=1,
        sensors={"rgb_camera": (RGBCamera, 1280, 720)}
    )
env = MetaDriveEnv(config)
env.seed(0)

prev_img = None
for i in range(20):
  print("step", i)
  obs, info = env.reset(seed=0)
  img = obs["image"][:, :, :, -1]
  # cv2.imwrite(f"img_step_{i}.png", img)
  if prev_img is not None:
    diff = np.where(img != prev_img)
    assert np.all(img == prev_img), "Not equal at step {}, diff len {}".format(i, len(diff[0]))

  prev_img = img
```

QuanyiLi, 2024-02-28:
What if setting `multi_thread_render=False`? Multi-thread rendering may produce inconsistent results I guess.

fredyshox, 2024-02-28:
It still happens with multi-thread rendering disabled

QuanyiLi, 2024-02-28:
My bad. It seems you are running on the same machine. I have no idea about this now. But before we move forward on identifying potential bugs on MetaDrive, could you create a discussion in the Panda3D forum? Just ask for the reproducibility of the panda3d engine.