metadriverse/metadrive #671: Set max curve radius in BIG_BLOCK_SEQUENCE

mimi89999, 2024-03-02:
Hello,
I noticed that the curves generated by `BIG_BLOCK_SEQUENCE` are very tight. Would it be possible to set the maximum curve radius?

QuanyiLi, 2024-03-02:
We didn't expose that API. The `Curve` block has a class parameter `PARAMETER_SPACE` specifying which space the parameters should be sampled. You can check its configuration at `metadrive/component/pg_space/BlockParameterSpace`.

mimi89999, 2024-03-02:
Is there any better way to generate a map then? Neither the `MapGenerateMethod.PG_MAP_FILE` type nor the map json file aren't really documented.

QuanyiLi, 2024-03-02:
Ideally, there should be something like a map editor. But we don't have it yet even if our backend supporters various map generation. The most user-friendly APIs are the `ScenarioMap` and `ScenarioBlock` which allow constructing a map from a list of map features including lane center lines/solid lines/dashed lines/crosswalks/sidewalks and so on. It has been used to construct the Waymo Map, nuScenes map, SUMO map, and OpenDrive map before. So it has a good compatibility. Or you can refer to the multi-agent environments where the maps are generated by combining several blocks manually.