metadriverse/metadrive #672: drive_in_real_env example fails with Waymo dataset

mimi89999, 2024-03-02:
```
$ python3 -m metadrive.examples.drive_in_real_env --waymo
Keyboard Shortcuts:
  W: Acceleration
  S: Braking
  A: Moving Left
  D: Moving Right
  R: Reset the Environment
  H: Help Message
  F: Switch FPS to unlimited / realtime
  Q: Third-person View Camera
  B: Top-down View Camera (control: WASD-=)
  +: Lift Camera
  -: Lower Camera
  Mouse click: move camera (top-down view)
  Esc: Quit

[INFO] Environment: ScenarioEnv
[INFO] MetaDrive version: 0.4.2.3
[INFO] Sensors: [lidar: Lidar(), side_detector: SideDetector(), lane_line_detector: LaneLineDetector(), main_camera: MainCamera(1200, 900), dashboard: DashBoard()]
[INFO] Render Mode: onscreen
[INFO] Horizon (Max steps per agent): None
[INFO] Assets version: 0.4.2.3
Xlib:  extension "XFree86-DGA" missing on display ":0".
[INFO] Known Pipes: glxGraphicsPipe
[WARNING] Since your screen is too small (1536, 864), we resize the window to (1036, 777). (engine_core.py:212)
[INFO] Start Scenario Index: 0, Num Scenarios : 3
[WARNING] No map is found in map region, size: [512, 512], center: [0, 0] (terrain.py:116)
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/michel/git/openpilot/.venv/lib/python3.11/site-packages/metadrive/examples/drive_in_real_env.py", line 41, in <module>
    o, _ = env.reset()
           ^^^^^^^^^^^
  File "/home/michel/git/openpilot/.venv/lib/python3.11/site-packages/metadrive/envs/base_env.py", line 528, in reset
    self.engine.reset()
  File "/home/michel/git/openpilot/.venv/lib/python3.11/site-packages/metadrive/engine/base_engine.py", line 374, in reset
    self.terrain.reset(center_p)
  File "/home/michel/git/openpilot/.venv/lib/python3.11/site-packages/metadrive/engine/core/terrain.py", line 141, in reset
    semantics = self.get_terrain_semantics(center_point)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/michel/git/openpilot/.venv/lib/python3.11/site-packages/metadrive/engine/core/terrain.py", line 591, in get_terrain_semantics
    semantics = self.engine.current_map.get_semantic_map(
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/michel/git/openpilot/.venv/lib/python3.11/site-packages/metadrive/component/map/base_map.py", line 255, in get_semantic_map
    points = [
             ^
  File "/home/michel/git/openpilot/.venv/lib/python3.11/site-packages/metadrive/component/map/base_map.py", line 259, in <listcomp>
    ] for x, y in polygon
          ^^^^
ValueError: too many values to unpack (expected 2)
```

QuanyiLi, 2024-03-02:
Hi, Michel

Do you install from pip? If you installed it from the latest GitHub, this should be fixed already.

mimi89999, 2024-03-02:
Thanks for your reply. I used the latest version from pip. Will check with the latest version in the repo.