metadriverse/metadrive #679: Memory leak during training with rllib probably because "Your env doesn't have a .spec.max_episode_steps attribute."

chrisgao99, 2024-03-18:
Hello, Thank you for the great simulation environment.

Recently, I've been trying to use rllib to train an agent with image obs. However, RAM usage consistently increases over training, eventually consuming all available memory (756GB) around the 20th epoch. And I received the warning below:
```
WARNING env.py:162 -- Your env doesn't have a .spec.max_episode_steps attribute. Your horizon will default to infinity, and your environment will not be reset.
```
So I guess rllib failed to close the env after collecting trajectories. Though I have already set horizon=800 in both metadrive env config and RLlib PPO config, the warning persists, along with the memory issue. I wonder if the memory issue is truly caused by the env not being closed. If so, how can i properly close the env during training? If not, is there other methods I can try to resolve this memory issue.

Here is most of my code including env_config and train_confing:
```
frame_size = (320, 240)
env_config=dict(
                map_config={BaseMap.GENERATE_TYPE: MapGenerateMethod.BIG_BLOCK_NUM,
                    BaseMap.GENERATE_CONFIG: 3},
                image_observation=True, 
                use_render=False,
                image_on_cuda=False,     
                sensors={
                    "rgb_camera": (RGBCamera, *frame_size), #w,h,c
                },
                vehicle_config={
                    "image_source": "rgb_camera",
                },
                stack_size = 1,       
                discrete_action=False, 
                horizon=800,
                random_spawn_lane_index=False,
                num_scenarios=1000,
                start_seed=0,
                traffic_density=0.2,
                accident_prob=0.2,
                log_level=50)
def env_creator3(env_config):
    env = MetaDriveEnv({"use_render": False, "image_observation": False,'horizon':10})
    env.reset()
    env.close() 
    print("The bullet env is open and closed")
    return MyWrapper(MetaDriveEnv(env_config))

if __name__ == "__main__":
    register_env("metadrive", env_creator3)
    ModelCatalog.register_custom_model("my_model", CustomModel)

    train_config = dict(
        env = "metadrive",
        env_config = env_config,
        model = {
            "custom_model": "my_model",
        },
        # gpu
        _disable_preprocessor_api = True,
        num_learner_workers = 4,
        num_gpus_per_learner_worker = 1,
        num_gpus = 4,  
        num_gpus_per_worker = 1,
        horizon = 1000,
        framework = "torch",
        # cpu
        num_rollout_workers = 4, 
        num_cpus_per_worker = 8, 
        num_cpus_for_local_worker = 4,
        num_envs_per_worker = 8,
        num_cpus_per_learner_worker = 1,
        remote_worker_envs = True,
        # training
        timesteps_per_iteration = 2000,
        rollout_fragment_length = 200,
        train_batch_size = 6400,
        sgd_minibatch_size=256,
        num_sgd_iter=10,
        lr=3e-4,
        **{"lambda": 0.95},
    )
    ray.init()
    algo = ppo.PPO(config=train_config) 
    for i in tqdm(range(1000)):
        result = algo.train()
```

chrisgao99, 2024-03-20:
Through off-screen rendering, I found out that even though I set horizon=800, the MetaDrive Env won't stop after 800 steps and it also won't stop after it arrives dest. The simulation can always continue unless I set a break condition. So how can I make Metadrive know when to stop and reset?

```
test_env = env_creator3(env_config)
obs,info = test_env.reset(seed=seed)
for i in range(1200):
            action = algo.compute_single_action(obs)
            obs, r, tm, tc, info = test_env.step(action)
            o = obs['image'][..., :] * 255
            frame = o.astype('uint8') 
            video_writer.write(frame)
            if info["arrive_dest"]:
                print("this scenario has {} steps".format(i))
                print("the max_step should be: ",info["max_step"])
                print("the horizon of the env is: ",test_env.config["horizon"])
                break
```

QuanyiLi, 2024-03-20:
Hi, I guess there is something wrong in your `MyWrapper`. The following is my script that manages to stop the environment by setting `horizon`.
```python
from metadrive.component.sensors.rgb_camera import RGBCamera
from metadrive.component.map.pg_map import MapGenerateMethod
from metadrive.component.map.pg_map import BaseMap
from metadrive.envs.metadrive_env import MetaDriveEnv

frame_size = (320, 240)
env_config=dict(
                map_config={BaseMap.GENERATE_TYPE: MapGenerateMethod.BIG_BLOCK_NUM,
                    BaseMap.GENERATE_CONFIG: 3},
                image_observation=True,
                use_render=False,
                image_on_cuda=False,
                sensors={
                    "rgb_camera": (RGBCamera, *frame_size), #w,h,c
                },
                vehicle_config={
                    "image_source": "rgb_camera",
                },
                stack_size = 1,
                discrete_action=False,
                horizon=800,
                random_spawn_lane_index=False,
                num_scenarios=1000,
                start_seed=0,
                traffic_density=0.2,
                accident_prob=0.2,
                log_level=50)

def env_creator3(env_config):
    env = MetaDriveEnv({"use_render": False, "image_observation": False,'horizon':10})
    env.reset()
    env.close()
    print("The bullet env is open and closed")
    return MetaDriveEnv(env_config)

test_env = env_creator3(env_config)
obs,info = test_env.reset(seed=0)
for i in range(1200):
    print(i)
    obs, r, tm, tc, info = test_env.step(test_env.action_space.sample())
    o = obs['image'][..., :] * 255
    frame = o.astype('uint8')
    if info["arrive_dest"] or tm or tc:
        print("this scenario has {} steps".format(i))
        print("the max_step should be: ",info["max_step"])
        print("the horizon of the env is: ",test_env.config["horizon"])
        break
```

Note that the max_step termination will only be indicated by `truncate flag`, which is a new required feature according to Gymnasium. If you wrapper is changing gymnasium style environment to gym-style one, you have to set done as True when truncate is True. Or you can use our official wrapper for converting the environemnt to gym-style env.

QuanyiLi, 2024-03-20:
Sorry, I didn't see the content before the last code block. It seems you are following the Gymnasium API correctly. So what we can confirm now is that the environment can be terminated by setting `horizon=xyz` with `truncate` flag. For futher debugging the memory issue, there are some code examples in `metadrive/tests/test_component/test_memory_leak***.py` which you can use to familiar with the memory debugging tools. Then you can debug the environment with these tools to see if your environment keeps leaking RAM even if `reset` is called when the environment is terminated.

chrisgao99, 2024-03-20:
Thank you so much for the reply.

I tried the test_close_and_restart_metadrive_env function from metadrive/tests/test_component/test_memory_leak***.py and it works fine. Does it mean the memory leak is caused by env not reset?

Although we can break down the simulation by these conditions,
```
if info["arrive_dest"] or tm or tc:
     break
```
Is it possible that rllib doesn't use the tm or tc to reset the env?

QuanyiLi, 2024-03-20:
Could you try this example? It shows that no memory leak happens in the environment with reset.

```python
from metadrive.component.sensors.rgb_camera import RGBCamera
from metadrive.component.map.pg_map import MapGenerateMethod
import matplotlib.pyplot as plt
from metadrive.tests.test_functionality.test_memory_leak_engine import process_memory
from metadrive.component.map.pg_map import BaseMap
from metadrive.envs.metadrive_env import MetaDriveEnv

frame_size = (320, 240)
env_config = dict(
    map_config={BaseMap.GENERATE_TYPE: MapGenerateMethod.BIG_BLOCK_NUM,
                BaseMap.GENERATE_CONFIG: 3},
    image_observation=True,
    use_render=False,
    image_on_cuda=False,
    sensors={
        "rgb_camera": (RGBCamera, *frame_size),  # w,h,c
    },
    vehicle_config={
        "image_source": "rgb_camera",
    },
    stack_size=1,
    discrete_action=False,
    horizon=100,
    random_spawn_lane_index=False,
    num_scenarios=1000,
    start_seed=0,
    traffic_density=0.2,
    accident_prob=0.2,
    log_level=50)

def env_creator3(env_config):
    env = MetaDriveEnv({"use_render": False, "image_observation": False, 'horizon': 10})
    env.reset()
    env.close()
    print("The bullet env is open and closed")
    return MetaDriveEnv(env_config)

test_env = env_creator3(env_config)

plt.ion()  # Turn on interactive mode
fig, ax = plt.subplots()
memory_usage = []

while True:
    obs, info = test_env.reset(seed=0)

    memory = process_memory(to_mb=True)
    print(memory)
    memory_usage.append(memory)
    ax.clear()
    ax.plot(memory_usage)
    ax.set_xlabel('Step')
    ax.set_ylabel('Memory Usage')
    plt.pause(0.05)  # A short pause to update the plot

    for i in range(100):
        print(i)
        obs, r, tm, tc, info = test_env.step(test_env.action_space.sample())
        o = obs['image'][..., :] * 255
    assert tc, test_env.episode_step

```

QuanyiLi, 2024-03-20:
If you only reset once, the memory leak won't happen as well. You can confirm this with the following code.
```python
obs, info = test_env.reset(seed=0)
while True:
    memory = process_memory(to_mb=True)
    print(memory)
    memory_usage.append(memory)
    ax.clear()
    ax.plot(memory_usage)
    ax.set_xlabel('Step')
    ax.set_ylabel('Memory Usage')
    plt.pause(0.05)  # A short pause to update the plot

    for i in range(100):
        print(i)
        obs, r, tm, tc, info = test_env.step(test_env.action_space.sample())
        o = obs['image'][..., :] * 255
    assert tc, test_env.episode_step
```

chrisgao99, 2024-03-21:
Thanks. Yes, no memory leak happens even though env.close() is not called. So this is not likely MetaDrive's problem. But the memory would increase over time using PPO from RLLib
```
algo = ppo.PPO(config=train_config) 
    for i in tqdm(range(1000)):
        result = algo.train()
```
Do you have any guess on the reason? It's fine If you are also not sure.

chrisgao99, 2024-03-22:
Ok, Thank you. I think the memory pressure will still increase over epochs but I could try a smaller batch size or a smaller camera.

chrisgao99, 2024-03-28:
I changed to stable baseline 3 which doesn't have any memory issue though i still don't understand why rllib trigger the memory leak

QuanyiLi, 2024-03-29:
It is strange. Could you try upgrading rllib to a higher version? It should be easy and won't change the training script too much.

chrisgao99, 2024-04-03:
yes it is strange. Actually the version I'm using is the newest version

closed 2024-04-25 by chrisgao99