metadriverse/metadrive #691: Possibility of adding Irregular customized intersections for MARL environments.

gcruiser, 2024-04-01:
Hi, thanks for the outstanding work!
I'm currently interested in MARL and here are a few questions:

1. How to customize an Irregular and unstructured intersection, such as defining road segments and connecting segments by specifying the start and end coordinates, since I wish to model a real intersection from a natural driving environment into a Metadrive environment.

2. Can the customized environment mentioned above be applied to the MARL environment? I noticed that the number of MARL environments is slightly less than the total number of metadrive environments, I wonder if there are any unique limitations of MARL environments.

QuanyiLi, 2024-04-01:
Hi, thanks for your interest!

1. In this case, you can build the map with `ScenarioMap` and `ScenarioBlocks`, which take a list of map features as input. This API is designed to build real-world maps from real-world datasets but can also serves as interface for customizing maps. The input , map features, should be a dictionary described here: https://metadrive-simulator.readthedocs.io/en/latest/scenario_description.html

2. If you use the PGMARL environment, the number of maps would be infinite as well. But for other MARL environments that have specific structures, the number is limited by the shape. You can have a set of real-world maps created and managed by a map manager so every episode you can load a different map for training. I don't think there are any limitations applying to MARL env.

QuanyiLi, 2024-04-15:
Sorry for the late reply.

It is definitely possible. To achieve this, just assign different policies for your vehicles, for example, some are neural networks, others are IDM. Here is an example: https://github.com/metadriverse/metadrive/blob/main/metadrive/manager/traffic_manager.py#L367

Though it may not work now, you can still get some insights from it.