metadriverse/metadrive #698: Unexpected Behavior with `out_of_route_done=False` in MetaDrive Simulator

ensan-hcl, 2024-04-08:
### Description
Currently, in the MetaDrive simulator, specifying `out_of_route_done=False` in the configuration should theoretically prevent the simulation from ending when the vehicle deviates from the route. However, the simulation terminates as soon as `vehicle.on_lane` becomes `False`, based on the [implementation](https://github.com/metadriverse/metadrive/blob/main/metadrive/envs/metadrive_env.py#L230-L238).

### Questions
Intended Behavior Clarification: Is it intended that the simulation ends when `vehicle.on_lane` becomes `False`, even when `out_of_route_done=False` is specified in the configuration?

### Improvement Suggestions:

Is it possible to modify the behavior so that the simulation does not end when `vehicle.on_lane` becomes `False`, even when `out_of_route_done=False`? This will be possible by just moving the `vehicle.on_lane` checks inside `if`.
Alternatively, could a new configuration parameter, such as `out_of_lane_done`, be introduced to control the termination of the simulation when the vehicle deviates from the lane?

### Additional Information
MetaDrive version: 0.4.2.3

QuanyiLi, 2024-04-15:
Hi,

Sorry for the late reply.

Yes, it is expected. To achieve your goal, you can make a new sub-environment with the [`done_function`](https://github.com/metadriverse/metadrive/blob/233a3a1698be7038ec3dd050ca10b547b4b3324c/metadrive/envs/metadrive_env.py#L129) overridden. So you can have any termination conditions according to your needs. We are not able to support this officially as different user has different requests. It is better to make it highly customizable.

ensan-hcl, 2024-04-16:
I understood. It sounds very reasonable. Thank you for your response!

closed 2024-04-16 by ensan-hcl