metadriverse/metadrive #700: Configuring fixed parameters for map blocks when creating maps using block_sequence?

XuanChen-xc, 2024-04-11:
Hi, thank you for the great work!

I have a question about the randomization of the parameters for the `PGBlock`. 
For example, we can set `config["map"] = "SCS"` if we want to create a map that contains a Straight block, followed by a Circular and another Straight block. But the parameters in each block are randomized. If I want to fix the direction and the radius of the Circular block, is there any way to achieve this, or specify a random seed somewhere to control the randomization? Looks like the `global_seed` in the engine does not control the randomization there.

Thank you.

QuanyiLi, 2024-04-15:
If you want to control the generation process, you can take a look at the multi-agent environment, like https://github.com/metadriverse/metadrive/blob/233a3a1698be7038ec3dd050ca10b547b4b3324c/metadrive/envs/marl_envs/marl_tollgate.py#L113

In this map class, we build blocks by specifying the parameters instead of sampling them from a pre-defined space. And then all blocks are connected into a whole map. You can also have some blocks constructed from specific parameters, but others from sampled parameters. So you can have deterministic behavior but some randomness as well.

XuanChen-xc, 2024-04-15:
Thank you for your reply, I will try that!

closed 2024-04-15 by XuanChen-xc