metadriverse/metadrive #701: What is the best practice for reproducibility?

Altriaex, 2024-04-11:
Hi, I am trying to improve the reproducibility of my experiments. 

In this page https://metadrive-simulator.readthedocs.io/en/latest/training.html# the set_random_seed function from stable-baselines3 is used. However, I notice that when resetting the MetaDriveEnv, _reset_global_seed function computes its own seed, if I do not pass a seed explicitly to the env.reset() function call.  So each time I run the program the agent gets different scenario_index. 

After reading base_env.py, I see that there is a class attribute _DEBUG_RANDOM_SEED that play the role of the real seed. After setting it to some integer (e.g. 0) I manage to have deterministic scenario_index. But modifying such a class attribute can be tricky when we use multiple environments. So I wonder what is the best practice for reproducibility.

QuanyiLi, 2024-04-15:
I am afraid that the only way is to pass a seed to `env.reset()` It controls all random behaviors of the simulator. Could you tell us what problem blocks you from doing so?

Altriaex, 2024-04-16:
It becomes a bit tricky when we use multiple envs via, e.g., stable-baselines3's SubProc, because the base environments (i.e. MetaDriveEnv) are reset automatically by SubProc. Such resetting behavior assumes no arguments.

pengzhenghao, 2024-04-16:
One solution is to wrap the environment with a global config call “global_seed” and instantiate a RNG with that seed and automatically generate next interger in each env.reset call from SubProcEnv and auto fill that integer into env.reset(random_seed=..)

Best,
Zhenghao