metadriverse/metadrive #702: verify_headless_installation error

Sephirex-X, 2024-04-12:
Following every things for installation,  poping
`AttributeError: 'NoneType' object has no attribute 'set_shader'` error when running `python -m metadrive.examples.verify_headless_installation`

QuanyiLi, 2024-04-15:
Ahh, it suggests that there is something wrong with your driver or OpenGL setting. Could you take a look at this question? It might be helpful: https://github.com/metadriverse/metadrive/issues/632

Sephirex-X, 2024-04-16:
> Ahh, it suggests that there is something wrong with your driver or OpenGL setting. Could you take a look at this two questions? they may be helpful: #632 and #334

Thx, I have checked multiple times, the problem can be narrowed down to something related to monitor (successful verfied on another machine with monitor). I'm wondering if ur guys can provide tutorials for those user develop on server without monitor, e.g. whats kind of dependencies are needed, or a docker version. That would help a lot :)

closed 2024-04-22 by Sephirex-X