metadriverse/metadrive #703: Black screen for all 3D rendering

mtribiere, 2024-04-13:
**Environment:**

OS: Windows 11 with WSL
Version: master branch

---

**Step to reproduce:**
Install the Python package with WSL in a venv

---

**What shouldn't happen:**
The screen shouldn't be completely blacked out for 3D camera renders.

[image]

Code used:
``` python
from metadrive.envs.metadrive_env import MetaDriveEnv
import gymnasium as gym

env = MetaDriveEnv(config={"use_render": True})
obs, info = env.reset()
for i in range(1000):
    obs, reward, terminated, truncated, info = env.step(env.action_space.sample())
    if terminated or truncated:
        env.reset()
env.close()
```
Logs:
```
[INFO] Environment: MetaDriveEnv
[INFO] MetaDrive version: 0.4.2.3
[INFO] Sensors: [lidar: Lidar(), side_detector: SideDetector(), lane_line_detector: LaneLineDetector(), main_camera: MainCamera(1200, 900), dashboard: DashBoard()]
[INFO] Render Mode: onscreen
[INFO] Horizon (Max steps per agent): None
[INFO] Assets version: 0.4.2.3
[INFO] Known Pipes: glxGraphicsPipe
[INFO] Start Scenario Index: 0, Num Scenarios : 1
```

---

**What should happen:**
The 3D render should be displayed properly

QuanyiLi, 2024-04-15:
Hi,

We haven't tried the simulator in WSL before... It might be due to the limitation of the game engine itself.

If you want to do some training, MetaDrive can run on Windows machines as well and for myself, I usually use Windows platform to train some agents with stable-baseline. So it is fine to just use Windows instead of WSL.

But if you really want to use WSL, you'd better make sure this issue is raised by MetaDrive instead of the game engine, so we can take a look at it. In this case, you can 1. install panda3d through pip and 2. run examples from https://github.com/panda3d/panda3d/tree/master/samples If these sample programs are broken as well, that could be a problem with the pand3d game engine, and we won't be able to fix it.

mtribiere, 2024-04-18:
Confirmed, it seems to be an issue with panda3d itself. 
I suppose the issue can be closed?

QuanyiLi, 2024-04-19:
sure. Thank you for your feedback

closed 2024-04-19 by QuanyiLi