metadriverse/metadrive #704: all navigation module assigns random destination using the same seed

pengzhenghao, 2024-04-17:
This makes all traffic vehicles choose the same destination all the time. However, I am not sure if setting random seed from base_vehicle will cause the system loss reproducibility:

(note that there previously the navigation module grabs global random seed to pick up the destination, instead of accepting a int as input.)

[image]

pengzhenghao, 2024-04-17:
Also, I notice that in TrafficManager, when creating the vehicle there is no random seed passing in.

[image]

pengzhenghao, 2024-04-17:
Another issue: 

```python
    @staticmethod
    def auto_assign_task(map, current_lane_index, final_road_node=None, random_seed=None):
        # TODO we will assign the route in the task manager in the future
        start_road_node = current_lane_index[0]
        if start_road_node is None:
            start_road_node = FirstPGBlock.NODE_1
        if final_road_node is None:
            current_road_negative = Road(*current_lane_index[:-1]).is_negative_road()
            # choose first block when born on negative road
            block = map.blocks[0] if current_road_negative else map.blocks[-1]
            sockets = block.get_socket_list()
            socket = get_np_random(random_seed).choice(sockets)
            while True:
                if not socket.is_socket_node(start_road_node) or len(sockets) == 1:
                    break
                else:
                    sockets.remove(socket)
                    if len(sockets) == 0:
                        raise ValueError("Can not set a destination!")
            # choose negative road end node when current road is negative road
            final_road_node = socket.negative_road.end_node if current_road_negative else socket.positive_road.end_node
        return final_road_node
```

In the line about, the sockets will only belong to the first block or the last block. It is impossible to pick a socket that is in the same block.

QuanyiLi, 2024-04-19:
1. Do you observe that all vehicles go to the same destination if the last block is an intersection? It is obviously not what we expect and is a bug.
2. Random seed will be auto-generate and assigned to the object generated with engine.spawn
3. Yes. It is expected. You can find the blocks after the current block, so you can get sockets of not only frst/last blocks as potential destination.