metadriverse/metadrive #709: verify_headless_installation fails

mimi89999, 2024-04-23:
Hello,

I noticed that running fails with the following error:
```
(.venv) michel@debian:/tmp$ python -m metadrive.examples.verify_headless_installation
[INFO] Environment: MetaDriveEnv
[INFO] MetaDrive version: 0.4.2.3
[INFO] Sensors: [lidar: Lidar(), side_detector: SideDetector(), lane_line_detector: LaneLineDetector()]
[INFO] Render Mode: none
[INFO] Horizon (Max steps per agent): None
[INFO] Assets version: 0.4.2.3
[INFO] Known Pipes: glxGraphicsPipe
[INFO] Start Scenario Index: 0, Num Scenarios : 1
[INFO] Episode ended! Scenario Index: 0 Reason: out_of_road.
Bullet physics world is launched successfully!
[INFO] Environment: MetaDriveEnv
[INFO] MetaDrive version: 0.4.2.3
[INFO] Sensors: [lidar: Lidar(), side_detector: SideDetector(), lane_line_detector: LaneLineDetector(), rgb_camera: RGBCamera(256, 256), depth_camera: DepthCamera(256, 256)]
[INFO] Render Mode: offscreen
[INFO] Horizon (Max steps per agent): None
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/tmp/.venv/lib/python3.11/site-packages/metadrive/examples/verify_headless_installation.py", line 11, in <module>
    verify_installation(args.cuda, args.camera)
  File "/tmp/.venv/lib/python3.11/site-packages/metadrive/tests/test_installation.py", line 84, in verify_installation
    capture_headless_image(cuda)
  File "/tmp/.venv/lib/python3.11/site-packages/metadrive/tests/test_installation.py", line 14, in capture_headless_image
    env = MetaDriveEnv(
          ^^^^^^^^^^^^^
  File "/tmp/.venv/lib/python3.11/site-packages/metadrive/envs/metadrive_env.py", line 103, in __init__
    super(MetaDriveEnv, self).__init__(config)
  File "/tmp/.venv/lib/python3.11/site-packages/metadrive/envs/base_env.py", line 299, in __init__
    self.agent_manager = self._get_agent_manager()
                         ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.venv/lib/python3.11/site-packages/metadrive/envs/base_env.py", line 396, in _get_agent_manager
    return VehicleAgentManager(init_observations=self._get_observations())
                                                 ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.venv/lib/python3.11/site-packages/metadrive/envs/base_env.py", line 393, in _get_observations
    return {DEFAULT_AGENT: self.get_single_observation()}
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.venv/lib/python3.11/site-packages/metadrive/envs/base_env.py", line 661, in get_single_observation
    o = ImageStateObservation(self.config) if img_obs else LidarStateObservation(self.config)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.venv/lib/python3.11/site-packages/metadrive/obs/image_obs.py", line 26, in __init__
    self.img_obs = ImageObservation(config, config["vehicle_config"]["image_source"], config["norm_pixel"])
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.venv/lib/python3.11/site-packages/metadrive/obs/image_obs.py", line 61, in __init__
    self.state = np.zeros(self.observation_space.shape, dtype=np.float32 if self.norm_pixel else np.uint8)
                          ^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.venv/lib/python3.11/site-packages/metadrive/obs/image_obs.py", line 67, in observation_space
    sensor_cls = self.config["sensors"][self.image_source][0]
                 ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^
  File "/tmp/.venv/lib/python3.11/site-packages/metadrive/utils/config.py", line 225, in __getitem__
    self._check_and_raise_key_error(item)
  File "/tmp/.venv/lib/python3.11/site-packages/metadrive/utils/config.py", line 211, in _check_and_raise_key_error
    raise KeyError(
KeyError: "'main_camera' does not exist in existing config. Please use config.update(...) to update the config. Existing keys: dict_keys(['lidar', 'side_detector', 'lane_line_detector', 'rgb_camera', 'depth_camera'])."
```

After switching to the rgb camera, I get a different error:
```
(.venv) michel@debian:/tmp$ python -m metadrive.examples.verify_headless_installation --camera rgb
[INFO] Environment: MetaDriveEnv
[INFO] MetaDrive version: 0.4.2.3
[INFO] Sensors: [lidar: Lidar(), side_detector: SideDetector(), lane_line_detector: LaneLineDetector()]
[INFO] Render Mode: none
[INFO] Horizon (Max steps per agent): None
[INFO] Assets version: 0.4.2.3
[INFO] Known Pipes: glxGraphicsPipe
[INFO] Start Scenario Index: 0, Num Scenarios : 1
[INFO] Episode ended! Scenario Index: 0 Reason: out_of_road.
Bullet physics world is launched successfully!
[INFO] Environment: MetaDriveEnv
[INFO] MetaDrive version: 0.4.2.3
[INFO] Sensors: [lidar: Lidar(), side_detector: SideDetector(), lane_line_detector: LaneLineDetector(), rgb_camera: RGBCamera(256, 256), depth_camera: DepthCamera(256, 256)]
[INFO] Render Mode: offscreen
[INFO] Horizon (Max steps per agent): None
[INFO] Assets version: 0.4.2.3
Xlib:  extension "XFree86-DGA" missing on display ":0".
[INFO] Known Pipes: glxGraphicsPipe
[WARNING] You are using too large buffer! The height is 256, and width is 256. It may lower the sample efficiency! Consider reducing buffer size or use cuda image by set [image_on_cuda=True]. (base_camera.py:46)
[WARNING] You are using too large buffer! The height is 256, and width is 256. It may lower the sample efficiency! Consider reducing buffer size or use cuda image by set [image_on_cuda=True]. (base_camera.py:46)
Assertion failed: _current_shader_context != nullptr at line 6939 of panda/src/glstuff/glGraphicsStateGuardian_src.cxx
Traceback (most recent call last):
  File "/tmp/.venv/lib/python3.11/site-packages/metadrive/component/sensors/depth_camera.py", line 142, in _dispatch_compute
    self.engine.graphicsEngine.dispatch_compute(
AssertionError: _current_shader_context != nullptr at line 6939 of panda/src/glstuff/glGraphicsStateGuardian_src.cxx
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/tmp/.venv/lib/python3.11/site-packages/metadrive/examples/verify_headless_installation.py", line 11, in <module>
    verify_installation(args.cuda, args.camera)
  File "/tmp/.venv/lib/python3.11/site-packages/metadrive/tests/test_installation.py", line 86, in verify_installation
    capture_headless_image(cuda, image_source="rgb_camera")
  File "/tmp/.venv/lib/python3.11/site-packages/metadrive/tests/test_installation.py", line 33, in capture_headless_image
    env.reset()
  File "/tmp/.venv/lib/python3.11/site-packages/metadrive/envs/base_env.py", line 519, in reset
    self.lazy_init()  # it only works the first time when reset() is called to avoid the error when render
    ^^^^^^^^^^^^^^^^
  File "/tmp/.venv/lib/python3.11/site-packages/metadrive/envs/base_env.py", line 406, in lazy_init
    initialize_engine(self.config)
  File "/tmp/.venv/lib/python3.11/site-packages/metadrive/engine/engine_utils.py", line 12, in initialize_engine
    cls.singleton = cls(env_global_config)
                    ^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.venv/lib/python3.11/site-packages/metadrive/engine/base_engine.py", line 88, in __init__
    self.warmup()
  File "/tmp/.venv/lib/python3.11/site-packages/metadrive/engine/base_engine.py", line 741, in warmup
    self.taskMgr.step()
  File "/tmp/.venv/lib/python3.11/site-packages/direct/task/Task.py", line 504, in step
    self.mgr.poll()
  File "/tmp/.venv/lib/python3.11/site-packages/metadrive/component/sensors/depth_camera.py", line 142, in _dispatch_compute
    self.engine.graphicsEngine.dispatch_compute(
AssertionError: _current_shader_context != nullptr at line 6939 of panda/src/glstuff/glGraphicsStateGuardian_src.cxx
```

QuanyiLi, 2024-04-29:
For the first error, you should explicitly declare that you need the main_camera as a sensor by putting it in the sensor config:
```python
"sensors":{"main_camera": ()
    ...                 
}
```
So you can access it even in offscreen mode.

The second error is raised by the depth camera because the GPU doesn't support OpenGL > 4.0 and the compute shader is not available. It usually happens for users using an integrated graphics card.

mimi89999, 2024-04-30:
My GPU should support OpenGL 4.6. Is there any specific extension that is required that I could check for?

QuanyiLi, 2024-06-09:
Sorry for the late reply. You would be better to try some compute shader that is irrelevant to MetaDrive and Panda3D to figure out whether it is the OpenGL/hardware problem. Then we can do further debugging on Panda3D and MetaDrive side.