metadriverse/metadrive #721: Headless: extracted image is colorless running in Docker container

bongbui321, 2024-05-26:
I'm currently running Metadrive in a Docker container and it seems that the image extracted from "rgb_camera" is all white (image below)
[image]

It runs fine on host machine (having colors and textures are correct). I have confirmed that textures and shaders are loaded (paths to assets are correct), but unsure why it is like the image above. I extracted the images from `rgb_camera.perceive()`

bongbui321, 2024-05-30:
My bad, this problem should have been solved, I was using Metadrive 4.2.3

closed 2024-05-30 by bongbui321

zhangqf2001, 2024-06-06:
I also encountered this problem, can I ask how you solved it?

yuyangzhang323, 2024-10-28:
I have the same problem, how do you solve it?

pengzhenghao, 2025-04-01:
Not sure if this issue still exists if you install metadrive via 
`pip install git+.....`

Feel free to reopen this issue.