metadriverse/metadrive #743: the background of simulator is not right

xiaoshizijiayou, 2024-07-17:
[image]
[image]
my simulator  backgroud is different from demo ,where is the problem?Is it about assets?how can i to fix it?
thanks!

maxime-desroches, 2024-07-28:
The shader for the skybox is declared for version 150 (OpenGL version 3.2) and the terrain shader is declared for version 330 (OpenGL version 3.3) 

https://github.com/metadriverse/metadrive/blob/aaed1f7f2512061ddd8349d1d411e374dab87a43/metadrive/shaders/terrain.frag.glsl#L1

https://github.com/metadriverse/metadrive/blob/aaed1f7f2512061ddd8349d1d411e374dab87a43/metadrive/shaders/skybox.frag.glsl#L1

So you would need a system with an OpenGL driver supporting at least version 3.3 in order to get both the sky and the road working.

You can check your OpenGL version using `glxinfo` from the `mesa-utils` apt package

pengzhenghao, 2024-12-15:
Could you try out the latest metadrive? (please avoid installing it from pypi)

I am not sure if this problem only happen in Windows