metadriverse/metadrive #760: How to upload a flat image in the simulator

ahoora79, 2024-09-10:
How to upload a flat image in the simulator environment? (to calibrate cameras for SVM technology)?

QuanyiLi, 2024-09-11:
The following code should work.

```
from panda3d.core import CardMaker

texture=engine.loader.loadTexture("path/to/your/texture")
cm = CardMaker('card')
scale = 4000
cm.setUvRange((0, 0), (scale / 10, scale / 10))
card = env.enginre.origin.attachNewNode(cm.generate())
card.set_scale(scale)
card.setPos(-scale / 2, -scale / 2, -0.1)
card.setZ(-.05)
card.setTexture(texture)
```

ahoora79, 2024-09-15:
thanks for answering look at pic : the black plane is now very big!
[image]

and this is my code which i want upload below image to env:
[image]

and this is my code:
`from metadrive import MetaDriveEnv
from panda3d.core import Texture, CardMaker, NodePath, AmbientLight, DirectionalLight

# Initialize MetaDrive environment
config = dict(
    use_render=True,
    manual_control=True,
    num_scenarios=10000,
    map=4,
    start_seed=10
)

env = MetaDriveEnv(config)

# Reset the environment to start a new episode
env.reset()

def create_checkerboard_texture(checkerboard_image_path):
    # Load the checkerboard image as a texture using the provided method
    texture = env.engine.loader.loadTexture(checkerboard_image_path)
    if not texture:
        print(f"Error loading texture from {checkerboard_image_path}")
        return

    print(f"Texture loaded successfully from {checkerboard_image_path}")

    # Create a plane (2D surface) using CardMaker with proper scaling
    card_maker = CardMaker('checkerboard_plane')

    # Set the scale for the UV mapping to repeat the texture
    scale = 4000
    card_maker.setUvRange((0, 0), (scale / 10, scale / 10))

    # Create a NodePath for the plane and add it to the scene
    card_node = env.engine.origin.attachNewNode(card_maker.generate())
    card_node.set_scale(scale)

    # Set the position at the origin (0, 0, 0)
    card_node.set_pos(0, 0, 0)
    card_node.set_z(0)  # Adjust the height to exactly 0
    card_node.set_texture(texture)

    # Optional: Print the node's details for debugging
    print(f"Plane details: {card_node}")

    # Add some basic lighting
    ambient_light = AmbientLight('ambient_light')
    ambient_light.set_color((0.5, 0.5, 0.5, 1))  # Light color
    ambient_light_node = NodePath(ambient_light)
    env.engine.render.set_light(ambient_light_node)

    directional_light = DirectionalLight('directional_light')
    directional_light.set_color((1, 1, 1, 1))  # Light color
    directional_light_node = NodePath(directional_light)
    directional_light_node.look_at(0, 0, -1)  # Direction of the light
    env.engine.render.set_light(directional_light_node)

# Path to the checkerboard image
checkerboard_image_path = 'checkerboard.png'

# Create the checkerboard in the environment
create_checkerboard_texture(checkerboard_image_path)

# Run the simulation loop
for _ in range(1000):  # Run for a fixed number of steps
    action = env.action_space.sample()  # Sample a random action
    env.step(action)  # Perform the action
    env.render()  # Render the environment to the screen

# Close the environment when done
env.close()
`

QuanyiLi, 2024-09-16:
The `card_node.set_scale(scale)` enlarges the image by `scale` times. So set the `scale` to 1 would make it 1m x 1m. Also, the `card_maker.setUvRange((0, 0), (scale / 10, scale / 10))` may need to be changed to `card_maker.setUvRange((0, 0), (1, 1))` to stop repeating the texture several times.

ahoora79, 2024-09-16:
thanks for answering!  i apply your commands in my code (scale = 1 and change `card_maker.setUvRange((0, 0), (scale / 10, scale / 10))` to `card_maker.setUvRange((0, 0), (1, 1))` but still black plane

 
[image]

my code :
```
from metadrive import MetaDriveEnv
from panda3d.core import Texture, CardMaker, NodePath, AmbientLight, DirectionalLight

# Initialize MetaDrive environment
config = dict(
    use_render=True,
    manual_control=True,
    num_scenarios=10000,
    map=4,
    start_seed=10
)

env = MetaDriveEnv(config)

# Reset the environment to start a new episode
env.reset()

def create_checkerboard_texture(checkerboard_image_path):
    # Load the checkerboard image as a texture using the provided method
    texture = env.engine.loader.loadTexture(checkerboard_image_path)
    if not texture:
        print(f"Error loading texture from {checkerboard_image_path}")
        return

    print(f"Texture loaded successfully from {checkerboard_image_path}")

    # Create a plane (2D surface) using CardMaker with proper scaling
    card_maker = CardMaker('checkerboard_plane')

    # Set the scale for the UV mapping to repeat the texture
    scale = 1
    card_maker.setUvRange((0, 0), (1, 1))

    # Create a NodePath for the plane and add it to the scene
    card_node = env.engine.origin.attachNewNode(card_maker.generate())
    card_node.set_scale(scale)

    # Set the position at the origin (0, 0, 0)
    card_node.set_pos(0, 0, 0)
    card_node.set_z(0)  # Adjust the height to exactly 0
    card_node.set_texture(texture)

    # Optional: Print the node's details for debugging
    print(f"Plane details: {card_node}")

    # Add some basic lighting
    ambient_light = AmbientLight('ambient_light')
    ambient_light.set_color((0.5, 0.5, 0.5, 1))  # Light color
    ambient_light_node = NodePath(ambient_light)
    env.engine.render.set_light(ambient_light_node)

    directional_light = DirectionalLight('directional_light')
    directional_light.set_color((1, 1, 1, 1))  # Light color
    directional_light_node = NodePath(directional_light)
    directional_light_node.look_at(0, 0, -1)  # Direction of the light
    env.engine.render.set_light(directional_light_node)

# Path to the checkerboard image
checkerboard_image_path = 'checkerboard.png'
print('hello')
# Create the checkerboard in the environment
create_checkerboard_texture(checkerboard_image_path)

# Run the simulation loop
for _ in range(1000):  # Run for a fixed number of steps
    action = env.action_space.sample()  # Sample a random action
    env.step(action)  # Perform the action
    env.render()  # Render the environment to the screen

# Close the environment when done
env.close()
```

ahoora79, 2024-09-25:
@QuanyiLi can you update this conversation ? i still have black plane