metadriverse/metadrive #761: Installing Metadrive. Error: subprocess-exited-with-error

oonuro, 2024-09-15:
Hello, I am getting an issue when I am installing library "metadrive". Is this issue from my site? What should I do fix it? 

```
(.venv) root@root:~/projectpy$ pip install metadrive-simulator
Collecting metadrive-simulator
  Using cached metadrive_simulator-0.2.6.0-py3-none-any.whl.metadata (8.6 kB)
Collecting gym==0.19.0 (from metadrive-simulator)
  Using cached gym-0.19.0.tar.gz (1.6 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [3 lines of output]
      /tmp/pip-build-env-ipjfv3xj/overlay/lib/python3.12/site-packages/setuptools/_distutils/dist.py:261: UserWarning: Unknown distribution option: 'tests_require'
        warnings.warn(msg)
      error in gym setup command: 'extras_require' must be a dictionary whose values are strings or lists of strings containing valid project/version requirement specifiers.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```
Python libraries in the virtual environment;
 
```(.venv) onur@onur:~/projectpy$ pip list
Package     Version
----------- -------
cloudpickle 3.0.0
gym         0.26.2
gym-notices 0.0.8
numpy       2.1.1
pip         24.2```

QuanyiLi, 2024-09-15:
The metadrive version collected by your `pip` is 0.2.6, but the latest version is `0.4.2.3`. Could you try `pip install metadrive-simulator==0.4.2.3` or install from github?

oonuro, 2024-09-17:
Hello, thank you very much for your help. I managed to install it.

closed 2024-09-17 by oonuro