metadriverse/metadrive #782: Car Doesn't Fully Stop When Applying Full Brake

lola831, 2024-12-07:
Hi MetaDrive Team,

While integrating MetaDrive into Scenic, I’ve encountered an issue where the vehicle does not fully stop when applying a throttle-brake input of [0, -1]. Instead, its speed oscillates near zero, and in the 2D render, the car visibly moves backward at a constant rate during the braking phase.

I found [issue #576](https://github.com/metadriverse/metadrive/issues/576), which describes similar behavior. Could you confirm if this is a known problem that is not planned to be addressed, or if there’s a potential fix or workaround? This is a real problem for us, as the car moving visibly backward can be confusing for users.

Here’s minimal reproducible code that demonstrates the issue:

```python
from metadrive.envs import MetaDriveEnv

config = {
    "use_render": False,
    "map": "y",
    "vehicle_config": {
        "spawn_position_heading": [[10, 10], 0],
    },
}

env = MetaDriveEnv(config)
env.reset()

try:
    # Driving forward phase
    for step in range(30):  # Drive forward for 30 steps
        env.step([0, 1])  # Full throttle, no steering
        print("Speed: ", env.agent.speed)
        env.render(mode="topdown")

    print("Starting Brake Phase")
    while True:  # Continue braking until the car stops
        env.step([0.0, -1.0])  # No throttle, apply brake
        print("Speed: ", env.agent.speed)
        env.render(mode="topdown")

        if env.agent.speed <= 0.01:  # Stop if speed is effectively zero
            print("Car has stopped.")
            break

finally:
    env.close()
```
Thank you for your time and assistance.

pengzhenghao, 2024-12-15:
Hi this is a known issue because we implement brake by applying negative acceleration to the agent via the game engine. To implement that I think a deadzone or a little PID for the negative acc will help. I can make a quick try now.

pengzhenghao, 2024-12-15:
The root cause is here

```python

    def _apply_throttle_brake(self, throttle_brake):
        max_engine_force = self.config["max_engine_force"]
        max_brake_force = self.config["max_brake_force"]
        for wheel_index in range(4):
            if throttle_brake >= 0:
                self.system.setBrake(2.0, wheel_index)
                if self.speed_km_h > self.max_speed_km_h:
                    self.system.applyEngineForce(0.0, wheel_index)
                else:
                    self.system.applyEngineForce(max_engine_force * throttle_brake, wheel_index)
            else:
                if self.enable_reverse:
                    self.system.applyEngineForce(max_engine_force * throttle_brake, wheel_index)
                    self.system.setBrake(0, wheel_index)
                else:
                    self.system.applyEngineForce(0.0, wheel_index)
                    self.system.setBrake(abs(throttle_brake) * max_brake_force, wheel_index)

```

metadrive/component/vehicle/base_vehicle.py

I'll try to fix it.

PR #786 (2024-12-15): Fix issue that car can't full stop if brake
## What changes do you make in this PR?

* Please describe why you create this PR

close #782 

## Checklist

* [x] I have merged the latest main branch into current branch.
* [x] I have run `bash scripts/format.sh` before merging.
* Please use "squash and merge" mode.
closed 2024-12-15 by pengzhenghao