metadriverse/metadrive #803: Slow FPS when running single agent example

alfredgu001324, 2025-01-20:
Hi, thanks for the great work! I am using RTX4090, but when I run `python -m metadrive.examples.drive_in_single_agent_env`, the fps is around 1.5x, but when I run `python -m metadrive.examples.profile_metadrive.py`, fps is around 1100. When I run `python -m metadrive.examples.top_down_metadrive`, it also looks normal. 

Is there a way to accelerate the rendering? I assume 4090 is a pretty good graphics card already

QuanyiLi, 2025-01-20:
Hi,

It is likely the GPU doesn't work. You can check it with `Nvidia-semi` to see whether there is the python program on it. Could you provide more context i.e laptop or PC, windows or linux(wsl) and whether it is headless?

alfredgu001324, 2025-01-20:
Thanks for the quick response! The python program is indeed on it, I can see that it is using 478MB of memory by using nvidia-smi. It is a workstation from my lab, and it is linux. I am remoting into the workstation directly using remote chrome desktop.

alfredgu001324, 2025-01-20:
And I tried both manual control and a torch PPO agent, but both have very low fps.

QuanyiLi, 2025-01-20:
There could be multiple potential issues. 

Could you run: `glxinfo | grep '^direct rendering:'` and `glxinfo | grep "version"` to see whether the output is YES and the version is higher than 4.0? 

Also, please provide me with the terminal output of the metadrive program you are running.

Another question is: Is the workstation headless (without a screen), and you only access the rendering result through some x11 service? If so, it is highly likely a problem with X11 server.

QuanyiLi, 2025-01-20:
BTW, the latest version supports MacOS/Linux/Windows/WSL, so you can try it locally.

alfredgu001324, 2025-01-20:
1. This is the output of the commands:
```
(base) guxunjia@tisl-ws23-0:~/masters_project/metadrive$ glxinfo | grep '^direct rendering:'
direct rendering: Yes
(base) guxunjia@tisl-ws23-0:~/masters_project/metadrive$ glxinfo | grep "version"
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
    Max core profile version: 4.5
    Max compat profile version: 4.5
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
OpenGL core profile version string: 4.5 (Core Profile) Mesa 23.2.1-1ubuntu3.1~22.04.3
OpenGL core profile shading language version string: 4.50
OpenGL version string: 4.5 (Compatibility Profile) Mesa 23.2.1-1ubuntu3.1~22.04.3
OpenGL shading language version string: 4.50
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 23.2.1-1ubuntu3.1~22.04.3
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
    GL_EXT_shader_implicit_conversions, GL_EXT_shader_integer_mix, 
```

2. 
```
(meta_drive) guxunjia@tisl-ws23-0:~/masters_project/metadrive/documentation$ python -m metadrive.examples.drive_in_single_agent_env
[INFO] Environment: MetaDriveEnv
[INFO] MetaDrive version: 0.4.3
[INFO] Sensors: [lidar: Lidar(), side_detector: SideDetector(), lane_line_detector: LaneLineDetector(), main_camera: MainCamera(1200, 900), dashboard: DashBoard()]
[INFO] Render Mode: onscreen
[INFO] Horizon (Max steps per agent): None
[INFO] Assets version: 0.4.3
Xlib:  extension "XFree86-DGA" missing on display ":20".
[INFO] Known Pipes: glxGraphicsPipe
[WARNING] Since your screen is too small (1908, 906), we resize the window to (1087, 815). (engine_core.py:234)
[INFO] Start Scenario Index: 10, Num Scenarios : 10000
Keyboard Shortcuts:
  W: Acceleration
  S: Braking
  A: Moving Left
  D: Moving Right
  R: Reset the Environment
  H: Help Message
  F: Switch FPS to unlimited / realtime
  Q: Third-person View Camera
  B: Top-down View Camera (control: WASD-=)
  +: Lift Camera
  -: Lower Camera
  Mouse click: move camera (top-down view)
  Esc: Quit

The observation is an numpy array with shape:  (259,)
[INFO] Use Torch PPO expert.
Navigation information:  forward
```

3. Yes, I think from the above, there is a message saying: Xlib:  extension "XFree86-DGA" missing on display ":20". When I run this file using MetaTopdownEnv, this message disappears. Maybe it is indeed related to the display. 

4. Thanks! Actually my laptop is very old and I don't have other devices other than this workstation, but thanks for letting me know haha.

QuanyiLi, 2025-01-20:
I asked GPT for the error message: `XFree86-DGA" missing on display ":20`: https://chatgpt.com/share/678eb691-a84c-800e-82f2-86c31002920b

I think its answer confirmed that your x server configuration is problematic. GPT also gives some solutions, but be careful if you are going to try them.

alfredgu001324, 2025-01-20:
Lol yes, I think it's because our workstation is running through Xvbf which does not support XFree86-DGA. Thank you so much for your help! Yes I will try to play around with the top down view first (do not want to destroy my workstation ahaha). 

Thank you again for your help! Greatly appreciate it.

closed 2025-01-20 by alfredgu001324