metadriverse/metadrive #817: Issue with obsevation_heading_diff

EhsanSabouni, 2025-02-12:
def heading_diff(self, target_lane):
        lateral = None
        if isinstance(target_lane, StraightLane):
            lateral = np.asarray(get_vertical_vector(target_lane.end - target_lane.start)[1])
        elif isinstance(target_lane, CircularLane):
            if not target_lane.is_clockwise():
                lateral = self.position - target_lane.center
            else:
                lateral = target_lane.center - self.position
        elif isinstance(target_lane, PointLane):
            lateral = target_lane.lateral_direction(target_lane.local_coordinates(self.position)[0])

        lateral_norm = norm(lateral[0], lateral[1])
        forward_direction = self.heading
        # print(f"Old forward direction: {self.forward_direction}, new heading {self.heading}")
        forward_direction_norm = norm(forward_direction[0], forward_direction[1])
        if not lateral_norm * forward_direction_norm:
            return 0
        cos = (
            (forward_direction[0] * lateral[0] + forward_direction[1] * lateral[1]) /
            (lateral_norm * forward_direction_norm)
        )
        # return cos
        # Normalize to 0, 1
        return clip(cos, -1.0, 1.0) / 2 + 0.5

This is the code snippet in component/vehicle/base_vehcie.py. 
This is supposed to calculate the difference between the road heading and ego heading. It is not clear to me why we need the vertical vector of the road (e.g. when the lane is straight). Upon obtaining the vertical vector, it proceeds to obtain the ego direction vector, and then calculates the inner product to obtain the cosine value of the angle. Shoud not we use the actual lane direction vector instead of the orthogonal one? How does this work?

QuanyiLi, 2025-02-13:
Good question. Take a look at the following cosine graph. 

[image]

If we use the real road direction, then when it aligns with the vehicle direction, the dot product is 1. The dot product will go under 1 but still beyond zero no matter whether the car is turning a bit right or left, which is very important information but lost here. 

If we use the vertical lane direction, then the dot product will be changed to 0, when the car is driving in the same lane direction. Turning right or left will also make the sign of the dot product change to reflect the direction of the angle. So the policy can know how to do the steering.

[image]

EhsanSabouni, 2025-02-25:
Thanks for your response. One more question I do have if you do not mind. I am trying understand what is the logic behind the the function you use to return radius of the lane through observation. here is the snippet of the code:

def update_localization(self, ego_vehicle):
        """
        Update current position, route completion and checkpoints according to current position.

        Args:
            ego_vehicle: a vehicle object

        Returns:
            None
        """
        position = ego_vehicle.position
        lane, lane_index = self._update_current_lane(ego_vehicle)
        long, _ = lane.local_coordinates(position)
        need_update = self._update_target_checkpoints(lane_index, long)
        assert len(self.checkpoints) >= 2

        # Update travelled_length for route completion
        long_in_ref_lane, _ = self.current_ref_lanes[0].local_coordinates(position)
        travelled = long_in_ref_lane - self._last_long_in_ref_lane
        self.travelled_length += travelled
        self._last_long_in_ref_lane = long_in_ref_lane
        # print(f"{self.travelled_length=}, {travelled=}, {long_in_ref_lane=}, "
        #       f"{self.route_completion=}, {self._last_long_in_ref_lane=}")

        # target_road_1 is the road segment the vehicle is driving on.
        if need_update:
            target_road_1_start = self.checkpoints[self._target_checkpoints_index[0]]
            target_road_1_end = self.checkpoints[self._target_checkpoints_index[0] + 1]
            target_lanes_1 = self.map.road_network.graph[target_road_1_start][target_road_1_end]
            self.current_ref_lanes = target_lanes_1
            self.current_road = Road(target_road_1_start, target_road_1_end)

            self._last_long_in_ref_lane = self.current_ref_lanes[0].local_coordinates(position)[0]

            # target_road_2 is next road segment the vehicle should drive on.
            target_road_2_start = self.checkpoints[self._target_checkpoints_index[1]]
            target_road_2_end = self.checkpoints[self._target_checkpoints_index[1] + 1]
            target_lanes_2 = self.map.road_network.graph[target_road_2_start][target_road_2_end]

            if target_road_1_start == target_road_2_start:
                # When we are in the final road segment that there is no further road to drive on
                self.next_road = None
                self.next_ref_lanes = None
            else:
                self.next_road = Road(target_road_2_start, target_road_2_end)
                self.next_ref_lanes = target_lanes_2

        self._navi_info.fill(0.0)
        half = self.CHECK_POINT_INFO_DIM
        # Put the next checkpoint's information into the first half of the navi_info
        self._navi_info[:half], lanes_heading1, checkpoint = self._get_info_for_checkpoint(
            lanes_id=0, ref_lane=self.current_ref_lanes[0], ego_vehicle=ego_vehicle
        )

        # Put the next of the next checkpoint's information into the first half of the navi_info
        self._navi_info[half:], lanes_heading2, _ = self._get_info_for_checkpoint(
            lanes_id=1,
            ref_lane=self.next_ref_lanes[0] if self.next_ref_lanes is not None else self.current_ref_lanes[0],
            ego_vehicle=ego_vehicle
        )

        self.navi_arrow_dir = [lanes_heading1, lanes_heading2]
        if self._show_navi_info:
            # Whether to visualize little boxes in the scene denoting the checkpoints
            pos_of_goal = checkpoint

            self._goal_node_path.setPos(panda_vector(pos_of_goal[0], pos_of_goal[1], self.MARK_HEIGHT))
            self._goal_node_path.setH(self._goal_node_path.getH() + 3)
            dest_pos = self._dest_node_path.getPos()
            self._draw_line_to_dest(start_position=ego_vehicle.position, end_position=(dest_pos[0], dest_pos[1]))
            navi_pos = self._goal_node_path.getPos()
            self._draw_line_to_navi(start_position=ego_vehicle.position, end_position=(navi_pos[0], navi_pos[1]))

In Function 
self._get_info_for_checkpoint(
            lanes_id=0, ref_lane=self.current_ref_lanes[0], ego_vehicle=ego_vehicle
        )
why do we only use the zero index of current_ref_lanes list as the ref_lane. What if the agent drives in lane with index 1? For example if an agent is driving on lane with index 1 the road has bigger radius than road with index 0 (at least in roundabout). Thanks for your time and efforts.