metadriverse/metadrive #836: [Weired Performance] Vehicle State Deviation in Replay: while trying to using multiple autonomous driving algorithm to control the traffic flow.

AOOOOOA, 2025-04-17:
Hi,

I'm using the latest version of MetaDrive and have encountered an issue when simulating personalized scenarios.

In my scenario, I aim to deploy several autonomous driving algorithms to control traffic flow independently while interacting with ego vehicles. To simulate this, I initially use multiple traffic managers to spawn and control the traffic flow vehicles.

However, I observed a significant deviation in vehicle states when replaying the recorded scenario using a single manager. Specifically:

- Initially, I used multiple traffic managers to control the vehicles in the traffic flow and recorded their states.

- Later, I replayed the recorded scenario using one manager to control all the traffic vehicles.

During replay, the vehicle states deviated significantly from the original recorded states. For example:

- In the replay video (attached below), you can see vehicles running off the road in the upper right corner.

- Additionally, there are vehicle collisions (e.g., between two black vehicles).

I have attached my code for your reference and would greatly appreciate any insights or guidance on resolving this issue.

Thank you!

**To run the code below:** 
1. First, run it in the multiple manager mode and record the data: 
   python test_multiple_traffic_manager.py --mode record --num_managers 5

2. Then replay the record data using a single manager:
   python test_multiple_traffic_manager.py --mode replay --recording recorded_data/traffic_recording_complete.json

https://github.com/user-attachments/assets/c59012b3-8bce-4339-9611-e741cff27ef4

```
#!/usr/bin/env python
"""
Record and replay traffic patterns from multiple lane-divided traffic managers
"""
import argparse
import time
import json
import os
import numpy as np
from collections import defaultdict

from metadrive.envs import MetaDriveEnv
from metadrive.manager.traffic_manager import TrafficManager, TrafficMode
from metadrive.policy.idm_policy import IDMPolicy
from metadrive.component.vehicle.base_vehicle import BaseVehicle
from metadrive.component.vehicle.vehicle_type import DefaultVehicle

class StrictLaneManager:
    """Lane manager that strictly divides lanes among traffic managers"""
    def __init__(self):
        self.all_lanes = []  # Will be populated during first call
        self.lane_assignments = {}  # manager_id -> list of assigned lanes
    
    def get_lanes_for_manager(self, manager_id, total_managers, all_available_lanes):
        """Strictly divide available lanes among managers, with no overlap"""
        # Initialize all lanes list if first call
        if not self.all_lanes:
            self.all_lanes = list(all_available_lanes)
            
            # Sort lanes to ensure consistent assignment
            self.all_lanes.sort(key=lambda lane: (lane.index[0], lane.index[1], lane.index[2]))
        
        # Check if assignment already exists
        if manager_id in self.lane_assignments:
            return self.lane_assignments[manager_id]
        
        # Divide lanes among managers
        total_lanes = len(self.all_lanes)
[... 917 lines cut ...]
    args = parser.parse_args()
    
    if args.mode == "record":
        # 运行记录模式
        recording_path = run_record_mode(args.num_managers, args.steps)
        print(f"Recording saved to {recording_path}")
        
    elif args.mode == "replay":
        # 运行重放模式
        if not args.recording:
            print("Error: --recording path must be specified for replay mode")
        else:
            run_replay_mode(args.recording, args.steps)
            
    elif args.mode == "compare":
        # 运行比较
        if not args.recording:
            print("Error: --recording path must be specified for compare mode")
        else:
            compare_vehicle_states(args.recording, args.steps)
            
    else:
        print(f"Unknown mode: {args.mode}")

```

AOOOOOA, 2025-04-23:
Hi @pengzhenghao @QuanyiLi , could you please follow up on this issue when you get a chance? Thanks!

QuanyiLi, 2025-04-25:
Hi @AOOOOOA,

Sorry for the late reply. It seems you made a new recording/replaying module, but I remember that we have an official implementation of this. The exact use example can be found here: https://github.com/metadriverse/metadrive/tree/main/metadrive/tests/test_export_record_scenario
The output is a dictionary containing the map and all objects.

Could you try using this instead, or modify it according to your needs? 

Best,
Quanyi

AOOOOOA, 2025-04-28:
> Hi [@AOOOOOA](https://github.com/AOOOOOA),
>
> Sorry for the late reply. It seems you made a new recording/replaying module, but I remember that we have an official implementation of this. The exact [...]

AOOOOOA, 2025-04-28:
> Hi [@AOOOOOA](https://github.com/AOOOOOA),
>
> Sorry for the late reply. It seems you made a new recording/replaying module, but I remember that we have an official implementation of this. The exact [...]

Sure, I will check whether the official replay module can meet our requirements and whether the issue still persists. Thanks.