metadriverse/metadrive #839: Initialize the ego car

StevenChengithub, 2025-05-02:
Hi, how can I spawn/initialize the ego vehicle at a specific coordinate (x, y)?
Thanks

baserker2, 2025-05-02:
try:
    env.reset()
    # obj_2 = env.engine.spawn_object(Pedestrian, position=[30, 10])
    # obj_2 = env.engine.spawn_object(Cyclist, position=[30, 10],heading=0)
    # obj_2 = env.engine.spawn_object(DefaultVehicle, position=[30, 10] ,vehicle_config=env.config["vehicle_config"],heading=0)
    # obj_2 = env.engine.spawn_object(ScenarioTrafficLight, position=[30, 10],lane)
    for i in range(150):
        # v.before_step([0, 0.5])
        o, r, d, _, _ = env.step([0,1])
        rgb_render()
        # top_down_render()
    generate_gif(frames,'demo2_6.gif')
    # gen_video(frames,'demo2')
finally:
    env.close()

just like this

StevenChengithub, 2025-05-02:
> try: env.reset() # obj_2 = env.engine.spawn_object(Pedestrian, position=[30, 10]) # obj_2 = env.engine.spawn_object(Cyclist, position=[30, 10],heading=0) # obj_2 = [...]

emmm I mean the ego car @QuanyiLi